earlephilhower/arduino-pico #1707: Feature Request: Better Joystick Library

OrhanYigitDurmaz, 2023-09-12:
We definitely need a better joystick library. The current one doesnt support the 16-bit resolution mode, nor changing the device descriptors

earlephilhower, 2023-09-12:
If you'd like to make a PR, that would be great.

Changing device names is pretty trivial to add.

The 16 bit issue is that TinyUSB doesn't have a built in HID descriptor with that resolution, so you'll need to build your own.  It's not very complicated, and once done it's relatively simple to change the HID report format.

You might want to do a PR upstream, there in fact.  HDJoystick or something added.

OrhanYigitDurmaz, 2023-09-13:
I'm bypassing the name problem by creating my own board using makeboards.py .

For the PR, I should start first with the bare Pico SDK first, then build the library using the information that i got from the Pico SDK.

Thank you for your response.

closed 2023-09-13 by OrhanYigitDurmaz

OrhanYigitDurmaz, 2023-09-14:
Here is a valuable document: https://interface.cqpub.co.jp/wp-content/uploads/0978f2e66a615b1023b910d4111ee347.pdf

OrhanYigitDurmaz, 2023-09-15:
Here is where the report descriptor templates are: https://github.com/hathach/tinyusb/blob/46f7cf4da2959b96d3af6b7f23d0d02ff7187f2f/src/class/hid/hid_device.h

Took a reaaly long time to find out.
Dont ask why 😁