earlephilhower/arduino-pico #1716: Platformio debugging: Watching class member variables doesn't work

LinusHeu, 2023-09-17:
[image]

In the bottom right, an error message pops up:

> There is no member or method named private. (from var-create watch_d1fa06f3898bc8ff4df7d0a42ff362c8} @ "obj.private.privat")

This bug already happened before the update. I just noticed it again.

earlephilhower, 2023-09-17:
I think this is a general Platform.IO thing and unrelated to the core running.  `public` and `private` are simple C++ keywords...

maxgerhardt, 2023-09-18:
The "public" and "private" displays are just in the debugger view so that you know where these variables were declared.

I think that in the watch window, you should just access the variables directly as you would in code, i.e.,
```
obj.i
obj.privat
```

closed 2023-09-18 by earlephilhower

LinusHeu, 2023-09-18:
Yes, that's it! Thanks!
Then I suppose it's just a UI/Platformion issue. Because right-clicking a variable -> "add to watch" automatically adds .private/.public.

maxgerhardt, 2023-09-18:
Huh I actually never tried to add a variable watch that way. I think you have a good case to open a bug at https://github.com/platformio/platformio-vscode-ide/issues/ to better the debugging UI?