earlephilhower/arduino-pico #1718: arduino-pico core has NO auto-reset support

mohdrais, 2023-09-19:
Hello Sir, 
I am using Windows11 Version 22H2, Arduino IDE 2.2.1 and Pico W ( Pico /RP2040 Version 3.5.0). The arduino-pico core has NO auto-reset. When the pico is attached to the PC, there is only PORT: COM1. The sketch can be uploaded by pressing the BOOTSEL but I need the other COM so as to use Serial Monitor. Last time there is no problem but I forgot both the version for Arduino IDE and Pico/RP2040. Thank you in advance for any suggestion.

maxgerhardt, 2023-09-19:
Yes the core has an auto-reset via serial, that was not removed. It seems more like the sketch you uploaded crashed the Pico or the wrong board was selected?

* What exact board do you have?
* What exact Tool menu settings have you set? 
* Can you run the minimal sketch
```cpp
void setup() { Serial.begin(115200); }
void loop() {}
```
on it and it will still not show a COM port?

mohdrais, 2023-09-19:
Sir,
I am really sorry for the mistake. It seems that the sketch I uploaded crash the pico. ( I tested with blink sketch and it works!) Thank you very much.

closed 2023-09-19 by earlephilhower