earlephilhower/arduino-pico #1723: W5500 not working on SPI1

Ing-Dom, 2023-09-21:
While I managed to get a W5500 working on SPI, it does not work on SPI1.

this is my init code:

```
Wiznet5500lwIP eth(PIN_SS_, SPI_INTERFACE);

...
    SPI_INTERFACE.setRX(PIN_MISO_);
    SPI_INTERFACE.setTX(PIN_MOSI_);
    SPI_INTERFACE.setSCK(PIN_SCK_);
    SPI_INTERFACE.setCS(PIN_SS_);
   eth.begin();
``` 

with these definitions used on a PiPico, it works.
``` 
#define SPI_INTERFACE SPI // SPI or SPI1, depends on the pins
#define PIN_MISO_ (16)
#define PIN_MOSI_ (19)
#define PIN_SCK_ (18)
#define PIN_SS_ (17)
``` 

with that, it does not work:
``` 
#define SPI_INTERFACE SPI1 // SPI or SPI1, depends on the pins
#define PIN_MISO_ (12)
#define PIN_MOSI_ (15)
#define PIN_SCK_ (14)
#define PIN_SS_ (13)
``` 

it looks like there is something wrong with SPI1 usage, but I cannot tell what.
Did I forget someting?

I can almost certain exclude wiring as a source of this error, because I had both setups running previously with the Ethernet_Generic stack.

earlephilhower, 2023-09-21:
What exactly is the way it fails?  During init it reports no Hw found?  It hangs the Pico?

Ing-Dom, 2023-09-21:
bool Wiznet5500::begin(const uint8_t* mac_address, netif *net)

returns false

earlephilhower, 2023-09-21:
I'm traveling so can't look into it now, but if you could enable Tools->Debug Port->Serial, and then enable Tools->Debug Level->SPI that would catch some handy logs.

Ing-Dom, 2023-09-21:
to really excluide the possibility that my code is failing here I got back to the w5500 example.
Same behaviour here.
I normally do not develop in the Arduino IDE, but let me switch over to a different machine...

Ing-Dom, 2023-09-21:
so, this is the output:

```

Starting Ethernet port
Link: 0
Link: 0
Eth begin
SPI::begin(0), rx=16, cs=17, sck=18, tx=19
SPI::beginTransaction(clk=4000000, bo=MSB)
SPI: initting SPI
SPI: actual baudrate=4000000
SPI: IRQ masks before = 00000000 00000000 00000000 00000000
SPI: IRQ masks after = 00000000 00000000 00000000 00000000
SPI::endTransaction()
SPI: IRQ masks = 00000000 00000000 00000000 00000000
SPI::beginTransaction(clk=4000000, bo=MSB)
SPI: Reusing existing initted SPI
SPI: IRQ masks before =SPI::endTransaction()
SPI: IRQ masks = 00000000 00000000 0000000No wired Ethernet hardware detected. Check pinouts, wiring.
```

the complete sketch 
```
/*
    This sketch establishes a TCP connection to a "quote of the day" service.
    It sends a "hello" message, and then prints received data.
*/

#include <W5500lwIP.h>

const char* host = "djxmmx.net";
const uint16_t port = 17;

Wiznet5500lwIP eth(17, SPI1);
WiFiUDP Udp;

void setup() {
  // Set up SPI pinout to match your HW
//  SPI.setRX(16);
//  SPI.setTX(19);
//  SPI.setSCK(18);
//  SPI.setCS(17);

    SPI1.setRX(12);
  SPI1.setTX(15);
  SPI1.setSCK(14);
  SPI1.setCS(13);

  Serial.begin(115200);
  delay(5000);
  Serial.println();
  Serial.println();
  Serial.println("Starting Ethernet port");

  // Start the Ethernet port
  IPAddress localip(192,168,123,15);
  IPAddress gwip(192,168,123,1);
  IPAddress netmask(255,255,255,0);
  IPAddress dns1(8,8,8,8);
  IPAddress dns2(4,4,4,4);
  IPAddress knx(4,4,4,4);

  Serial.print("Link: ");
  Serial.println(eth.isLinked());

  //eth.config(localip, gwip, netmask, dns1, dns2);

  Serial.print("Link: ");
  Serial.println(eth.isLinked());
  
  Serial.println("Eth begin");
  uint32_t start = millis();
  bool r = eth.begin();
  if (!r)
  {
    Serial.println("No wired Ethernet hardware detected. Check pinouts, wiring.");
    while (1) {
      delay(1000);
    }
  }
 uint8_t* mymac;
 mymac = eth.getNetIf()->hwaddr;

 Serial.print("MAC-Adress: ");
 Serial.print(mymac[0], HEX);
 Serial.print(" ");
  Serial.print(mymac[1], HEX);
 Serial.print(" ");
  Serial.print(mymac[2], HEX);
 Serial.print(" ");
  Serial.print(mymac[3], HEX);
 Serial.print(" ");
  Serial.print(mymac[4], HEX);
 Serial.print(" ");
  Serial.print(mymac[5], HEX);
 Serial.println(" ");

  while (!eth.connected()) {
    Serial.print(".");
    delay(500);
  }

  Serial.println("");
  Serial.println("Ethernet connected");
  Serial.println("IP address: ");
  Serial.println(eth.localIP());

}

void loop() {
  Serial.print("Link: ");
  Serial.println(eth.isLinked());
  delay(1000);
}

```

Ing-Dom, 2023-09-21:
and this is the ouput of the working setup with SPI:'

```
Starting Ethernet port
Link: 0
Link: 0
Eth begin
SPI::begin(0), rx=16, cs=17, sck=18, tx=19
SPI::beginTransaction(clk=4000000, bo=MSB)
SPI: initting SPI
SPI: actual baudrate=4000000
SPI: IRQ masks before = 00000000 00000000 00000000 00000000
SPI: IRQ masks after = 00000000 00000000 00000000 00000000
SPI::endTransaction()
SPI: IRQ masks = 00000000SPI::beginTransaction(clk=4000000, bo=MSB)
SPI: Reusing existing initted SPI
SPI: IRQ masks before = 00000000 00000000 00000000 SPI::endTransaction()
SPI: IRQ masks = 00000000 00000000 0000000SPI::transfer(00), cpol=0, cpha=0
SPI: read back 01
SPI::transfer(00), cpol=0, cpha=0
SPI: read back 02
SPI::transfer(04), cpol=0, cpha=0
SPI: read back 03
SPI::transfer(80), cpol=0, cpha=0
```

Ing-Dom, 2023-09-21:
now, I see it. Somehow, the wrong SPI interface is used?!

It seems that Wiznet5500lwIP eth(17, SPI1);
does not correctly select SPI1.. but SPI is used.

earlephilhower, 2023-09-21:
Could well be since the ESP8266 I took out from only has the one port!

It should be a search and replace in the W5500.cpp file of `SPI` with (iirc) `_spis`.

Ing-Dom, 2023-09-21:
no, in w5500.cpp, _spi is used. I already checked that.

I think the assignment does not work as intended.
It must admit this is a little above my C++ skills...

```
template<class RawDev>
class LwipIntfDev: public LwipIntf, public RawDev {
public:
    LwipIntfDev(int8_t cs = SS, SPIClass& spi = SPI, int8_t intr = -1) :
        RawDev(cs, spi, intr), _mtu(DEFAULT_MTU), _intrPin(intr), _started(false), _default(false) {
        _spiUnit = spi;
        memset(&_netif, 0, sizeof(_netif));
    }
```

PR #1725 (2023-09-22): _spiUnit is a reference, and when initialized with SPI, it cannot be …
…changed in the constructor afterwards.

So initialize it in the constructor's declaration.

fixes #1723
closed 2023-09-22 by earlephilhower