earlephilhower/arduino-pico #1728: Unable to flash Raspberry Pico W over the OTA

Jaishankar872, 2023-09-24:
Hi
   I am using Raspberry Pico W official board from Raspberry Pi

   And I used the example of BasicOTA, 
[image]

Flash the Raspberry Pico W over the OTA, getting failed with below Error

08:45:01 [ERROR]: Bad Answer: ERR: No Filesystem
Failed uploading: uploading error: exit status 1

[image]

earlephilhower, 2023-09-24:
>  08:45:01 [ERROR]: Bad Answer: ERR: No Filesystem

You need to have a Filesystem to do OTA.   Use the Tool->Flash Size to assign a FS setaside.

closed 2023-09-24 by earlephilhower

Jaishankar872, 2023-09-24:
Hi 
Thanks for your fast response

Now, new error is coming as below

Usage: espota.py [options]

espota.py: error: option -p: invalid integer value: '{upload.port.properties.port}'
Failed uploading: uploading error: exit status 2

[image]

bartgrefte, 2024-03-07:
> Hi Thanks for your fast response
>
> Now, new error is coming as below
>
> Usage: espota.py [options]
>
> espota.py: error: option -p: invalid integer value: '{upload.port.properties.port}' Failed uploading: [...]

Same error here:
```
Sketch uses 398820 bytes (19%) of program storage space. Maximum is 2093056 bytes.
Global variables use 73356 bytes (27%) of dynamic memory, leaving 188788 bytes for local variables. Maximum is 262144 bytes.
Usage: espota.py [options]

espota.py: error: option -p: invalid integer value: '{upload.port.properties.port}'
Failed uploading: uploading error: exit status 2
```

Also running into https://forum.arduino.cc/t/ide-2-0-ota-mdns-discovery-only-intermittently-finds-nodes/1036850/2 , not sure if related. 

As for the error, I've checked both the firewall on my router as well as ESET Internet Security on the Arduino IDE W10 host, nothing is being blocked as far as I can tell.