earlephilhower/arduino-pico #1730: Feature Request: Non-Blocking I2C

Rastloser, 2023-09-24:
I don't know of a non-blocking or perhaps even DMA I2C-Library for the RP2040-Arduino-Core. There is a DMA-library for FreeRTOS under [PICO-I2C-DMA](https://github.com/fivdi/pico-i2c-dma/tree/master), but it would be great to have something Arduino compatible...

earlephilhower, 2023-09-24:
That library may use DMA, but it definitely is blocking on the caller (i.e. a call to i2c-read-write-dma takes exactly as many wall seconds to complete as a i2c-read-write-polling call would).

I guess in the case of FreeRTOS that lets you run another task instead of busy waiting, but here we don't have tasks (unless you use FreeRTOS...in which case the library may almost work as-is).

I'm not sure what the API would be here, though, where there are no tasks.  You'd need a callback function or something (called ar IRQ time) or a flag your app could poll to see if the I2C op is really done or not.

Rastloser, 2023-09-24:
I'd personally go with a callback, but a polled function/flag would be a solid choice, too. Maybe even preferably since some people might put too much code into the callback-handler...
The [Teensy-Version](https://github.com/Richard-Gemmell/teensy4_i2c) uses a finished flag, but does expose IRQ-based callbacks with ample warning not to put lengthy code into it...

earlephilhower, 2024-05-21:
If anyone looking at this request could take a look at #2167 and try their hand at using it in their application I'd much appreciate it.  I have the TalkingToMyself test running in asynchronous mode stably, but don't seem to have any I2C gadgets to hoop up and try on external HW...

PR #2167 (2024-05-21): Add asynchronous I2C read and write operations
Fixes #1730

Uses DMA operations to avoid the need to bit-bang or busy wait for I2C operations that might be very slow.  Optional, adds new API calls to enable.  Simple example included.
commit 11814823 (2024-05-21): Add asynchronous I2C read and write operations (#2167)

Fixes #1730

Uses DMA operations to avoid the need to bit-bang or busy wait for I2C operations
that might be very slow.  Optional, adds new API calls to enable.  Simple example
included.
closed 2024-05-21 by earlephilhower