earlephilhower/arduino-pico #1735: Bluetooth Characteristic Callback Issue

jhughesbiot, 2023-09-28:
Hardware: Pi Pico W
Library Version: 3.6.0

I am not sure if it I am doing something wrong with the Bluetooth implementation or this is an actual bug but it does appear the callbacks are not working as expected in the current Bluetooth stack.

I am connecting to a Nano33BLESense that has 1 characteristic that is setup to notify and writes a new value every 5 seconds.  I have tested the notify options with an android app to verify notification updates are working on the nano 

I modified the LECentral code slightly to try to get updates from characteristic subscriptions but these callbacks do not appear to work.   

Basic state order:
1. Connect to peripheral (Nano33BLESense)
2. Discover characteristics
3. Subscribe to characteristic
4. Wait for  characteristic updates

1-2 work as expected
3. Subscribe callback will always return BLEState of BLE_STATUS_OTHER_ERROR when you attempt to subscribe, no matter how many times you attempt.  If you perform a read of the characteristic, then subscribe, you will receive a subscribe call back status of BLE_STATUS_OK
4. Never receives any callback even after a subscribe callback of BLE_STATUS_OK

Additional testing on just reading the characteristic value every second resulted in a subscribe callback every time a read callback was completed.  No further subscriptions are occurring past the very first successful subscription but a successful read callback in turn receives a subscription callback. 

The gattCharacteristicNotification is never called despite successful subscription callback status and continual update of the characteristic in the nano.

Console Output showing connection and initial subscription failure, read, then success:
```
10:52:49.616 -> BLE ShieldService V2 found!
10:52:49.616 -> 
10:52:49.804 -> Device connected!
10:52:50.692 -> Service Discovered: :1800
10:52:50.692 -> Service Discovered: :1801
10:52:50.786 -> Service Discovered: :B8E06067-62AD-41BA-9231-206AE80AB550
10:52:50.786 -> Our service located!
10:52:50.879 -> Service discovery finished
10:52:50.972 -> Characteristic Discovered: F897177B-AEE8-4767-8ECC-CC694FD5FCEE, handle 0xC
10:52:50.972 -> Characteristic found: RX
10:52:50.972 -> Characteristic Discovered: BF45E40A-DE2A-4BC8-BBA0-E5D6065F1B4B, handle 0xF
10:52:50.972 -> Characteristic found: TX
10:52:50.972 -> Characteristic Discovered: 2FBC0F31-726A-4014-B9FE-C8BE0652E982, handle 0x11
10:52:50.972 -> Characteristic found: Baudrate
10:52:50.972 -> Characteristic Discovered: 1D132F28-5D85-11EE-8C99-0242AC120002, handle 0x13
10:52:50.972 -> Characteristic found: Intent
10:52:50.972 -> Characteristic discovery finished, status 1
10:52:50.972 -> Subscribing to charRX
10:52:50.972 -> Subscribing
10:52:51.066 -> 
10:52:51.066 -> Subscribe Callback: 4
10:52:51.066 -> BLE_STATUS_OTHER_ERROR
10:52:51.160 -> Read callback
10:52:51.160 -> Read Value:108 
10:52:51.160 -> 
10:52:51.160 -> Subscribe Callback: 0
10:52:51.160 -> BLE_STATUS_OK
```

One second reads showing a updates from nano but also a seemingly random subscription callback during every read and no gattCharacteristicNotification ever received:
```
`10:52:58.091 -> Subscribe Callback: 0
10:52:58.091 -> BLE_STATUS_OK
10:52:58.978 -> Reading...
10:52:59.071 -> Read callback
10:52:59.071 -> Read Value:80 
10:52:59.071 -> 
10:52:59.071 -> Subscribe Callback: 0
10:52:59.071 -> BLE_STATUS_OK
10:52:59.956 -> Reading...
10:53:00.049 -> Read callback
10:53:00.049 -> Read Value:80 
10:53:00.049 -> 
10:53:00.049 -> Subscribe Callback: 0
10:53:00.049 -> BLE_STATUS_OK
[... 430 lines cut ...]
  Serial.println("- Discovering central device...");
  delay(500);

  if (central) {
    Serial.println("* Connected to central device!");
    Serial.print("* Device MAC address: ");
    Serial.println(central.address());
    Serial.println(" ");
    lastWrite = millis();
    byte val = 0x50;
    while (central.connected()) {
      BLE.poll();
      central.poll();
        if( (millis() - lastWrite) > 5000){ 
          Serial.print("Writing RX: ");
          Serial.println(val);
          picoVoiceCharacteristicRx.writeValue(val);
          lastWrite = millis();
          val+=1;
        }
    }
    Serial.println("* Disconnected to central device!");
  }   
}
```

earlephilhower, 2023-09-29:
@jhughesbiot thanks for the detailed MCVE.  Would you know of a phone app or something similar that could replace the NanoBLE?  I don't have one or anything compatible with `ArduinoBLE.h` AFAIK, so can't really look into this.

Because this is using the BTStack Arduino lib (and maybe some native calls) I would also suggest opening up an issue on the https://github.com/bluekitchen/btstack repo.  They might be able to just look at the code and spot something.

jhughesbiot, 2023-09-29:
I only have the NanoBLE, no app I know of that can simulate it's behavior.  According to the official ArduinoBLE docs, these are all the supported boards, but I only have the NanoBLE myself.

[image]

I reported the issue directly to btstack as well.

jhughesbiot, 2023-10-04:
Feedback from BTstack is to use the native BTstack API instead of that project, not really sure what that means or if it even works here.  And the other thing to help diagnose is to turn on the debugging using this as a reference

https://github.com/raspberrypi/pico-examples/pull/427/files

I don't see this option or equivalent in the current API so not really sure how to get the extra debug available for troubleshooting further.

earlephilhower, 2023-10-08:
Closing as this is a ArduinoBLE issue per the BlueKitchen folks, being tracker in the Arduino repo.

closed 2023-10-08 by earlephilhower