earlephilhower/arduino-pico #1742: ArduinoOTA with W5100S-EVB-Pico

ZinnerC, 2023-10-01:
I might need some help.

I try to implement OTA for a project of mine.
The server and all the Ethernet stuff is working.
But if I start to add ArduinoOTA (based on example/BasicOTA), my setup() hangs at ArduinoOTA.begin() and I don't know why.

I tried example/BasicOTA and only removed Wifi stuff and included Ethernet but I always get "IP unset".

Could somebody point me in the right direction.

The codes I use are the following:

working server with OTA added first and second code sections.
With this code every thing is fine as long as ArduinoOTA.begin is commented.
As soon I uncomment this line the code hangs there.

The third code section is example/BasicOTA with the change to Ethernet

```c++
/****************************************************************************************************************************
  PostServer.h - Dead simple web-server for Ethernet shields

  EthernetWebServer is a library for the Ethernet shields to run WebServer

  Based on and modified from ESP8266 https://github.com/esp8266/Arduino/releases
  Built by Khoi Hoang https://github.com/khoih-prog/EthernetWebServer
  Licensed under MIT license
 *****************************************************************************************************************************/
/*
   The Arduino board communicates with the shield using the SPI bus. This is on digital pins 11, 12, and 13 on the Uno
   and pins 50, 51, and 52 on the Mega. On both boards, pin 10 is used as SS. On the Mega, the hardware SS pin, 53,
   is not used to select the Ethernet controller chip, but it must be kept as an output or the SPI interface won't work.
*/

#include "defines.h"
#include "Wire.h"
#include "jled.h"
#include "ArduinoJson.h"
#include "ArduinoOTA.h"

#define DEBUG           true
#if DEBUG
    #define debug(x)       Serial.print(x)
    #define debugF(x,y)     Serial.print(x,y)
    #define debugln(x)     Serial.println(x)
    #define debugFln(x,y)   Serial.println(x,y)
#else
    #define debug(x)
    #define debugF(x,y)
    #define debugln(x)
    #define debugFln(x,y)
#endif

#define ETHERNET_DHCP   true

#define PROXY           1

#define PROXY_FIRMWARE 0.25

// DONT MODIFY BELOW HERE

/* CLIENT VARIABLES BELOW */
/*------------------------*/

// I2C Client Status Definitions
#define STATUS_LAST_COMMAND_SUCCESS 1
#define STATUS_LAST_COMMAND_KNOWN 2
#define STATUS_COMMAND_PROCESSING 3

EthernetWebServer server(80);
[... 875 lines cut ...]
  });
  ArduinoOTA.onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) {
      Serial.println("Auth Failed");
    } else if (error == OTA_BEGIN_ERROR) {
      Serial.println("Begin Failed");
    } else if (error == OTA_CONNECT_ERROR) {
      Serial.println("Connect Failed");
    } else if (error == OTA_RECEIVE_ERROR) {
      Serial.println("Receive Failed");
    } else if (error == OTA_END_ERROR) {
      Serial.println("End Failed");
    }
  });
  ArduinoOTA.begin();
  Serial.println("Ready");
  Serial.print("IP address: ");
  Serial.println(Ethernet.localIP());
}

void loop() {
  ArduinoOTA.handle();
}
```

JAndrassy, 2023-10-01:
this library doesn't have a BasicOTA example

earlephilhower, 2023-10-01:
I'm kind of confused with your example code.

I *think* you're mixing the native `LWIPEthernet` stuff and the Arduino-separate `Ethernet`, and this core's native ArduinoOTA vs the generic multi-board ArduinoOTA.  That's never going to work, you'll need to pick one or the other set to use.

See the [W5100 example](https://github.com/earlephilhower/arduino-pico/blob/master/libraries/lwIP_w5100/examples/WiFiClient-W5100/WiFiClient-W5100.ino) and the [docs](https://github.com/earlephilhower/arduino-pico/blob/master/docs/ethernet.rst) for info on how to use the integrated LWIP code.

ZinnerC, 2023-10-01:
Thank you for your response.

@JAndrassy Well probably I was not precise enough.
[image]
I ment the example from the RP2040 board core (see screenshot)

@earlephilhower Thanky you for the hint. I just took for "BasicOTA_Ethernet.ino" the BasicOTA from the examples from ArduinoOTA from within RP2040 board core (see screenshot).

I will check out how I can use the mentioned example as a starting point. How can I "force" to use the core's native ArduinoOTA? Does it have an other name?

Based on this information I for sure can find a solution. Sadly not in the coming week. Will keep this thread informed when I succeeded. Might help one or the other at a later point.

earlephilhower, 2023-10-01:
> @earlephilhower Thanky you for the hint. I just took for "BasicOTA_Ethernet.ino" the BasicOTA from the examples from ArduinoOTA from within RP2040 board core (see screenshot).

That's the problem, you're using `Ethernet` which does not actually run IP on the Pico and is incompatible with the ArduinoOTA in this core.  See the example I listed before, where it includes `W5100lwIP.h` and instantiates its own object in the sketch to enable the core's built-in wired Ethernet support.

Alternatively, you should be able to use the Arduino `Ethernet` and the generic `ArduinoOTA` , but I can't really help you there.  You should open something on their repos with questions related to it.

earlephilhower, 2023-10-07:
Closing as this is just a mixing of the core LWIP wired Ethernet and the Arduino `Ethernet` library.

closed 2023-10-07 by earlephilhower