earlephilhower/arduino-pico #1746: SPISlave cannot communicate with other arduino with MODE0 or MODE2

asukiaaa, 2023-10-02:
Thank you for sharing useful environment.
I tried SPIClass and have a problem.

I tried SPI communication with the following connection.
Arduino Uno R3 as SPI Master - 5V to 3.3V level shifter - Pi Pico SPI Slave

I succeeded in communication with MODE1 or MODE3 but cannot with MODE0 or MODE2.
Do you have any idea to resolve problem of MODE0 and MODE2?
Is it problem of pico-sdk?

The code what I tried.

Pi Pico slave
```cpp
#include <Arduino.h>
#include <SPISlave.h>

#if true
#define PIN_SPI_CS 17
#define PIN_SPI_MOSI_RX 16
#define PIN_SPI_MISO_TX 19
#define PIN_SPI_SCK 18
#define SPI_SLAVE SPISlave
#else
#define PIN_SPI_CS 9
#define PIN_SPI_MOSI_RX 8
#define PIN_SPI_MISO_TX 11
#define PIN_SPI_SCK 10
#define SPI_SLAVE SPISlave1
#endif

SPISettings spisettings(1000000, MSBFIRST, SPI_MODE0); // OK for MODE 1 and 2, bad for 0 and 2

void setup() {}
void loop() { delay(5000); }

volatile bool recvBuffReady = false;
const uint8_t lenBuff = 20;
char recvBuff[lenBuff] = "";
int recvIdx = 0;
void recvCallback(uint8_t *data, size_t len) {
  if (recvIdx + len >= lenBuff) {
    len = lenBuff - recvIdx;
  }
  if (len > 0) {
    memcpy(recvBuff + recvIdx, data, len);
    recvIdx += len;
  }
  while (recvIdx >= lenBuff) {
    recvBuffReady = true;
    recvIdx -= lenBuff;
  }
}

int sendcbs = 0;
char sendBuff[10];
int sendIndex = 0;
void sentCallback() {
  SPI_SLAVE.setData((uint8_t *)&sendBuff[sendIndex], 1);
  ++sendIndex;
}

void setup1() {
  SPI_SLAVE.setRX(PIN_SPI_MOSI_RX);
  SPI_SLAVE.setCS(PIN_SPI_CS);
  SPI_SLAVE.setSCK(PIN_SPI_SCK);
  SPI_SLAVE.setTX(PIN_SPI_MISO_TX);
  sentCallback();
  SPI_SLAVE.onDataRecv(recvCallback);
  SPI_SLAVE.onDataSent(sentCallback);
  SPI_SLAVE.begin(spisettings);
  delay(3000);
  Serial.println("S-INFO: SPISlave started");
}

void loop1() {
  if (digitalRead(PIN_SPI_CS) == HIGH && recvIdx > 0) {
    Serial.printf("S-RECV: '%s'\n", recvBuff);
    recvBuffReady = false;
    recvIdx = 0;
    sendIndex = 0;
    ++sendcbs;
    sprintf(sendBuff, "hi %d", sendcbs);
    Serial.printf("set sendBuff '%s'\n", sendBuff);
  }
}
```

Arduino Uno R3 master
```cpp
#include <Arduino.h>
#include <SPI.h>

SPISettings spisettings(1000000, MSBFIRST, SPI_MODE0); // OK for MODE 1 and 2, bad for 0 and 2
#define PIN_CS SS
#define SPI_MASTER SPI

void setup(void) {
  Serial.begin(115200);
  Serial.println();
  pinMode(PIN_CS, OUTPUT);
  digitalWrite(PIN_CS, HIGH);
  SPI_MASTER.begin();
  SPI_MASTER.setDataMode(SPI_MODE0);
}

byte transferAndWait(const byte what) {
  byte a = SPI_MASTER.transfer(what);
  delayMicroseconds(20);
  return a;
}

void loop(void) {
  char data[10];
  sprintf(data, "hello ");
  SPI_MASTER.beginTransaction(spisettings);
  digitalWrite(PIN_CS, LOW);
  SPI_MASTER.transfer(data, sizeof(data));
  SPI_MASTER.endTransaction();
  digitalWrite(PIN_CS, HIGH);
  delay(10);
  Serial.println("hello results:");
  for (size_t i = 0; i < sizeof(data); i++) {
    // Serial.print(data[i]);
    Serial.print(data[i], HEX);
    Serial.print(' ');
  }
  Serial.println();
  for (size_t i = 0; i < sizeof(data); i++) {
    Serial.print(data[i]);
    // Serial.print(data[i], HEX);
  }
  Serial.println();

  Serial.println("at " + String(millis()));
  Serial.flush();
  delay(1000);
}
```

earlephilhower, 2023-10-02:
Unfortunately I haven't had access to an Uno in something like 10 years.  

What exactly do you mean it's not communicating?  That it's not sending/receiving at all?  That it's getting bad data?  That things are shifted, etc?

Are you able to get traces of the SPI signals between the two devices in the passing and failing modes?  That would help diagnose things.

FWIW MODE0/2 are both `SPI_CHPA_0` meaning the opposite edge is used to sample data from MODE1/3, so that's worth exploring.  Especially if there's HW between the 2 devices like the level shifters, could the setup/hold times be broken because of it?

asukiaaa, 2023-10-08:
Thank you for the respoinse.

> What exactly do you mean it's not communicating? That it's not sending/receiving at all? That it's getting bad data? That things are shifted, etc?

Slave basically receive nothing but sometimes get first byte.

> Are you able to get traces of the SPI signals between the two devices in the passing and failing modes? 

I checked the signals from Pico and ProMicro(ATMega32u4) then found that SS signal is different betreen Pico and Arduino(ATMega32u4).

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

earlephilhower, 2023-10-08:
Sorry, but it's pretty impossible to tell anything when the clocks and MOSI/MISO are on different graphs and overlaid.  Since the voltage swings seem fine, can you use a logic analyzer instead and capture all channels at the same time to look at?

earlephilhower, 2023-10-08:
Also, we need enough detail to identify when the data is changing relative to the clock on MOSI and MISO, especially since this is a weird clocking mode being debugged.  So higher sampling rate is better. :)

asukiaaa, 2023-10-09:
Thank you for the response but I don't have logic analyzer so I cannot show information any more now.

earlephilhower, 2023-10-09:
Unfortunately, I don't think there's anything we can do here then since this is hardware and timing related.  The different SPI modes only differ in 2 setup calls (the CPOL and CPHA settings).  Looking at the code, the settings for MODE0..3 match the info found in the Analog Devices whitepaper.  Given the clock/data relationships this could well be related to setup and hold times which are outside our control here.

So, closing for now.  If we can get some good hi-res traces with all signals we can look at it some more...

asukiaaa, 2023-10-10:
The problem is not resolved so I think it's better to keep open state.

earlephilhower, 2023-10-11:
Looking into it, those calls are not used in the SPISlave object, only in the normal SPI master one.

closed 2023-10-11 by earlephilhower