earlephilhower/arduino-pico #1749: Core freeze due to stack memory

Ing-Dom, 2023-10-04:
We (@OpenKNX) had some weird issues, especially somehow linked to dualcore and sprintf.. drove me crazy for quite some time.
Finally we tracked it down to stack size issues - there seems to be a fairly small max. stack size (compared to the sram size of the rp2040).

There were discussion in the rpi forums about this as well:
https://forums.raspberrypi.com/viewtopic.php?t=335833

For now, we assume that the stack is limited to 2k each core when using arduino-pico ?
Is that true?

The pico-sdk is precompiled, so I did not find the values for 
PICO_STACK_SIZE
PICO_CORE1_STACK_SIZE
PICO_HEAP_SIZE

but I assume they are default?

That would mean 2k for each core.

In the memmap 4k seem to be reserved..
https://github.com/earlephilhower/arduino-pico/blob/master/lib/memmap_default.ld#L24-L30

1) Can you confirm this?

2) Is there a possibility to increase max stack size without changig the core? And if not, how do you think about having some modifications to the core enabling this ?
There is also multicore_launch_core1_with_stack()
https://www.raspberrypi.com/documentation/pico-sdk/high_level.html#gac1926bb2da43016e5d14eb52c8c123f4

earlephilhower, 2023-10-04:
We use the SDK defaults for stacks, so with only 1 core running you have 8K but with 2 cores each has 4K.  For a microcontroller it's a pretty common size in my experience.

The Pico has a 256K shared SRAM, used as the heap here, a (2?4?)k RAM in the USB controller for buffers, and 2 sets of 4K fast-access RAM used as the stack for the cores.  These blocks are non-contiguous in RAM.

There's no simple way to change this allocation without some hacking in the linker file.  You could change it to point to a reserved spot in main SRAM for one core, for example.  You could also use thunks and a heap-allocated stack (see the BearSSL implementation which allocated a ~7K stack for operations).

There are GCC options to print warnings when your static function stacks exceed a defined value, those are simple to add and can help pinpoint where you're allocating way too much stack space (and should consider moving things to the heap).

closed 2023-10-07 by earlephilhower

Ing-Dom, 2023-10-16:
Sorry Earle for not responding on time.
I tried your solution and it worked for me.

It could be a little bit more flexible: stacksize of core1 could be configurable, but I'm not sure if this is a relevant use case.

Thank you!

PR #1750 (2023-10-07): Allow full 8K stack for both cores, optionally
Fixes #1749

Defining a global true `bool core1_separate_stack = true` will separate the two cores' stacks, with core 0 using the scratch RAM while core 1 will use 8K from the heap.
commit 7868ddee (2023-10-07): Allow full 8K stack for both cores, optionally (#1750)

Fixes #1749

Defining a global true `bool core1_separate_stack = true` will separate
the two cores' stacks, with core 0 using the scratch RAM while core 1
will use 8K from the heap.