earlephilhower/arduino-pico #1755: Additional doco information on Analog Output Restrictions

palmerr23, 2023-10-07:
Earle, you could update the Analog Output Restrictions section in the Analog I/O doco page to add a sentence to the following, as the extra information is hard to find elsewhere. 

`The PWM generator source clock restricts the legal combinations of frequency and ranges. At a CPU frequency of 133MHz, the 16 bit maximum range decreases by 1 bit for every doubling of the default PWM frequency of 1 kHz. For example, at 1MHz only about 6 bits of range are possible. 
...`

I wrote a little bit of code that set the desired frequency and watched when analogWrite(PIN,1) resulted in no pulse as I increased the analogWriteResolution. Once I saw the pattern, the overall logic became clear.

Again, thanks for the doco - it's really helpful!

Richard

earlephilhower, 2023-10-07:
Again, thanks for reading the docs.  They're always the last thing done but I do try and make sure there's at least a little bit for each new feature included.

Would you like to make a PR with the addition?  To me, honestly, it seems kind of obvious so I probably am not the right person to write it up.

palmerr23, 2023-10-07:
OK, a PR is on its way.

closed 2023-10-07 by earlephilhower