earlephilhower/arduino-pico #1770: Keyboard doesn't work with TINY USB enabled

mkalkbrenner, 2023-10-15:
With platformio I use this simple main.cpp:
```c++
#include <Arduino.h>
#include <Keyboard.h>

void setup() {
  Serial.begin(115200);
  Keyboard.begin();
}

void loop() {
}
```

This works fine unless I switch to Tiny USB by adding `-DUSE_TINYUSB=1`.
With Tiny USB enabled I get a linker error:
```
.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/10.3.0/../../../../arm-none-eabi/bin/ld: .pio/build/pico/lib9e6/Keyboard/Keyboard.cpp.o: in function `_ZN9Keyboard_10sendReportEP9KeyReport':
Keyboard.cpp:(.text._ZN9Keyboard_10sendReportEP9KeyReport+0x22): undefined reference to `_Z24__USBGetKeyboardReportIDv'
```

maxgerhardt, 2023-10-15:
I only see this being defined in RP2040USB.cpp. Does it help when you add the (trivial) function code to your sketch directly?

https://github.com/earlephilhower/arduino-pico/blob/d42f0ab4b0e0da77f59fe4618e5626df81c273f2/cores/rp2040/RP2040USB.cpp#L127-L129

earlephilhower, 2023-10-15:
The Keyboard library and TinyUSB are not compatible.  You need to implement your own Keyboard-like HID device.  The library has examples like this.  Good luck!

closed 2023-10-15 by earlephilhower

maxgerhardt, 2023-10-15:
Ah, I guess he needs [Adafruit_TinyUSB_Arduino/examples/hid_boot_keyboard.ino](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/b17ab37a0e5f4722bbe7996d2ce5a696b2283904/examples/HID/hid_boot_keyboard/hid_boot_keyboard.ino) as a start then.

mkalkbrenner, 2023-10-15:
Thanks for your answers. 
At least, the incompatibility should documented at https://arduino-pico.readthedocs.io/en/latest/usb.html

mkalkbrenner, 2023-10-15:
@earlephilhower @maxgerhardt 
I can't re-open this issue, but want to ask before opening another one.

I think this is a bug. I can't implement anything on my own because the linker already happens if I simply switch to Tiny USB and open a the serial port, even without using the keyboard:

main.cpp:
```c++
#include <Arduino.h>

void setup() {
  Serial.begin(115200);
}

void loop() {
}
```

platformio.ini:
```
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower
board_build.filesystem_size = 0.5m
build_flags =
    -DUSE_TINYUSB=1
```

maxgerhardt, 2023-10-16:
This code should not produce a "undefined ref to _Z24__USBGetKeyboardReportIDv" error -- make sure to delete the `.pio` folder of the project. The code may also need a `#include <Adafruit_TinyUSB.h>` at the start.