earlephilhower/arduino-pico #1778: WiFi.localIP() from WifiNina.h not working

SachiWeera, 2023-10-20:
Hi,

I'm using the Arduino Nano RP2040 connect to connect to my phone's hotspot.

The board does connect to my phone's hotspot, however always consistently freezes at the line **IPAddress ip = WiFi.localIP();**,

My code is given below.

`/*

#include <SPI.h>
#include <WiFiNINA.h>

#include "arduino_secrets.h" 
///////please enter your sensitive data in the Secret tab/arduino_secrets.h
char ssid[] = SECRET_SSID;        // your network SSID (name)
char pass[] = SECRET_PASS;    // your network password (use for WPA, or use as key for WEP)
int keyIndex = 0;                                // your network key index number
int status = WL_IDLE_STATUS;                     // the WiFi radio's status

void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }

  // attempt to connect to WiFi network:
  while (status != WL_CONNECTED) {
    Serial.print("Attempting to connect to WEP network, SSID: ");
    Serial.println(ssid);
    status = WiFi.begin(ssid, keyIndex, pass);

    // wait 10 seconds for connection:
    delay(10000);
  }

  // once you are connected :
  Serial.print("You're connected to the network");
  printCurrentNet();
  printWifiData();
}

void loop() {
  // check the network connection once every 10 seconds:
  delay(5000);
  printCurrentNet();
}

void printWifiData() {
  // print your board's IP address:
  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);
  Serial.println(ip);

  // print your MAC address:
  byte mac[6];
  WiFi.macAddress(mac);
  Serial.print("MAC address: ");
  printMacAddress(mac);
}

void printCurrentNet() {
  // print the SSID of the network you're attached to:
  Serial.print("SSID: ");
  Serial.println(WiFi.SSID());

  // print the MAC address of the router you're attached to:
  byte bssid[6];
  WiFi.BSSID(bssid);
  Serial.print("BSSID: ");
  printMacAddress(bssid);

  // print the received signal strength:
  long rssi = WiFi.RSSI();
  Serial.print("signal strength (RSSI):");
  Serial.println(rssi);

  // print the encryption type:
  byte encryption = WiFi.encryptionType();
  Serial.print("Encryption Type:");
  Serial.println(encryption, HEX);
  Serial.println();
}

void printMacAddress(byte mac[]) {
  for (int i = 5; i >= 0; i--) {
    if (mac[i] < 16) {
      Serial.print("0");
    }
    Serial.print(mac[i], HEX);
    if (i > 0) {
      Serial.print(":");
    }
  }
  Serial.println();
}
`

earlephilhower, 2023-10-20:
Sorry, but I don't think anyone uses this core with the official Arduino board.  It has a completely different Internet setup and is not compatible with the IP infrastructure built in to this core.

I suggest sticking with the official Arduino core for that board.  Good luck.

closed 2023-10-20 by earlephilhower

jaron780, 2023-12-16:
Hello, im getting this exact same issue using the "SimpleWebServerWIFI" example in the WiFiNINA library using your pico core on an Arduino Nano RP2040 connect. The code will run up to that point and then just stop when it reaches the WiFi.localIP() line. I specifically chose the RP2040 connect for my project so that i could host a simple webserver to listen for events and run that webserver/wifi connection on one single core and then run the rest of my project on the other core to handle the actual inputs/outputs of the pins. i need them on separate cores because of the things i run with my project is an stepper motor which is very sensitive to timing when running each step. and having both the wifi/webserver and my main project code on a single thread even using scheduler just does not work properly and causes the stepper motor to have issues. So I was really hoping there would be some way to make use of the second physical core to handle the wifi and web server. To add to this issue a bit for my own projects code, if i remove the call to localIP it can let wifi connect for a bit but after i do something with the AccelStepper library it instantly causes a similar effect where the wifi module seems to freeze or lock up

earlephilhower, 2023-12-16:
Again, as per https://github.com/earlephilhower/arduino-pico/issues/1778#issuecomment-1773306895, you're not going to be able to use the Arduino Nano WiFi with this core.

jaron780, 2023-12-17:
Well, that's the thing, I am able to use WiFi with your pico core on this Arduino Nano RP2040 connect. I found an interesting detail that makes the .localIP() function actually work and its very bizarre. basically you load up the "SimpleWebServerWiFi" example from the WiFiNINA library and right before the call to "printWifiStatus();" you add this: 

  byte bssid[6];
  WiFi.BSSID(bssid);
  
When that simple part is called before .localIP() it for whatever reason it allows .localIP() to be called without freezing the Arduino and WiFi will work mostly properly. its just three's some weird things that cause it to freeze like some memory could be corrupted or something. Like for example if i add the AccelStepper library to the script and call stepper.move(500); the moment i call that the Arduino freezes in the same way that .localIP(); does without the BSSID call. Basically Wifi can work and its so very close to working properly it just seems like three's some weird behavior happening that causes it to freeze when you do something specific. If you would like i have a ino file that i could share that is just a slightly modified "SimpleWebServerWifi" script that has an additional button on the webpage that you can click that calls "stepper.move(500)" which instantly freezes the arduino, i even setup a ping -t on my desktop and pinged the ip of the rp2040 nano and the moment that stepper.move gets called the pings stop responding. so its very easy to replicate on demand.

Basically its seems so very close to being fully functional it just seems like there's some bug somewhere causing this freeze/crash to happen which feels like some form of corrupted memory since unrelated things can seemingly cause the issue to happen

earlephilhower, 2023-12-17:
I'm really surprised at that.  Can you grab a stack dump when it hangs using a picoprobe?  That would at last give a hint.  Given that the MBED core from Arduino has a real time OS running being the scenes in even studied it compiles.

jaron780, 2023-12-17:
I dont actually have a picoprobe or know how to create stacks, bit too new to all of this so still alot to learn

jaron780, 2023-12-17:
Heres the actual code to replicate it, hopefully you or someone else that has a Arduino Nano RP2040 could try to replicate it and get the stack. Just really hope this is fixable because its so very close to working.

Basically just change the SSID/PASS in the arduino_secrets.h file or directly in the code. Then once its booted and connected to your wifi open its ip in a browser and test the top two buttons to turn the blue led on and off, and then press the 3rd button and it should freeze/crash/lockup the arduino. can test without accel stepper by removing the related lines and just removing the call to the .BSSID() line or using the original example provided by WiFiNINA

```
#include <SPI.h>
#include <WiFiNINA.h>
#include <AccelStepper.h>

#define STEPPER_DIR_PIN 2
#define STEPPER_STEP_PIN 3
AccelStepper stepper(AccelStepper::DRIVER, STEPPER_STEP_PIN, STEPPER_DIR_PIN);

#include "arduino_secrets.h" 
///////please enter your sensitive data in the Secret tab/arduino_secrets.h
char ssid[] = SECRET_SSID;        // your network SSID (name)
char pass[] = SECRET_PASS;    // your network password (use for WPA, or use as key for WEP)
int keyIndex = 0;                 // your network key index number (needed only for WEP)

int status = WL_IDLE_STATUS;
WiFiServer server(80);

void setup() {
  Serial.begin(9600);      // initialize serial communication
  pinMode(LEDB, OUTPUT);      // set the LED pin mode
  while(!Serial){
    ;
  }

  stepper.setMaxSpeed(2500);
  stepper.setAcceleration(2700);

  // check for the WiFi module:
  if (WiFi.status() == WL_NO_MODULE) {
    Serial.println("Communication with WiFi module failed!");
    // don't continue
    while (true);
  }

  String fv = WiFi.firmwareVersion();
  if (fv < WIFI_FIRMWARE_LATEST_VERSION) {
    Serial.println("Please upgrade the firmware");
  }

  // attempt to connect to WiFi network:
  while (status != WL_CONNECTED) {
    Serial.print("Attempting to connect to Network named: ");
    Serial.println(ssid);                   // print the network name (SSID);

    // Connect to WPA/WPA2 network. Change this line if using open or WEP network:
    status = WiFi.begin(ssid, pass);
    // wait 10 seconds for connection:
    delay(10000);
  }
  server.begin();                           // start the web server on port 80
  byte bssid[6];
  WiFi.BSSID(bssid); ////WHEN THIS LINE IS REMOVED .locaIP() causes a system freeze/crash
  printWifiStatus();                        // you're connected now, so print out the status
}

void loop() {
  WiFiClient client = server.available();   // listen for incoming clients

  if (client) {                             // if you get a client,
    Serial.println("new client");           // print a message out the serial port
    String currentLine = "";                // make a String to hold incoming data from the client
    while (client.connected()) {            // loop while the client's connected
      if (client.available()) {             // if there's bytes to read from the client,
        char c = client.read();             // read a byte, then
        Serial.write(c);                    // print it out to the serial monitor
        if (c == '\n') {                    // if the byte is a newline character

          // if the current line is blank, you got two newline characters in a row.
          // that's the end of the client HTTP request, so send a response:
          if (currentLine.length() == 0) {
            // HTTP headers always start with a response code (e.g. HTTP/1.1 200 OK)
            // and a content-type so the client knows what's coming, then a blank line:
            client.println("HTTP/1.1 200 OK");
            client.println("Content-type:text/html");
            client.println();

            // the content of the HTTP response follows the header:
            client.print("Click <a href=\"/H\">here</a> turn the LED on pin 9 on<br>");
            client.print("Click <a href=\"/L\">here</a> turn the LED on pin 9 off<br>");
            client.print("Click <a href=\"/S\">here</a> move stepper 500 steps<br>");

            // The HTTP response ends with another blank line:
            client.println();
            // break out of the while loop:
            break;
          } else {    // if you got a newline, then clear currentLine:
            currentLine = "";
          }
        } else if (c != '\r') {  // if you got anything else but a carriage return character,
          currentLine += c;      // add it to the end of the currentLine
        }

        // Check to see if the client request was "GET /H" or "GET /L":
        if (currentLine.endsWith("GET /H")) {
          digitalWrite(LEDB, HIGH);               // GET /H turns the LED on
        }
        if (currentLine.endsWith("GET /L")) {
          digitalWrite(LEDB, LOW);                // GET /L turns the LED off
        }
        if (currentLine.endsWith("GET /S")) {
          Serial.println(stepper.currentPosition());
          stepper.move(500);//WHEN THIS LINE IS CALLED THE WEBSERVER/ARDUINO STOPS FUNCTIONING
          Serial.println("Moving stepper 500");
        }
      }
    }
    // close the connection:
    client.stop();
    Serial.println("client disconnected");
  }

  stepper.run();
}

void printWifiStatus() {
  // print the SSID of the network you're attached to:
  Serial.print("SSID: ");
  Serial.println(WiFi.SSID());

  // print your board's IP address:
  IPAddress ip = WiFi.localIP();
  Serial.print("IP Address: ");
  Serial.println(ip);

  // print the received signal strength:
  long rssi = WiFi.RSSI();
  Serial.print("signal strength (RSSI):");
  Serial.print(rssi);
  Serial.println(" dBm");
  // print where to go in a browser:
  Serial.print("To see this page in action, open a browser to http://");
  Serial.println(ip);
}
```

LinusHeu, 2024-01-20:
Someone on the official Arduino discord server said:
>  I tried wifinina generic's scan network example and it compiles and works flawlessly... quite quickly in fact
> Other examples also work pretty well, even the RGB LED

WiFiNINA_Generic: https://github.com/khoih-prog/WiFiNINA_Generic/

I don't know more.

riywo, 2024-07-03:
Same here. `WiFi.BSSID();` saved me. Thanks!!