earlephilhower/arduino-pico #1782: Switching the Pi Pico W LED on / off [digitalWrite (LED_BUILTIN, HIGH or LOW)] seems very slow

XZCE, 2023-10-22:
Hi Earle,

I know the LED on the Pico W is hanging off the WiFi chip, so it takes longer to control than a normal IO pin normally would, but I think I'm seeing access times of around 1ms (I've not timed it precisely) which is really quite a long time for a Pico.

For example, I use the LED to indicate disk access in my project, and one way to access my project's emulated disks is via sector I/O (copying a disk therefore takes 1000's of accesses both reading and writing)...  One example takes just under 30 seconds to complete the full copy, but if I activate and deactivate the LED for each sector access, it takes just under 33 seconds.  A 10% overhead, which I can't optimise due to the way the host computer is sending commands for disk access.

Is there something that can be done to make it faster, other than moving onto a normal IO pin (for which, sadly, I have none spare)?

Thanks.

earlephilhower, 2023-10-22:
This core just calls the SDK routine which does it's magic to send a request to the cyw chip to toggle the LED.  You could ask in the SDK repo I suppose, but my guess is there cyw is not set to minimize latency for that use case.

closed 2023-10-22 by earlephilhower

XZCE, 2023-10-23:
I've now looked, and it's pretty horrible,

cyw43_do_ioctl is the function that ends up being used for IO pins with a wait loop.  It waits up to 1 second, but with a loop delay of 1ms.  So first time around, it's not done, 2nd time around it is because the delay is so massive.  I hex-edited your noipv6-nobtc-noble lib (the one I use) to make it a 100uS wait instead (THUMB code is horrible too, it was loading a register with 250 and shifting left 2 positions to make 1000, so I made the instruction load register with 25 - so the shift made it 100).

There's almost no difference in times now between using the LED and not using it (for me at least, with my 1000's of sector accesses).

Of course, I did all this with (Ghidra and) a hex-editor because I don't have any of the SDK working here, so I'm not going to report it to the SDK repo either, as I'll look pretty silly explaining all this.