earlephilhower/arduino-pico #1783: How to specify Boot Stage 2 option for PlatformIO

ayan4m1, 2023-10-22:
I read the documentation for [Boot Stage 2 options](https://arduino-pico.readthedocs.io/en/latest/ide.html#boot-stage-2-options-for-generic-rp2040) and [this comment](https://github.com/earlephilhower/arduino-pico/issues/743#issuecomment-1210905307) but I can't get it working in PlatformIO.

I following [this doc](https://arduino-pico.readthedocs.io/en/latest/platformio.html#current-state-of-development) to get a platformio.ini of:

```ini
[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower
board_build.boot2=boot2_w25q64jv_4_padded_checksum
monitor_speed = 115200
lib_deps =
  fastled/FastLED@^3.6.0
  lennarthennigs/Button2@^2.2.4
```

I have tried using `board_build.boot2` and `build.boot2` and in both cases, during build I get:

```
Building in release mode
Compiling .pio\build\pico\FrameworkArduinoBootloader\boot2_w25q080_2_padded_checksum.S.o
```

And the resulting UF2 does not boot. Any idea what I'm doing wrong?

Thank you!

ayan4m1, 2023-10-23:
I ended up defining a custom board according to [this doc](https://arduino-pico.readthedocs.io/en/latest/contrib.html#adding-a-new-board).

closed 2023-10-23 by ayan4m1

maxgerhardt, 2023-10-24:
I think you'd have to go down the same path as the real JSON does, so with

https://github.com/earlephilhower/arduino-pico/blob/a4ad8ae0fdbec37ae31348a815e1b790cce9379c/tools/json/adafruit_feather.json#L1-L5

It would be
```ini
board_build.arduino.earlephilhower.boot2_source = boot2_w25q64jv_4_padded_checksum.S
```
(untestedd)