earlephilhower/arduino-pico #1788: I2C Wire::endTransmission is hanging

ZinnerC, 2023-10-26:
I currently encounter the issue that some communication is not terminated.

The function Wire::endTransmission() keeps hanging. When I look at two signals with a logic analyser I see that the SCL line is continously low. Something is holding it down.
There could be several reasons for this such as noise since we "abuse" the I2C communication for inter MCU communication off PCB.

However, when inspecting the Wire functions and I see that there is no timeout option as there is in other I2C implementations for other boards.
An addition of the functions could solve my current symptom of having hanging "endTransmissions". The problem would not be solved but handled.

Would be much appreciated.

earlephilhower, 2023-10-27:
For the timeout, is it in the official Arduino documents or is it just a defacto standard?

I'm currently traveling, but if you have a PR impending this I would be happy to review from the hostel!

ZinnerC, 2023-10-27:
Sure. I can give it a shot.

Here you can see it from Arduino

https://www.arduino.cc/reference/en/language/functions/communication/wire/setwiretimeout/

earlephilhower, 2023-10-29:
Great, thanks for the link.

ZinnerC, 2023-11-01:
Created a PR.

https://github.com/earlephilhower/arduino-pico/pull/1793

closed 2023-11-10 by earlephilhower

thestumbler, 2025-04-15:
I just ran into a possible need for this timeout. This seems like a minor point, but I noticed that the Arduino documentation. the word `Wire` is included in these I2C timeout function names. Therefore, these functions don't 100% align from a naming perspective, although I suspect they function the same.  For example,

Arduino docs:
`  setWireTimeout(), getWireTimeoutFlag(), clearWireTimeout()`
Arduino-pico (from Wire.h): 
`  setTimeout(),     getTimeoutFlag(),     clearTimeoutFlag()`