earlephilhower/arduino-pico #1794: If already mounted (automounting removable drives), boot mode upload fails.

sleemanj, 2023-10-30:
If a (generic in my case) RP2040 board is connected in boot mode, and your OS (Ubuntu in my case) automounts the drive, then `uf2conv.py` fails because it attempts to mount the already mounted drive.

Somwhat annoyingly for an unknown reason as yet the Ubuntu Virtualbox VM I'm testing on is very persistent in automounting, as soon as you unmount, it remounts again despite settings set to otherwise.  Which makes it impossible to upload to a (fresh, without firmware loaded) RP2040.

A simple fix to `uf2conv.py` is to just see if the mount failed because it was already mounted...

    # Existing Line 277 
                      if proc_out.returncode == 0:
                          stdoutput = proc_out.stdout.decode("UTF-8")
                          match = re.search(r'Mounted\s+.*\s+at\s+([^\.\r\n]*)', stdoutput)
                          if match:
                              drives = [match.group(1)]
    # Add the following else clause
                      else:
                          stderror = proc_out.stderr.decode("UTF-8")
                          match = re.search(r'already mounted at\s+`([^\.\r\n']+)', stderr)
                          if match:
                              drives = [match.group(1)]

earlephilhower, 2023-10-30:
I'm traveling, so can't make a PR myself, but it looks like you not only had a problem but found a solution.  Would you like to make a PR with the fix?

sleemanj, 2023-10-30:
Sure have done so

closed 2023-11-05 by earlephilhower