earlephilhower/arduino-pico #1801: TinyUSB not working

traxanos, 2023-11-02:
Hi,

i have been sitting here for a few days and wanted to emulate a storage.
the low level way via `__USBInstallMassStorage()` works. but the way via Adafruit-TinyUSB does not work.

my structure looks roughly like this
```
    -D USE_TINYUSB
    -D CFG_TUD_ENABLED // optional tried
    -D CFG_TUSB_CONFIG_FILE=\"custom_tusb_config.h\"

    Adafruit_USBD_MSC usb_msc;

    usb_msc.setID("My", "External Flash", "1.0");
    usb_msc.setReadWriteCallback(msc_read_cb, msc_write_cb, msc_flush_cb);
    usb_msc.setStartStopCallback(msc_start_stop_cb);
    usb_msc.setReadyCallback(msc_ready_cb);
    usb_msc.setCapacity(EXCHANGE_FLASH_SIZE / 512, 512);

    // the beginn fails
    usb_msc.begin();
```

no matter what I do, it just does not work. can it be that a bug is responsible here? by the way, the serial doesn't work anymore, even if I set up my own via Adafruit_USBD_CDC.

maxgerhardt, 2023-11-04:
I added a whole new filesystem library (FAT16) in https://github.com/earlephilhower/arduino-pico/pull/1731 and exposed it via `Adafruit_USBD_MSC` and that worked fine. One thing I've noticed is that the `setup()` has to completely rather quickly so that USB enumeration of that device goes through (e.g., a call to erase the flash would already miss that time window).

>by the way, the serial doesn't work anymore

Then you might be dealing with a crash instead. Hard to know without the full code given. Can you upload it in full? Or use a debugger setup like [Picoprobe](https://arduino-pico.readthedocs.io/en/latest/platformio.html#debugging) to debug the project?

traxanos, 2023-11-04:
I work with a JLink, I can see when something is not working. It looks like the TinyUSB just doesn't do anything.

In the end I did the same as you:
https://github.com/earlephilhower/arduino-pico/pull/1731/files#diff-eef2061dfd7775404f5b4e2f6e731d4f3591e78a9241334b735a2cedcbfe069aR50-R58

You can view the current status here. But OpenKNX is quite extensive. Difficult for external users without prior knowledge...
https://github.com/OpenKNX/OAM-Dummy/compare/v1..usb

Incidentally, I have similar plans. The approach is different. I simulate a virtual file system and copy the files to the internal LittleFS.
I use the existing FATLib instead of ff15 and simply simulate a block device.
And since I have my own flash implementation that recognizes whether a flash erase is necessary at all, I have no problems with the 512b instead of 4096 :D

earlephilhower, 2023-11-10:
Without a complete MCVE, there's little we can do to help you here.  The example [here](https://github.com/adafruit/Adafruit_TinyUSB_Arduino/blob/b17ab37a0e5f4722bbe7996d2ce5a696b2283904/examples/MassStorage/msc_ramdisk/msc_ramdisk.ino) seems to work fine and exports a micro-sized RAM disk image over USB.

Please post a small, complete code sample showing the failure.

traxanos, 2023-11-13:
i posted an example above. and i know the examples by heart. they just don't work either.

actually it's enough that i only have 
-D USE_TINYUSB
after that nothing works anymore. here I would have expected that at least the console would continue to work.

for the time being, i have decided to continue on the low-level path. that works.

i can now transfer files directly to the internal littlefs, as well as offer files for download.

if you are interested, you can find the repo here: https://github.com/openKNX/ofm-usbexchange

earlephilhower, 2023-11-13:
OK, I'll close this for now then.  Just FYI, though, if you choose TinyUSB then your app must make the CDC USB serial port...there is no default one in that case since the whole idea is you get 100% control in that mode.

closed 2023-11-13 by earlephilhower