earlephilhower/arduino-pico #1806: Feature Request:  lwIP_esp32 with Espressif "esp_hosted" firmware for esp32

earlephilhower, 2023-11-10:
### Discussed in https://github.com/earlephilhower/arduino-pico/discussions/1795

<div type='discussions-op-text'>

<sup>Originally posted by **JAndrassy** October 29, 2023</sup>
Espressif has for esp32 a fw which makes it a MAC level WiFi adapter. The host MCU then has the TCP/IP stack and a network interface driver for the esp32. 
This could be used to allow esp32 as WiFi network interface for Pico LwIP. 
The repository 
https://github.com/espressif/esp-hosted/tree/master/esp_hosted_fg

Arduino uses this for their new Portenta C33 board which has STM32 MCU with esp32-C3 for WiFi with LwIP. 
Here is the driver
https://github.com/arduino/ArduinoCore-renesas/tree/main/libraries/ESPhost/src

There are multiple RP2040 boards with esp32 on SPI for WiFi. The Challenger RP2040 WiFi/BLE MkII board, the Arduino Nano Connect

cc: @PontusO </div>

earlephilhower, 2023-11-11:
Thanks for the info @JAndrassy!   I don't have any ESP32+RP2040 boards handy, but I think I've got some OG ESP32s floating around in the junk drawer...

Looks like that firmware also supports a standard Bluetooth HCI implementation.  So BTStack might also be persuaded to play nicely with it.

JAndrassy, 2023-11-12:
I tested it with Uno R4 with esp32 on SPI. but Uno R4 is in the same platform as the Portenta C33 for which the ESPHosted library was written. It uses HAL SPI. I did a rewrite with Arduino SPI library but for now that is slow or what. It misses the events from esp32.
but the ESPHosted library is ugly so maybe not a good starting point. maybe the host example code in the esp-hosted repository is better to start with.
I had to rebuild the firmware because the prebuild version used the other SPI (than documented). But it was easy on Linux. The ESP-IDF build tools are very good now.

JAndrassy, 2023-11-17:
here are the files for the ESPHosted library which use Arduino SPI library:
[esphostdriver.zip](https://github.com/earlephilhower/arduino-pico/files/13390287/esphostdriver.zip)

PontusO, 2023-11-20:
I can send out a couple of boards to anyone wanting to try this out @earlephilhower , @JAndrassy ?

PontusO, 2023-11-20:
I noticed the design uses an additional (data ready) pin which differs from the esp-at SPI implementation which our boards are based on. But I guess repurposing the boot select pin would be doable enabling full functionality.

JAndrassy, 2023-12-17:
another small step 
https://github.com/arduino/ArduinoCore-renesas/issues/212

JAndrassy, 2024-01-01:
I made some progress. I extracted the ESPHost library from the Renesas Core into [a separate repository](https://github.com/JAndrassy/ESPHost) and I modified it to run on any ARM Arduino. The library has **a test** in examples. The test does the WiFi related functions like connecting to AP, scan and similar. The test works with **this core**.

I integrated ESPHost into Arduino Mbed Core networking in form of [ESPHosted-EMAC library](https://github.com/JAndrassy/ESPHost-EMAC). It is just a rough draft, but it works. The ESPHostEMAC.cpp shows **how to integrate it with LwIP**.

earlephilhower, 2024-01-01:
That's awesome work!  I can tell you've been very busy over the holidays!  Will you be leaving those repos up?  I've got home renovations taking up most of my "free" time and won't be able to give it a deep dive for a couple weeks.

JAndrassy, 2024-01-01:
> Will you be leaving those repos up?

of course

maybe I even start drafting the integration here. but first I want to fix the WiFi library in the Mbed Core, which I finally can test.

JAndrassy, 2024-01-14:
this weekend I was able to make the esp-hosted firmware with the ESPHost library on my new "Arduino Nano RP2040 Connect".
it was a challenge with two major problems: 
* the fw [has a problem](https://github.com/espressif/esp-hosted/issues/315) with esp32 pins > 32 and the Nano has pin 33 as ACK.
* the Nano connects one io of RP2040 to two esp32 pins, SPI and UART. so the SPI CS pins is wired to esp32 UART TX pin too. and the esp32 with the hosted firmware has a pullup on UART TX even the firmware doesn't use UART (I guess the IDF sets the pull-up). The RP2040 was unable to pull down two pulled up io of the esp32.

JAndrassy, 2024-02-10:
@PontusO Hello. I want to build the esp-hosted firmware for the Challenger boards so you can test the lwIP_ESPHost library.
Please confirm that for both the C3 and C6 the pins are:
MISO IO2
CLK IO6
MOSI IO7
SS IO10
HS IO3
and IO9 is connected too

earlephilhower, 2024-02-10:
I'm having some issues with the esp-hosted-fg (built using their defaults) on an ESP32C6 board and the LWIP driver here, would appreciate any suggestions on how to debug.

I've double checked the wiring between the RPIPico board and the ESP32-C6 and have everything agreeing AIUI, but other than an initial SPI transaction being dumped I see no other SPI transactions while enabling full core debug.

The ESP32C6 dumps the following config info to serial:
````
I (471) NETWORK_ADAPTER: *********************************************************************
I (481) NETWORK_ADAPTER:                 ESP-Hosted-FG Firmware version :: 0.0.5                        
I (491) NETWORK_ADAPTER:                 Transport used :: SPI only                      
I (501) NETWORK_ADAPTER: *********************************************************************
I (511) NETWORK_ADAPTER: Supported features are:
I (521) NETWORK_ADAPTER: - WLAN over SPI
I (521) ESP_BT: - BT/BLE
I (521) ESP_BT:    - HCI Over SPI
I (531) ESP_BT:    - BLE only
I (531) NETWORK_ADAPTER: capabilities: 0xe8
I (541) phy_init: phy_version 202,b4b3263,May 17 2023,20:14:14
controller lib commit: [5cacafa]
I (591) NETWORK_ADAPTER: ESP Bluetooth MAC addr: 40:4c:ca:4e:de:1a
I (591) SPI_DRIVER: Using SPI interface
I (591) gpio: GPIO[3]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (601) gpio: GPIO[4]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (611) SPI_DRIVER: SPI Ctrl:1 mode: 2, InitFreq: 10MHz, ReqFreq: 26MHz
GPIOs: MOSI: 7, MISO: 2, CS: 10, CLK: 6 HS: 3 DR: 4

I (621) SPI_DRIVER: Hosted SPI queue size: Tx:20 Rx:20
I (631) gpio: GPIO[10]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (641) gpio: GPIO[10]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (651) pp: pp rom version: 5b8dcfa
I (651) net80211: net80211 rom version: 5b8dcfa
````

My boards.local.txt is:
````
rpipico.build.extra_flags=-DESPHOST_RESET=D5 -DESPHOST_HANDSHAKE=D7 -DESPHOST_DATA_READY=D6 -DESPHOST_CS=D1 -DESPHOSTSPI=SPI
````

I've wired as follows:
````
ESP7 (MOSI) <-> GP3
ESP2 (MISO) <-> GP0
ESP10 (CS)  <-> GP1
ESP6 (CLK)  <-> GP2
ESP3 (HS)   <-> GP7
ESP4 (DR)   <-> GP6
ESP-RST     <-> GP5
ESP-GND     <-> RPIPICO-GND
````

And configured SPI to use the appropriate pins at the beginning of `setup()` before any code is executed
````
  SPI.setRX(0);
  SPI.setCS(1);
  SPI.setSCK(2);
  SPI.setTX(3);
````

I've started going down the driver code in GDB but it's real slow going.  I was wondering if you have any suggestions on where to start, or if there are certain version dependencies or something higher level.  Everything is the `git head` version (your library and the ESP FW)

JAndrassy, 2024-02-11:
@earlephilhower your initial log of the fw should end with something like "leaving main()" (the handling tasks started).
you build the fw from master branch? the library only works with 0.0.5 tag version and now I see C6 is unknown there

PontusO, 2024-02-11:
> @PontusO Hello. I want to build the esp-hosted firmware for the Challenger boards so you can test the lwIP_ESPHost library. Please confirm that for both the C3 and C6 the pins are: MISO IO2 CLK IO6 MOSI IO7 SS IO10 HS IO3 and IO9 is connected too

@JAndrassy For the C3 board that is correct. The C6 board is connected a bit differently. MISO IO3, MOSI IO7, CLK IO6, SS IO12, HS IO2 and IO9 is also connected.

JAndrassy, 2024-02-13:
@PontusO 
https://github.com/JAndrassy/ESPHost/tree/main/extras/firmware/Challenger2040WiFiBLE
https://github.com/JAndrassy/ESPHost/blob/main/README.md#configuration

earlephilhower, 2024-02-24:
@JAndrassy any reason to keep this open?  You seem to have completed it. 😄

JAndrassy, 2024-02-24:
you can close it.

closed 2024-02-24 by earlephilhower

JAndrassy, 2024-03-24:
@PontusO the configuration for C3  has 30 MHz SPI so use `-DESPHOSTSPI_MHZ=30`

JAndrassy, 2024-04-26:
@PontusO sorry. the upload command in the README had a wrong address for the bootloader. on C3 the bootloader address is 0x0