earlephilhower/arduino-pico #1812: Pico W WiFi + FreeRTOS `xSemaphoreTake` in loop results in freeze

GUVWAF, 2023-11-13:
Using version 3.6.0 of this core, calling FreeRTOS' `xSemaphoreTake` repeatedly in a loop while WiFi is enabled on a Raspberry Pi Pico W eventually results in a freeze. From the [WiFi documentation](https://arduino-pico.readthedocs.io/en/latest/wifi.html) I understood that using FreeRTOS like this should be possible; if I misunderstood this, please disregard this issue.
In the application (Meshtastic), the semaphore is used to create an “interruptable delay”. In an ISR, `xSemaphoreGiveFromISR` is used to early-exit the delay. 

Here's an MCVE, adapted from the WiFiServer example: 
```
#include <WiFi.h>
#include "FreeRTOS.h"
#include <semphr.h>

SemaphoreHandle_t semaphore;

#ifndef STASSID
#define STASSID "your-ssid"
#define STAPSK "your-password"
#endif

const char* ssid = STASSID;
const char* password = STAPSK;

int port = 4403;

WiFiServer server(port);

void setup() {
  Serial.begin(115200);
  semaphore = xSemaphoreCreateBinary();
  WiFi.mode(WIFI_STA);
  WiFi.config(IPAddress(192, 168, 0, 123)); // Probably not needed, but somehow I need a static IP for my router
  delay(5000);
  Serial.printf("Connecting to '%s' with '%s'\n", ssid, password);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(100);
  }
  Serial.printf("\nConnected to WiFi\n\nConnect to server at %s:%d\n", WiFi.localIP().toString().c_str(), port);

  server.begin();
}

void loop() {
  static int i;
  delay(1000);
  Serial.printf("--loop %d\n", ++i);
  delay(10);
  WiFiClient client = server.available();
  if (!client) {
    Serial.printf("Take semaphore\n");
    xSemaphoreTake(semaphore, pdMS_TO_TICKS(10));
    return;
  }
}
```

After a couple loops, the device freezes with the following stack trace:

> #0  sleep_until (t=<optimized out>) at /home/earle/src/pico-quick-toolchain/arduino/pico-sdk/src/common/pico_time/time.c:398
#1  0x100096e4 in sleep_us (us=<optimized out>) at /home/earle/src/pico-quick-toolchain/arduino/pico-sdk/src/common/pico_time/time.c:414
#2  0x100096fe in sleep_ms (ms=<optimized out>) at /home/earle/src/pico-quick-toolchain/arduino/pico-sdk/src/common/pico_time/time.c:430
#3  0x1000572c in rtosFatalError ()
#4  0x10003b30 in xQueueSemaphoreTake ()
#5  0x10003c80 in xQueueTakeMutexRecursive ()
#6  0x100055f2 in __freertos_recursive_mutex_take ()
#7  0x10007eae in __retarget_lock_acquire_recursive ()
#8  0x10019b9c in __malloc_lock (ptr=ptr@entry=0x200143f8) at /workdir/repo/newlib/newlib/libc/stdlib/mlock.c:44
#9  0x100197d4 in _malloc_r (reent_ptr=0x200143f8, bytes=bytes@entry=12) at /workdir/repo/newlib/newlib/libc/stdlib/mallocr.c:2362
#10 0x10019752 in malloc (nbytes=12) at /workdir/repo/newlib/newlib/libc/stdlib/malloc.c:164
#11 0x100086da in __wrap_malloc ()
#12 0x1000b456 in operator new (n=<optimized out>)
    at /home/earle/src/pico-quick-toolchain/arduino/pico-sdk/src/rp2_common/pico_standard_link/new_delete.cpp:15
#13 0x10005cce in std::_Function_handler<void (), std::_Bind<signed char (LwipIntfDev<CYW43>::*(LwipIntfDev<CYW43>*))()> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) ()
#14 0x1001ff00 in std::function<void ()>::function(std::function<void ()> const&) ()
#15 0x1001ff4a in ethernet_timeout_reached(async_context*, async_work_on_timeout*) ()
#16 0x1000ebb6 in async_context_base_execute_once (self=self@entry=0x20001bf0 <cyw43_async_context_threadsafe_background>)
    at /home/earle/src/pico-quick-toolchain/arduino/pico-sdk/src/rp2_common/pico_async_context/async_context_base.c:96
#17 0x1000e79c in process_under_lock (self=self@entry=0x20001bf0 <cyw43_async_context_threadsafe_background>)
    at /home/earle/src/pico-quick-toolchain/arduino/pico-sdk/src/rp2_common/pico_async_context/async_context_threadsafe_background.c:251
#18 0x1000e8fe in low_priority_irq_handler ()
    at /home/earle/src/pico-quick-toolchain/arduino/pico-sdk/src/rp2_common/pico_async_context/async_context_threadsafe_background.c:293
#19 <signal handler called>
#20 0x10004310 in vTaskSuspendAll ()
#21 0x10004f1c in vTaskDelay ()
#22 0x1000581c in __usb(void*) ()
#23 0x10005908 in vPortStartFirstTask ()

I would like to hear some thoughts on this.

earlephilhower, 2023-11-13:
Thanks for the stack trace.  It looks like `std::function` is allocating memory during an interrupt which is going to give everyone a bad day.  Not sure the semaphore taking has anything to do with it, other than maybe affecting timing to make it happen more frequently.

Let me double-check I can reproduce this here and then I'll have to rework the Ethernet worker callback chain...

earlephilhower, 2023-11-13:
@GUVWAF can you give #1815 a try and report back?  I am now at 802 loops w/o a crash using it (but had immediate crashes before the change).

GUVWAF, 2023-11-14:
Thanks for the (quick!) fix. I indeed can run the MCVE for a long time now too. 
However, I still experience a freeze when running Meshtastic, which seems to happen after a longer time now. The stack trace doesn’t give me anything useful: 

```
#0  0x10041d46 in prvIdleTask ()
#1  0x10042d28 in vPortStartFirstTask ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

I will try to isolate the problem. I can open a new issue when I find something, or comment here again; please let me know what works best for you.

commit b8d14238 (2023-11-13): Avoid memory allocation in Ethernet callbacks

std::bind can cause a memory allocation to occur during the periodic
polling interrupt which is a very bad thing.  Use a lambda instead.

Fixes #1812
PR #1815 (2023-11-14): Avoid memory allocation in Ethernet callbacks
std::bind can cause a memory allocation to occur during the periodic polling interrupt which is a very bad thing.  Use a lambda instead.

Fixes #1812
commit 4625d2c8 (2023-11-14): Avoid memory allocation in Ethernet callbacks (#1815)

std::bind can cause a memory allocation to occur during the periodic
polling interrupt which is a very bad thing.  Use a lambda instead.

Fixes #1812
closed 2023-11-14 by earlephilhower