earlephilhower/arduino-pico #1822: Problems with MDNS

traxanos, 2023-11-17:
Hi Earle,

we habe some problems with the mDNS Service.

```
MDNS.begin(_hostName);
MDNS.enableArduino(8080, false);
MDNS.addService("http", "tcp", 80);
```

For _arduino._tcp. (enableArduino way) i got a valid responses with ip and port.
For _http._tcp. (addService way) i got invalid responses without ip (empty) and invalid port (-1).

Additional i need add MDNS.update() in my loop(). no idea if this is intentional.
Do you have an idea? We use an ethernet port with your lwip and leamdns.

earlephilhower, 2023-11-17:
> Additional i need add MDNS.update() in my loop(). no idea if this is intentional.

Yes, that's required and expected, like the `HTTPServer`/etc.  OTW apps would need to do locking/etc. to avoid potential re-entrancy.

Can you give a small MCVE that shows the failure, please?  This may need to go upstream as I just borrowed the MDNS library from the 8266 (who took it from somewhere else)  and really haven't needed to look at the internals, ever.  IT always "just worked" on the 8266 and here for ArduinoOTA.

traxanos, 2023-11-18:
here an extracted version:

```
#include "LEAmDNS.h"
#include <Arduino.h>
#include <W5500lwIP.h>
#include <lwip/dhcp.h>

#define PIN_ETH_SS (29)
#define PIN_ETH_MISO (28)
#define PIN_ETH_MOSI (27)
#define PIN_ETH_SCK (26)
#define ETH_SPI_INTERFACE SPI1
#define KNX_NETIF Eth

Wiznet5500lwIP KNX_NETIF(PIN_ETH_SS, ETH_SPI_INTERFACE);
WiFiUDP Udp;

void setup()
{
    ETH_SPI_INTERFACE.setRX(PIN_ETH_MISO);
    ETH_SPI_INTERFACE.setTX(PIN_ETH_MOSI);
    ETH_SPI_INTERFACE.setSCK(PIN_ETH_SCK);
    ETH_SPI_INTERFACE.setCS(PIN_ETH_SS);

    const char *hostname = "MyTest";
    KNX_NETIF.hostname(hostname);
    KNX_NETIF.begin();
    MDNS.begin(hostname);
    MDNS.enableArduino(80);
    MDNS.addService("http", "tcp", 123);
    MDNS.announce();
}

void loop()
{
    MDNS.update();
}
```
[image]
[image]

maxgerhardt, 2023-11-19:
Weird. I just tried this sketch on a Pico W:

```cpp
#include <Arduino.h>
#include <WiFi.h>
#include "LEAmDNS.h"

#define WIFI_SSID "YOURSSIDHERE"
#define WIFI_PW  "YOURPASSWORDHERE"

void setup()
{
    WiFi.mode(WIFI_STA);
    WiFi.begin(WIFI_SSID, WIFI_PW);
    while (WiFi.status() != WL_CONNECTED)
    {
        delay(500);
        Serial.print(".");
    }
    Serial.println("");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
    const char *hostname = "MyTest";
    MDNS.begin(hostname);
    MDNS.addService("http", "tcp", 123);
    MDNS.announce();
}

void loop()
{
    MDNS.update();
}
```
and this tool is showing the proper data

[image]

What app are you using to verify the result?

earlephilhower, 2023-11-19:
I'm also unable to reproduce any problem using my wired W5100 board from WizNet under Ubuntu and `avahi-discover`:

[image]

[image]

earlephilhower, 2023-11-19:
My code, same as yours but adjusted to the 5100 pinout:

````
#include <LEAmDNS.h>
#include <Arduino.h>
#include <W5100lwIP.h>

Wiznet5100lwIP eth(17, SPI);

void setup()
{
    SPI.setRX(16);
    SPI.setCS(17);
    SPI.setSCK(18);
    SPI.setTX(19);
    
    const char *hostname = "MyTest";
    eth.hostname(hostname);
    eth.begin();
    MDNS.begin(hostname);
    MDNS.enableArduino(80);
    MDNS.addService("http", "tcp", 123);
    MDNS.announce();
}

void loop()
{
    MDNS.update();
}
````

traxanos, 2023-11-20:
i have tested with different tools on iphone and mac. now i have also tried with zeroconfservicebrowser and there i get the entry.

the only difference between all the other devices i have and this one is that they all have TXT records. and when i add them it also works with the other tools.

so this works
MDNS.addService("http", "tcp", 80);
MDNS.addServiceTxt("http", "tcp", "info", "no");

what had confused me, I had already tried it with txt record, like this
MDNS.addService("device-info", "tcp", 0);
MDNS.addServiceTxt("device-info", "tcp", "info", "no");

but that didn't work either. the reason here is that the lib does not allow port 0!

but i can get around this by using port 1 or -1. not nice but it works.

thanks for your time.

closed 2023-11-20 by earlephilhower