earlephilhower/arduino-pico #1824: linker issue, pico w, arduino framework and platform io

sverrham, 2023-11-18:
Hello, not sure if this is the right place, so sorry if it is wrong.

I hade this working but then I removed the platform io pakages to get a clean update, (something might have gone wrong here)

But now I can't seem to link with fhe following error when using the #include "pico/cyw43_arch.h"
```
Linking .pio\build\pico\firmware.elf
c:/xxx/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/12.3.0/../../../../arm-none-eabi/bin/ld.exe: .pio\build\pico\libFrameworkArduino.a(lwip_wrap.cpp.o): in function `_ZN9LWIPMutexC2Ev':
lwip_wrap.cpp:(.text._ZN9LWIPMutexC2Ev[_ZN9LWIPMutexC5Ev]+0x28): undefined reference to `__isPicoW'
c:/xxx/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/12.3.0/../../../../arm-none-eabi/bin/ld.exe: .pio\build\pico\libFrameworkArduino.a(lwip_wrap.cpp.o): in function `_ZN9LWIPMutexD2Ev':
lwip_wrap.cpp:(.text._ZN9LWIPMutexD2Ev[_ZN9LWIPMutexD5Ev]+0x28): undefined reference to `__isPicoW'
collect2.exe: error: ld returned 1 exit status
```

Not sure if this is something related to any update or just on my side, (I expect it is on my side but I am struggling to figure it out).

PACKAGES:
 - framework-arduinopico @ 1.30600.0+sha.41b0686
 - tool-rp2040tools @ 1.0.2
 - toolchain-rp2040-earlephilhower @ 5.120300.230911 (12.3.0)

earlephilhower, 2023-11-18:
Double check you set the board to the Pico W in your ini file.  This variable is only present in the PicoW board variant.

sverrham, 2023-11-18:
Thank you for quick comment, I have this, I thought it work, did something change?

```

[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower
upload_protocol = cmsis-dap
debug_tool = cmsis-dap
debug_init_break = tbreak loop

monitor_speed = 115200
build_flags = -D ARDUINO_RASPBERRY_PI_PICO_W
```

earlephilhower, 2023-11-18:
Nope, you're trying to build for the plain pico.  The board line need to be fixed to either `picow` or `rpipicow` or appropriate (I don't use P.IO so can't really say, but this has caught others as well).

sverrham, 2023-11-18:
facepalm, how I am abel to get up in the mornings suprises me 🤣 
worked at once with rpipicow, dont know how I got it compiled before.

earlephilhower, 2023-11-19:
Hehe, no worries!  We'll chalk it up to the quantum nature of builds and leave it at that.  :laughing:

closed 2023-11-19 by earlephilhower