earlephilhower/arduino-pico #1833: wifi  + freertos freezing

k3341095, 2023-11-21:
```c++
#include <FreeRTOS.h>
// #include <task.h>
#include <SPI.h>
#include <WiFi.h>
#include <MQTT.h>

const char ssid[] = "iptimepure";
const char pass[] = "123456789a";
int i = 0;
// 정방향 역방향 지정
boolean directionReverse = false;
boolean switchL1 = false;
boolean switchL2 = false;

WiFiClient net;
MQTTClient client;

unsigned long lastMillis = 0;

void WifiConnect() {
  Serial.print("checking wifi...");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(1000);
  }

  Serial.print("\nconnecting...");
  while (!client.connect("arduino", "public", "public")) {
    Serial.print(".");
    delay(1000);
  }

  Serial.println("\nconnected!");

  client.subscribe("/hello");
  // client.unsubscribe("/hello");
}

void messageReceived(String& topic, String& payload) {
  Serial.println("incoming: " + topic + " - " + payload);

  // Note: Do not use the client in the callback to publish, subscribe or
  // unsubscribe as it may cause deadlocks when other things arrive while
  // sending and receiving acknowledgments. Instead, change a global variable,
  // or push to a queue and handle it in the loop after calling `client.loop()`.
}

void Task1(void* arg) {
  while (1) {
    Serial.println("task1");
    // vTaskDelay(10000 / portTICK_PERIOD_MS);
  }
}
void Task2(void* arg) {
  while (1) {
    Serial.println("task2");
    // vTaskDelay(10000 / portTICK_PERIOD_MS);
  }
}

void MqttTask(void* arg) {
  while (1) {
    Serial.println("test");
    // client.loop();

    // if (!client.connected()) {
    //   connect();
    // }

    // // publish a message roughly every second.
    // if (millis() - lastMillis > 1000) {
    //   lastMillis = millis();
    //   i += 1;
    //   Serial.println(String(i) + "puslish /helloserver");
    //   client.publish("/helloserver", String(i) + "world");
    // }
  }
}

void setup() {

  Serial.begin(115200);
  WiFi.begin(ssid, pass);

  client.begin("192.168.0.105", net);
  client.onMessage(messageReceived);

  WifiConnect();
  // put your setup code here, to run once:
  // xTaskCreate(Task1, "Task1", 100, NULL, 1, NULL);
  // xTaskCreate(Task2, "Task2", 100, NULL, 2, NULL);
  // xTaskCreate(MqttTask, "MqttTask", 100, NULL, 3, NULL);
}

void loop() {
  // put your main code here, to run repeatedly:
  client.loop();
}

``` 
At the top 
If you just put #include <FreeRTOS.h>, then since it was uploaded, the
after uploading. 
I can't even make a new upload... I have to press the RESET button and unplug the USB before I can make a new upload.

If I remove the #include <FreeRTOS.h>, it works fine and the 
I can get wifi fine, connect to mqtt server fine and get event fine.

Even if you put freertos in and don't create a task, just include it and it should work. 
it stops, what should I fix?

I've tried this and that but the comments are messed up.... Sorry.
I'm using version 3.6.0. / raspberry pi pico W

earlephilhower, 2023-11-21:
This really isn't a reproducible sketch due to the MQTT server requirement and the comments.

What MQTT library are you using?  There are something like 30 listed in Arduino library manager.

Some general rules for FreeRTOS and WiFi:
* Only setup/loop or a single task can ever do anything that ends up touching the WiFi stack.
* Only core 0 can use the WiFi stack in any way
* Re-entrancy for the WiFi stack (and pretty much all other Arduino libraries) is illegal and will make things go boom.

Since this is dependent on your infrastructure, the best way to get anywhere would be for you to connect up a PicoProbe (or other) debugger and get a full stack trace for core 0 and core 1.  That will better explain where things are crashing and is often enough to pinpoint things.

k3341095, 2023-11-22:
> This really isn't a reproducible sketch due to the MQTT server requirement and the comments.
>
> What MQTT library are you using? There are something like 30 listed in Arduino library manager.
>
> Some [...]

```c++
// #include <FreeRTOS.h>
// #include <task.h>
#include <SPI.h>
#include <WiFi.h>

const char ssid[] = "iptimepure";
const char pass[] = "123456789a";
int i = 0;
// 정방향 역방향 지정
boolean directionReverse = false;
boolean switchL1 = false;
boolean switchL2 = false;

WiFiClient net;

unsigned long lastMillis = 0;

void WifiConnect() {
  Serial.print("checking wifi...");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(1000);
  }

  Serial.println("\nconnected!");
}

void setup() {

  Serial.begin(115200);
  WiFi.begin(ssid, pass);

  WifiConnect();

}

void loop() {
  // put your main code here, to run repeatedly:
}

``` 

I'm not talking about MQTT. 
I'm talking about simply adding a 
freertos to the wifi functionality and your device will die.
You have to unconditionally unplug the usb, press reset and plug it back in. 
the firmware.

// #include <FreeRTOS.h>
// #include <task.h>
After commenting it out 
11:32:00.296 -> checking wifi...
11:32:00.296 -> connected!

But if I uncomment 
the device crashes.

earlephilhower, 2023-11-22:
Ah, that's a much better MCVE, thanks! 

This is a duplicate of #1812, fixed by #1815.  Use `git master` until I publish a new release to get the fix.

closed 2023-11-22 by earlephilhower