earlephilhower/arduino-pico #1837: Can't Run BT and WiFi Simultaneously on Pico W

CliveMcF, 2023-11-22:
Google tells me that I can run Bluetooth and WiFi simultaneously on the Raspberry Pi Pico W, but no luck so far! I'm trying to send the WiFi network credentials to the Pico via BT which works fine, but as soon as the Pico connects to the WiFi network it drops the BT connection.

I'm programming the Pico via the Arduino IDE and running the code below. To reproduce the error:
With the code running on the Pico, pair it to a BT Terminal app (in my case, running on an Android phone)
Send a newline from the terminal app to let the Pico it's connected
The Pico sends the SSID prompt text to the terminal
Send the correct SSID from the terminal
The Pico prompts for the password
Send the correct password to the Pico
The BT connection to the terminal drops out
The serial monitor on the Arduino IDE displays the "WiFi connected" message with the IP address

Here are the results from the possible combinations of correct/incorrect SSID and password:
SSID correct, password correct -> BT disconnects, WiFi connects
SSID wrong, password correct -> BT stays connected, WiFi doesn't connect
SSID correct, password wrong -> BT disconnects, WiFi doesn't connect
SSID wrong, password wrong -> BT stays connected, WiFi doesn't connect

Not as logical as I'd thought it would be! 
Here's the code:
`#include <SerialBT.h>
#include <WiFi.h>

char ssid[32];      // Character array to store SSID
char password[32];  // Character array to store password
bool DEBUG = true; 

WiFiMulti multi;

void setup() {
  if (DEBUG) Serial.begin(9600);
  SerialBT.begin();
  while (!SerialBT.available()) // Wait for BT Terminal to connect
    ;
  int bytesRead = SerialBT.read();
  SerialBT.println("Please enter the WiFi SSID: ");
  while (!SerialBT.available())
    ;
  bytesRead = SerialBT.readBytesUntil('\n', ssid, sizeof(ssid));
  ssid[bytesRead] = '\0';  
  if (DEBUG) Serial.println(ssid);

  SerialBT.println("Please enter the WiFi Password: ");
  while (!SerialBT.available())
    ;
  bytesRead = SerialBT.readBytesUntil('\n', password, sizeof(password));
  password[bytesRead] = '\0'; 
  if (DEBUG) Serial.println(password);

  multi.addAP(ssid, password); // Connect to WiFi network

  if (DEBUG) {
    if (multi.run() != WL_CONNECTED) Serial.println("Unable to connect to network");
    else {
      Serial.print("WiFi connected via IP address: ");
      Serial.println(WiFi.localIP());
    }
  }
}

void loop(){

}`

earlephilhower, 2023-11-22:
`WiFi.begin` and the equivalent BT/BLE calls all basically restart the CYW43 chip to ensure it's in a safe state to start their respective operations.  Kicking the CYW43 will, of course, kill whatever was running before.  When `WiFi.end` is called behind the scenes the same effect happens.

While I'm sure it's possible to rearchitect the WiFi and BT stacks to share state and let it happen, it's definitely a very low priority for me.

If you'or someone else is very interested, though, I'd be happy to get a PR or two implementing it, though!

CliveMcF, 2023-11-22:
Thanks Earle, I think reworking the stacks would be well beyond my meagre capabilities but hopefully some generous soul with more expertise will take it on. I imagine there would be many many situations where BT/BLE & WiFi would need to coexist.

djpearman, 2023-11-30:
Hi,

It appears to be possible to use Bluetooth LE (via BTStackLib.h's BTStackManager) simultaneously with WiFi (via WiFi.h). Since I just added WiFi to a somewhat larger project on the Raspberry Pi Pico W that already had BLE set up, I have not done much testing with it however. Here are some of my preliminary observations:

1. During `setup()`, `Wifi.begin` is called before setting up BTStackManager. It appears to work since the latter is started with its `.setup` and `.startAdvertising` functions, which apparently don't restart the CYW43 chip.
2. Calling 'WiFi.begin' after starting BLE caused the former to somewhat unpredictably with either a `WL_CONNECT_FAIL`' or a `WL_DISCONNECTED`, but didn't affect BLE. Not sure what is going on here, so I need to investigate further.
3. Connecting WiFi takes quite a while, extending initial setup time from about 2 to 15 seconds.
5. Calling 'WiFi.end` also kills Bluetooth LE, thus requiring the latter to be re-setup and re-connected.

I used the [Arduino WiFi guide for setting up a web server](https://docs.arduino.cc/library-examples/wifi-library/WiFiWebServer), though I have yet to proceed past pinging the board once it is connected to WiFi. This is my very first foray into using WiFi, so I'm still finding my way around the library. Perhaps this information will help you. I'll update once I know more.

CliveMcF, 2023-12-08:
I took another look at the sequence in which BT and WiFi are started up and can happily confirm @djpearman 's observation that if WiFi successfully connects then BT doesn't drop out. So now my setup()  process is:
Get WiFi credentials from a saved file
Attempt to connect to WiFi using these credentials
Start up BT

The problem comes if WiFi doesn't connect, in which case I need to prompt for WiFi credentials via BT, save the new credentials, connect to WiFi, then restart BT as it will have been kicked out by the WiFi connection. Clunky but doable at a pinch, and certainly an improvement on where I was. Would still love to have BT/BLE and WiFi just playing nice all the time, for a whole load of reasons!

jhmaloney, 2024-12-03:
Just want to +1 this request.

Simultaneous use of BT and WiFi worked well for me back around the 3.7.0. Contrary to what others report, I was actually able to do a WiFi.begin() without losing the current BT connection.

However, that no longer works in the latest release. I'm guessing things stopped working in the transition to 4.0. 

I use PlatformIO. What's the recommended way to install earlier releases of the framework in PlatformIO? I'd like to use the latest release to build for the RP2530-based Pico 2 W while using the older release to build for the classic Pico W. Of course, it would be even better if the latest release supported smooth WiFi/Bluetooth co-existence, but I can wait for that as long as I can still build code that supports that on the classic Pico W.

jhmaloney, 2024-12-04:
Thanks for the pointer to lwip_cyw43 ::end.

What's odd is that in the 3.x version it worked to start Bluetooth first, then do a WiFi.begin() to connect to a network. Earlier posts in this issue suggest that that would not work because WiFi.begin() restarted the CYW43 chip, but I think that issue must have gotten fixed in the cyw43 driver. Unfortunately, that no longer works in the latest release.

Is there a way to specify a specific arduino-pico release when using PlatformIO? I tried using something like:

  platform = https://github.com/maxgerhardt/platform-raspberrypi.git@3.7.0

but ran into a error with one of the tools. (Can't remember the exact error.) I realize that PlatformIO support is done by Max Gerhardt, not you; just thought you might know.

djpearman, 2024-12-05:
> Is there a way to specify a specific arduino-pico release when using PlatformIO? I tried using something like:

You can use `platform_packages`, i.e. for release 3.9.5:

`platform_packages = framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git#3.9.5`

jhmaloney, 2024-12-13:
Thanks! That worked, but now I'm getting this mysterious linker error. Any ideas?

Linking .pio/build/pico-w-ble/firmware.elf
/Users/johnmaloney/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: 
/workdir/repo/newlib/newlib/libc/stdio/findfp.c:93:(.text+0xda): undefined reference to `__retarget_lock_init_recursive'
/Users/johnmaloney/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: 
/Users/johnmaloney/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: 
/Users/johnmaloney/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/14.2.0/../../../../arm-none-eabi/bin/ld: 
/workdir/repo/newlib/newlib/libc/stdio/findfp.c:201:(.text+0x1ea): undefined reference to `__retarget_lock_init_recursive'

InfiniteCoder01, 2025-02-28:
I tried with WiFi connected and then A2DP and it seems to work fine (WiFi is still connected, haven't tested if it works tho)

closed 2025-09-30 by earlephilhower

XZCE, 2025-11-18:
https://github.com/raspberrypi/pico-sdk/blob/8fcd44a1718337861214ba5499a8faceea2bfa1d/src/rp2_common/pico_cyw43_driver/cybt_shared_bus/cybt_shared_bus_driver.c#L16

I'm questioning if this issue should have been closed.  It's great that connections can now be completed in any order, but the title of this issue is about using both devices at the same time - but I don't think it's stable, because of the link above.  I've settled with the work-around that connects WiFi first, and much later scanning for BT devices once connected.

Has this been tested for stable concurrent access to WiFi and BT?  Should the Pico SDK guys be told to remove that comment in the SDK?  I've got consumers asking me why WiFi is flakey [granted, on an older firmware] if they have BT devices configured (which starts the BT firmware later as described in the workaround).

I don't mind it being an issue [only] in the SDK, but if this issue is closed, I'd hope it works here if I go to the trouble of bringing this change into my firmware.  I don't think it will change anything though???

XZCE, 2025-11-30:
Yes, my understanding is it's logging to catch the error condition..  If it was working now, I don't see why they shouldn't remove it (and the comment).  Anyway, I've found some time to work on a MCVE, it requires a Windows build (linked to WSOCK32.lib) for something to talk to the PicoW and 30 seconds later starts up a bit of the BT libs - here's what I get:

[image]

[TestWIFI_PC.c](https://github.com/user-attachments/files/23837536/TestWIFI_PC.c)

[TestWIFI.txt](https://github.com/user-attachments/files/23837537/TestWIFI.txt)

The .txt file is the .ino but renamed, since github doesn't allow .ino files to be attached.

XZCE, 2025-11-30:
It might not be obvious, but try_connect() was written to connect more than 1 device (my real firmware, which this is based on, supports 6 devices and has a 1000ms delay between starting each one).  Actually, there is no issue with BT devices in my firmware, that part does work.  I've cut it down to just 1 device here, and it just starts the connect process (doesn't matter that you don't have the device with the MAC address).  Once started, all that BT code sleeps.

The thing is, my test was with the current Arduino Pico release, the WiFi was not having any issues passing data between the PC and Pico until 30 seconds later when the BT was started up - and then it suddenly has issues and stops being reliable.

Locking/unlocking BT shouldn't matter to what I'm having issues with, which is WiFi stalling.  They are meant to be separate systems, but I suspect they are really not...  And is why I pointed out the comment in the Pico SDK - that there are issues using both at the same time.

earlephilhower, 2025-11-30:
You need to do the locking because the Wi-Fi and the Bluetooth share the cyw43 driver.  The Pico doesn't use the UART port on the CYW chip for HCI, it's not wired up.  It uses the same SPI as the Wi-Fi comms.

The lockBluetooth ensures no background WIFI or BT (or interrupts from the same) run while you're calling BT code.  Even if you don't corrupt Bluetooth state, you could corrupt the cyw43 driver or SPI interface without it.

XZCE, 2025-11-30:
Hmmm, I had assumed the single pass-through before the "sleep" wouldn't make a difference...  It's pretty reliable as-is to show an issue.

If you can show me how/where to make the change, so both devices work, that would be great.

XZCE, 2025-12-01:
Reading the docs, it says locks should be around all BT API calls - so that's what I've done - please let me know if I've got it wrong.

It doesn't change anything though, I get the exact same output I've shown above.

[TestWIFI.txt](https://github.com/user-attachments/files/23846152/TestWIFI.txt)

XZCE, 2025-12-02:
I added this in loop(), but it never detects a disconnect.

```
    if (!WiFi.connected()) {
        WiFi.begin (SSID, PASS);
        while (!WiFi.connected()) {
        }
        SERIALOUT.printf ("Reconnected\n");
    }
```

[TestWIFI.txt](https://github.com/user-attachments/files/23874532/TestWIFI.txt)

earlephilhower, 2025-12-04:
So, that did it and it's running fine for me now, but does require a TCP reconnection from the host.  It could be an issue with the WiFiServer or the way it's being handled in the sketch, but it's transient and non-fatal.

Here's my slightly augmented test with some state dumping (no changes to the host test other than IP):
````
#include <WiFi.h>

#include <btstack_tlv.h>
#include "btstack_config.h"
#include "btstack.h"
#include "classic/sdp_server.h"

#define SSID      "xxx"
#define PASS      "xxx"
#define SZ        256
#define SERIALOUT Serial
#define FIXEDBAUD 115200
#define S_TX      8
#define S_RX      9

static WiFiServer wfs(1000);
static WiFiClient wfc;
static int sz = 0;
static unsigned char buffer [SZ];
static unsigned char buff2 [SZ];

#define DEVICES 1

static char btjoy1 [18] = "58:10:31:04:4D:7B";

static char * devices [DEVICES] = { &(btjoy1 [0]) };
static bd_addr_t remote_addr [DEVICES];
static uint16_t hid_host_cid [DEVICES] = { 0 };

#define MAX_ATTRIBUTE_VALUE_SIZE (512 * DEVICES)
static uint8_t hid_descriptor_storage [MAX_ATTRIBUTE_VALUE_SIZE];

static int try_next = 1;
static void packet_handler (uint8_t packet_type, uint16_t channel, uint8_t *packet, uint16_t size) {Serial.println("HID-ph");}

void try_connect () { // To be called once every 1 ms
    static int device = DEVICES - 1;
    uint8_t status;

    if (device < 0)
        return;

    if (try_next > 1) {
        try_next--;
        return;
    }
    while ((try_next) && (device >= 0)) {
        if ((devices [device][0]) && (hid_host_cid [device] == 0)) {
            __lockBluetooth ();
            uint32_t s = millis();
            status = hid_host_connect (remote_addr [device], HID_PROTOCOL_MODE_REPORT, &(hid_host_cid [device]));
            uint32_t m = millis();
            Serial.printf("hhc took %d\n",m - s);
            __unlockBluetooth ();
            if (status != ERROR_CODE_SUCCESS)
                SERIALOUT.printf ("hid_host_connect command %d failed: 0x%02x\n", device, status);
            else {
                try_next = 0;
                SERIALOUT.printf ("hid_host_connect command %d success\n", device);
                device--;
                break;
            }
        }
        device--;
    }
}
[... 156 lines cut ...]
13:35:03.473 -> hhc took 1
13:35:03.473 -> hid_host_connect command 0 success
13:35:04.468 -> 8887/s
13:35:05.463 -> 0/s

<ctrl-c old pinger, start new one on host>
....
13:35:18.489 -> 0/s
13:35:18.842 -> HID-ph
13:35:19.484 -> 0/s
....
13:35:25.482 -> 0/s
13:35:25.771 -> new conn TCP
13:35:25.771 -> Error detected 00 != 2C
13:35:26.477 -> 50662/s
13:35:27.472 -> 73073/s
13:35:28.472 -> 73036/s
13:35:29.470 -> 72702/s
13:35:30.466 -> 73337/s
13:35:31.467 -> 73458/s
````

So it looks like the BT scanning may override WiFi connectivity during the time of scans.  At that point the TCP timeout hits and the old connection gets dropped allowing a new one in (the TCP backlog in LWIP is not so great).

So you could bring up something @ the RPI repo about this.  Normal BT slave mode seems fine, pairing as a slave seems fine, but (at least) master scanning seems to pause the WiFi side of the CYW43.  Whether this is in the cyw43-driver (IMHO unlikely) or in the BYW43 binary firmware blob, that's the question...

PR #3160 (2025-09-30): Rewrite CYW43 WiFi connection scheme
Fixes #3157
Fixes #1837

The SDK's CYW43 WiFi connection scheme combined both link layer (WiFi net connection) and IP (getting an IP address from DHCP) into a single step. We worked around it with a hack wrapped function, but it resulted in the WiFi.begin() timeout being 2x as long as specified.

Remove the SDK's cyw43_arch_wifi_connect_timeout_ms and replace with our own custom CYW43::begin which will only look at the link layer. Note that the SDK docs are hokey and the cyw43_link_status is set to 1 *while network connection attempts are underway.*  This m [... cut]
commit 8208b1fe (2025-09-30): Rewrite CYW43 WiFi connection scheme (#3160)

Fixes #3157
Fixes #1837

The SDK's CYW43 WiFi connection scheme combined both link layer (WiFi net
connection) and IP (getting an IP address from DHCP) into a single step.
We worked around it with a hack wrapped function, but it resulted in
the WiFi.begin() timeout being 2x as long as specified.

Remove the SDK's cyw43_arch_wifi_connect_timeout_ms and replace with
our own custom CYW43::begin which will only look at the link layer.
Note that the SDK docs are hokey and the cyw43_link_status is set to 1
*while network connection attempts are underway. [... cut]