earlephilhower/arduino-pico #1840: analogWrite() sets two pins

elral, 2023-11-23:
Within our project we have a flexible way to define different devices (ouputs, inputs, 7Segment displays,...), configured by a program running on a PC. This program transfers a configuration via serial communication and it gets stored in the EEPROM.
Afterwards this configuration gets read from the EEPROM and the devices are set up.

One of this device is a simple output where e.g. LED's canbe connected. Now one of our user has reported, that setting the value of one output will also set a second one. We have figured out that:
* both pins must be outputs
* the difference between the pins is 16, e.g. Pin 2 and 18 or Pin 3 and 19

So doing an analogWrite(2, 255) will also set pin 18. But it does only happen in our the way we have setup our code. Initializing and setting the output is done in a class with an overlaying function. If we move initializing of the pin to the overlaying function everything works fine, For now we have no idea why this does happen. On the AVR's we haven't seen this behaviour.

Does anyone have a hint?

```
#include <Arduino.h>

/* ********************************************************* */
class MFOutput
{
public:
    MFOutput();
    MFOutput(uint8_t pin);
    void set(uint8_t value);

private:
    uint8_t _pin;
    uint8_t _value;
};

MFOutput::MFOutput() {
    _value = 1;
}

MFOutput::MFOutput(uint8_t pin) {
    _pin   = pin;
    pinMode(_pin, OUTPUT_12MA);           // <-------------
    set(_value);
}

void MFOutput::set(uint8_t value)
{
    _value = value;
    analogWrite(_pin, _value);
}
/* ********************************************************* */

namespace Output
{
    #define MAX_OUTPUT_ARRAY    5

    MFOutput  outputs[MAX_OUTPUT_ARRAY];
    uint8_t   outputsRegistered = 0;

    void Add(uint8_t pin)
    {
        if (outputsRegistered == MAX_OUTPUT_ARRAY)
            return;
        outputs[outputsRegistered] = MFOutput(pin);
//pinMode(pin, OUTPUT_12MA);          // <--- uncomment this line to get it working, no need to delete it in MFOutput::MFOutput (unles it's better to do it)
        outputsRegistered++;
    }

    void OnSet(uint8_t pin, uint8_t state)
    {
        outputs[pin].set(state);
    }
}
/* ********************************************************* */

void setup() {
  Serial.begin(115200);
  Output::Add(2);           // add Pin 2
  Output::Add(3);           // add Pin 3
  Output::Add(18);          // add Pin 18
  Output::Add(19);          // add Pin 19
}

void loop() {
  Serial.println("Setting output to 255");
  Output::OnSet(0,255);     // 0.th registered Output, it's Pin 2
  Output::OnSet(1,255);     // 1.st registered Output, it's Pin 3
  delay(1000);
  Serial.println("Setting output to 0");
  Output::OnSet(0,0);
  Output::OnSet(1,0);
  delay(1000);
}

```

maxgerhardt, 2023-11-23:
```cpp
MFOutput::MFOutput(uint8_t pin) {
    _pin   = pin;
    pinMode(_pin, OUTPUT_12MA);           // <-------------
    set(_value);
}

void MFOutput::set(uint8_t value)
{
    _value = value;
    analogWrite(_pin, _value);
}
```
Calling into Arduino API in the constructor? How do you know that the ctors for your object are running after all the ctors that theh Arduino core needs? Russian roulette, reminds me of https://github.com/earlephilhower/arduino-pico/issues/1085.

maxgerhardt, 2023-11-23:
Besides the point above, the
>the difference between the pins is 16

is also related to hardware: PWM can only be initialized in *slices*, which are two GPIO pins, that wrap at GPIO16. See [datasheet](https://files.seeedstudio.com/wiki/XIAO-RP2040/res/rp2040_datasheet.pdf) page 546.

[image]

Not sure if Arduino-Pico has a bug there that it also initializes `pin + 16` for PWM or if that's just the case when calling into APIs from a constructor when all objects are not yet fully initialized.

elral, 2023-11-23:
@maxgerhardt thank you very much for your reply.

> Calling into Arduino API in the constructor?

Up to now I thought that this must not be done if the class is statically initialized. As we do the initialization dynamically on runtime after entering main(), I thought that the Arduinio cores are initialized and it can be done.

elral, 2023-11-23:
And moving `pinMode()` from the constructor to an separate function does not change the situation.
But it still might be my not sufficient programing knowledge...

```
#include <Arduino.h>

/* ********************************************************* */
class MFOutput
{
public:
    MFOutput();
    void attach(uint8_t pin);
    void set(uint8_t value);

private:
    uint8_t _pin;
    uint8_t _value;
};

MFOutput::MFOutput() {
    _value = 1;
}

void MFOutput::attach(uint8_t pin) {
    _pin   = pin;
    pinMode(_pin, OUTPUT_12MA);
    set(_value);
}

void MFOutput::set(uint8_t value)
{
    _value = value;
    analogWrite(_pin, _value);
}
/* ********************************************************* */

namespace Output
{
    #define MAX_OUTPUT_ARRAY    5

    MFOutput  outputs[MAX_OUTPUT_ARRAY];
    uint8_t   outputsRegistered = 0;

    void Add(uint8_t pin)
    {
        if (outputsRegistered == MAX_OUTPUT_ARRAY)
            return;
        outputs[outputsRegistered] = MFOutput();
        outputs[outputsRegistered].attach(pin);
//pinMode(pin, OUTPUT_12MA);          // <--- uncomment this line to get it working, even if not deleted it in MFOutput::MFOutput (unles it's better to do it)
        outputsRegistered++;
    }

    void OnSet(uint8_t pin, uint8_t state)
    {
        outputs[pin].set(state);
    }
}
/* ********************************************************* */

void setup() {
  Serial.begin(115200);
  Output::Add(2);           // add Pin 2
  Output::Add(3);           // add Pin 3
  Output::Add(18);          // add Pin 18
  Output::Add(19);          // add Pin 19
}

void loop() {
  Serial.println("Setting output to 255");
  Output::OnSet(0,255);     // 0.th registered Output, it's Pin 2
  Output::OnSet(1,255);     // 1.st registered Output, it's Pin 3
  delay(1000);
  Serial.println("Setting output to 0");
  Output::OnSet(0,0);
  Output::OnSet(1,0);
  delay(1000);
}
```

maxgerhardt, 2023-11-23:
```cpp
namespace Output
{
    #define MAX_OUTPUT_ARRAY    5

    MFOutput  outputs[MAX_OUTPUT_ARRAY];
```
But.. this is like a global variable with an array of `MFOutput` objects, so I also assume they'd be initialized inside the global constructors? Can you try and change that to an array of pointers to `MFOutput` objects, which will be all `nullptr` in the beginning, then using `new MFOutput();` to assign a new instance of the object?

elral, 2023-11-23:
That's where I am coming from. Our code does it in this way you said and I just "reduced" it this way.
But I will test it again.

elral, 2023-11-23:
Still the same behaviour even with change;
```
MFOutput  *outputs[MAX_OUTPUT_ARRAY];
    uint8_t   outputsRegistered = 0;

    void Add(uint8_t pin)
    {
        if (outputsRegistered == MAX_OUTPUT_ARRAY)
            return;
        outputs[outputsRegistered] = new MFOutput();
        outputs[outputsRegistered]->attach(pin);
// pinMode(pin, OUTPUT_12MA);          // <--- uncomment this line to get it working, no need to delete it in MFOutput::MFOutput (unles it's better to do it)
        outputsRegistered++;
    }
```

maxgerhardt, 2023-11-23:
Is this reproducible with just a plain sequence of `pinMode()` and `analogWrite()` calls in `setup()`, not using any objects?

elral, 2023-11-23:
> Is this reproducible with just a plain sequence of `pinMode()` and `analogWrite()` calls in `setup()`, not using any objects or classes?

No, it is not. In this case it's working as expected. It is also working if pinMode() is called from Output::Add() (see the comment in the code).

earlephilhower, 2023-11-24:
@elral can you please repost your latest complete MCVE after the changes?

@maxgerhardt 's comments about global object constructors not calling Arduino APIs is very valid and has caught many folks by surprise (i.e. there is no guarantee that then your object's constructor has been called that the other necessary objects constructors have been called before).  

FWIW, there's only one call to [set the PWM output to a GPIO](https://github.com/earlephilhower/arduino-pico/blob/cebdb6c917016252c94228728e465739d7051155/cores/rp2040/wiring_analog.cpp#L121) pin in the core:
````
earle@amd:~/Arduino/hardware/pico/rp2040$ grep -rn GPIO_FUNC_PWM cores/
cores/rp2040/wiring_analog.cpp:121:    gpio_set_function(pin, GPIO_FUNC_PWM);
````
and looking at it, unless there is a bug in the Pico SDK `gpio_set_function` call I'm hard pressed to see any possibility of a double-PWM set.

elral, 2023-11-25:
@earlephilhower It's still [this](https://github.com/earlephilhower/arduino-pico/issues/1840#issuecomment-1824445511) code. And [these](https://github.com/earlephilhower/arduino-pico/issues/1840#issuecomment-1824546353) changes do also not change the situation.

earlephilhower, 2023-11-25:
Ah, I can reproduce your results and it's just a simple misunderstanding:  the app is doing something that's just not possible due to RP2040 HW limitations.

You can `analogWrite(n)` or `analogWrite(n+16)`, but not both.  Pin N and pin N+16 are hardwired to the *same PWM generator* (see https://github.com/earlephilhower/arduino-pico/issues/1840#issuecomment-1824178021).

So if you `analogWrite(2, 0)` and `analogWrite(18, 255)`  then the last PWM setting will go out on *both* pins.  Setting `pinMode` after the fact would end up resetting the function mux to one of them to send out the GPIO, not PWM signal on that pin.

For more info see [section 4.5](https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf) in the RP2040 datasheet.

closed 2023-11-25 by earlephilhower

elral, 2023-11-28:
@earlephilhower thank you very much for your answer and my late replay.

Took me the weekend to think about what happened and realized that I always only tested setting pin 2 and checking pin 16 and mainly that it depends on the order we are intialazing and setting the pins (what you have explained).

Again thank you.