earlephilhower/arduino-pico #1841: Pimorini Tiny2040 SPI errors...

richm-dev, 2023-11-23:
When compiling code without any SPI bus includes, errors are kicked up as follows:

/Users/r/Library/Arduino15/packages/rp2040/hardware/rp2040/3.6.1/libraries/SPI/src/SPI.cpp:367:36: error: 'PIN_SPI1_MISO' was not declared in this scope; did you mean 'PIN_SPI0_MISO'?
  367 | SPIClassRP2040 SPI1(__SPI1_DEVICE, PIN_SPI1_MISO, PIN_SPI1_SS, PIN_SPI1_SCK, PIN_SPI1_MOSI);
      |                                    ^~~~~~~~~~~~~
      |                                    PIN_SPI0_MISO
/Users/r/Library/Arduino15/packages/rp2040/hardware/rp2040/3.6.1/libraries/SPI/src/SPI.cpp:367:51: error: 'PIN_SPI1_SS' was not declared in this scope; did you mean 'PIN_SPI0_SS'?
  367 | SPIClassRP2040 SPI1(__SPI1_DEVICE, PIN_SPI1_MISO, PIN_SPI1_SS, PIN_SPI1_SCK, PIN_SPI1_MOSI);
      |                                                   ^~~~~~~~~~~
      |                                                   PIN_SPI0_SS
/Users/r/Library/Arduino15/packages/rp2040/hardware/rp2040/3.6.1/libraries/SPI/src/SPI.cpp:367:64: error: 'PIN_SPI1_SCK' was not declared in this scope; did you mean 'PIN_SPI0_SCK'?
  367 | SPIClassRP2040 SPI1(__SPI1_DEVICE, PIN_SPI1_MISO, PIN_SPI1_SS, PIN_SPI1_SCK, PIN_SPI1_MOSI);
      |                                                                ^~~~~~~~~~~~
      |                                                                PIN_SPI0_SCK
/Users/r/Library/Arduino15/packages/rp2040/hardware/rp2040/3.6.1/libraries/SPI/src/SPI.cpp:367:78: error: 'PIN_SPI1_MOSI' was not declared in this scope; did you mean 'PIN_SPI0_MOSI'?
  367 | SPIClassRP2040 SPI1(__SPI1_DEVICE, PIN_SPI1_MISO, PIN_SPI1_SS, PIN_SPI1_SCK, PIN_SPI1_MOSI);
      |                                                                              ^~~~~~~~~~~~~
      |                                                                              PIN_SPI0_MOSI

exit status 1

Compilation error: exit status 1

maxgerhardt, 2023-11-24:
Interesting, so the SPI library seems to expect that __SPI1_DEVICE and PIN_SPI1_x always exists even if the board doesn't have those pinned out. PR https://github.com/earlephilhower/arduino-pico/pull/1699/files explicitly didn't add SPI1 pins because it's not physically available on the board. Maybe the SPI library needs a protecting `#ifdef PIN_SPI1_MISO` around the creation of the `SPI1` object to safeguard against it, or the board has to define these pins as "31" (=[not available](https://github.com/earlephilhower/arduino-pico/blob/39a2bbd78811be87edac56b545d8b3a3b3b5bca4/variants/pimoroni_plasma2040/pins_arduino.h#L57-L61)).

If you wanna quickfix it yourself, I guess you can go to `/Users/r/Library/Arduino15/packages/rp2040/hardware/rp2040/3.6.1/variants/pimoroni_tiny2040/pins_arduino.h` and add the 
```cpp
#define PIN_SPI1_MISO  (31u)
#define PIN_SPI1_MOSI  (31u)
#define PIN_SPI1_SCK   (31u)
#define PIN_SPI1_SS    (31u)
```
block manually.

richm-dev, 2023-11-24:
I'll try that thanks. BTW, V3.5 and later have this issue, I wasn't updating beyond V3.4.1, but got fed up with being nagged by Arduino about updating avery time I started the IDE. I don't think that the Piromoni Tiny2020 was supported in 3.4.1 and earlier, which would explain the above "bug" (or is is an issue?) above. I've tried your suggetion and it works a treat thankyou.

PR #1842 (2023-11-24): Only create SPI/Wire instances if variant defined
Fixes #1841
commit 6a878cde (2023-11-24): Only create SPI/Wire instances if variant defined (#1842)

Fixes #1841
closed 2023-11-24 by earlephilhower