earlephilhower/arduino-pico #1844: BTstackLib LE Peripheral Notify client on Characteristic Change

savejeff, 2023-11-25:
Hi, 
I'm trying to get the BTstack working to send data to a connected client (for example Android Smartphone) 
I want to get the quasi-standard UART Service working to send data bi-directional.

for this i have these two Characteristics: 
```
BTstack.addGATTService(new UUID(SERVICE_UUID_UART));
	pCharacteristic_UART_RX = BTstack.addGATTCharacteristicDynamic(new UUID(CHARACTERISTIC_UUID_UART_RX), ATT_PROPERTY_WRITE, 0); // <- client writes to this char to send data to RP2040
	pCharacteristic_UART_TX = BTstack.addGATTCharacteristicDynamic(new UUID(CHARACTERISTIC_UUID_UART_TX), ATT_PROPERTY_NOTIFY, 0); // <- RP2040 writes to this char. to trigger a notify message to the connected client to send data to remote device
```

The problem is, that i cant find functionality to write to the pCharacteristic_UART_TX (which should then notify the connected client that the characteristic has changed) 

In general, how can i write to created characteristics created with addGATTCharacteristicDynamic? 

I know the library is still under development, i looked at this around half a year ago, and there seemed to be some changes, but i still can't find this functionality. Am i overlooking something? I feel like this functionality of having a device with a readable characteristic that is updated with sensor data is pretty much the basic use case for BLE GATT.

I also looked at the underlying btstack library but its far too complicated and i would rather wait for the arduino wrapper to get the functionality i need.

earlephilhower, 2023-11-25:
The BTStackLib is directly from the BTstack folks with only compiler fixes done here to let it build without errors.

You may get a better response going to their [repo](https://github.com/bluekitchen/btstack) and opening this question...

Alternatively, you can use the raw BTstack APIs (as done in the SerialBT and HID_Bluetooth libraries).

closed 2023-11-25 by earlephilhower

savejeff, 2023-11-26:
Okay thanks. i though with it having android .ino examples, it would be specific for this Arduino Core.