earlephilhower/arduino-pico #1845: I2C connection broken with last update?

LazaroFilm, 2023-11-25:
I need to dig in further but it looks like my BNO085 isn't connecting to my RP2040 since the last update which contained some wire modifications. How can I revert to the previous version or arduino-pico on PlatformIO to troubleshoot this?

earlephilhower, 2023-11-25:
We had 4 PRs affecting I2C from 3.6.0 to 3.6.1:
* Adding capability to enable timeout for I2C (https://github.com/earlephilhower/arduino-pico/pull/1793)
* Avoid calling spi_set_format during transactions (https://github.com/earlephilhower/arduino-pico/pull/1762)
* Avoid calling spi_set_format during SPI transfers (https://github.com/earlephilhower/arduino-pico/pull/1762)
* Don't break transfer16 into 2 8-bit CS periods (https://github.com/earlephilhower/arduino-pico/pull/1764)

I believe if you're using the `git` repo as the sources in platform.ini, you can add a `@xxxx` (where xxx is the git shortrev) to pick a specific version.  @maxgerhardt might be of more help, as I don't use P.IO myself...

LazaroFilm, 2023-11-26:
Thanks! I'll check out it that works the same way with pio

LazaroFilm, 2023-11-27:
Reverting to https://github.com/earlephilhower/arduino-pico.git#3ce902183ea70a3328041923ce86c362972e8837 worked.

To do so, I [forked](https://github.com/LazaroFilm/platform-raspberrypi) @maxgerhardt 's repo and [modified](https://github.com/LazaroFilm/platform-raspberrypi/commit/eec5e45c38e22e5721bc3e0c6d9efa5e55dee5ae) the link to Arduino-Pico.

Now, I'll do more testing to see why this happens.

earlephilhower, 2023-11-29:
Any update on your debug?

LazaroFilm, 2023-11-29:
It's not Arduino-pico. I pulled an older version of my code and it's working fine. I must have broken something else somewhere the same day arduino-pico updated.

closed 2023-11-29 by LazaroFilm

earlephilhower, 2023-11-30:
Thank you for the update!  One less thing to worry about here. :)