earlephilhower/arduino-pico #1850: Multicore FIFO functions no longer recognized after update to 3.6.1

soulofancient, 2023-11-28:
Prior to the most recent update I was using multicore processing and pushing values between cores using the Arduino IDE as outlined by the documentation at: https://arduino-pico.readthedocs.io/en/latest/multicore.html#bool-rp2040-fifo-push-nb-uint32-t 

Following the update the IDE no longer recognizes the class nor functions and upon exploring the file directory I found that they no longer seem to exist. Furthermore, reverting to previous versions does not fix the problem. If there's some other protocol for multicore communication now, an explanation would be much appreciated as I have yet to find the solution.

maxgerhardt, 2023-11-28:
The `rp2040` object of type `RP2040` with a member object `fifo` still exists in https://github.com/earlephilhower/arduino-pico/blob/master/cores/rp2040/RP2040Support.h and this is standard-included in `Arduino.h`, so this should be by-default available.

Can you show the exact error message when you try
```cpp
#include <Arduino.h>

void setup() {
  rp2040.fifo.push(0xdeadbeef);
}

void loop() {}
```
?

earlephilhower, 2023-11-28:
The `rp2040` class is still in the core, as is the FIFO.  Max's code does build OK with this and prior versions.  This is not a core problem, but probably just an installation issue.

I suggest removing your current installation completely, restarting the IDE, and then reinstalling.  Be very sure to install the `Raspberry Pi Pico/RP2040` in the Boards Manager or  `board_build.core = earlephilhower` with PIO.  If you select the `Arduino MBED` version it will not have the `rp2040` object or lots of other things here (but feel free to use it if you're in the MBED ecosystem!).  See the [Platform.IO](https://github.com/earlephilhower/arduino-pico/blob/master/docs/platformio.rst) or [Arduino IDE](https://github.com/earlephilhower/arduino-pico/blob/master/docs/install.rst) docs for more info.

closed 2023-11-28 by earlephilhower