earlephilhower/arduino-pico #1852: Serial Modbus communication with version 3.6.1 not working.

codev123, 2023-11-29:
Thank you sir for this amazing project. Earlier i was communicating with energy meters modbus interface with serial1 of pico pi, using arduinoModbus library and it was working flawlessly. with 3.6.1 update my rp2040 gateway cant have faithful communication with modbus device anymore and returns error like invalid data or modbus response is not from requested address. but earlier same code with same library with version 3.6.0 was working fine.

earlephilhower, 2023-11-29:
Sorry, without an MCVE that can run w/o your special hardware there's not much anyone other than you can do here.

This could be related to #1845 but I'm not even sure what modbus is or how it's connected.

If this is hardware dependent, I think if you can catch a good waveform (say 3.6.0) and a bad waveform (3.6.1) at the same point in the app (i.e. chip init or something) we can eyeball things and see what's up.

codev123, 2023-12-01:
Thankyou sir for your response, however there was some issues with pre and post delay times for modbus re de line. By changing it, my code is working fine. Thankyou sir

closed 2023-12-01 by codev123

earlephilhower, 2023-12-01:
Thanks for the update!  Glad it's all good.