earlephilhower/arduino-pico #1865: TalkToMySelf.ino and I2C issues

victoryred, 2023-12-01:
Hi Everyone,
I'm trying to get the Wire/I2C libraries working with both the TalkToMySelf.ino and I2C_Scan.ino sketches on a Raspberry Pi Pico. Long story short, I believe the the Wire.endTransmission() function/subfunction chain has issues. I'm using version 3.6.1 from github.

Indeed, the endTransmission() function exits with a value of 4. "4: other error. Either !running || !txBegin"

I used the connection setup as instantiated here in Wire.cpp;
TwoWire Wire(__WIRE0_DEVICE, PIN_WIRE0_SDA, PIN_WIRE0_SCL);
TwoWire Wire1(__WIRE1_DEVICE, PIN_WIRE1_SDA, PIN_WIRE1_SCL);

I made no pin changes in the code. BTW what is the clock speed? Can't verify it.

Regards

earlephilhower, 2023-12-01:
What exactly is your output from TalkingToMyself.ino?  It is working just fine and unchanged from 3.6.0 to 3.6.1 here:

3.6.1:
````
Sending...
buff: 'pass 0'

recv: '0002FD'
Sending...
buff: 'pass 1'

recv: '0002FE'
Sending...
buff: 'pass 2'

recv: '0002FF'
Sending...
buff: 'pass 3'

recv: '000300'
Sending...
buff: 'pass 4'

recv: '000301'
Sending...
buff: 'pass 5'

recv: '000302'
Sending...
buff: 'pass 6'

recv: '000303'
Sending...
buff: 'pass 7'

recv: '000304'
Sending...
buff: 'pass 8'

recv: '000305'
Sending...
buff: 'pass 9'

recv: '000306'
````

3.6.0:
````
Sending...
buff: 'pass 0'

recv: '0002FD'
Sending...
buff: 'pass 1'

recv: '0002FE'
Sending...
buff: 'pass 2'

recv: '0002FF'
Sending...
buff: 'pass 3'

recv: '000300'
Sending...
buff: 'pass 4'

recv: '000301'
Sending...
buff: 'pass 5'

recv: '000302'
Sending...
buff: 'pass 6'

recv: '000303'
Sending...
buff: 'pass 7'

recv: '000304'
Sending...
buff: 'pass 8'

recv: '000305'
Sending...
buff: 'pass 9'

recv: '000306'
````

You obviously need to read the sketch and do the hookup of GPIOs, or it won't be able to talk to itself.  0->2, 1->3 (note on the RP2040 the GPIO # is not the same as the board pinout, there's a GND between GP1 and GP2).

victoryred, 2023-12-01:
Hi Earle,
Thank you for your help and advice. This is the output from my testing.

12:36:58.037 -> 
12:36:58.037 -> Wire receiving from Wire1: '' recv:  Cnt=   0
12:36:59.037 -> Beginning transmission. Error code- 0:Success
12:36:59.037 -> Finished transmission. Error code- 4: other error. Either !running || !txBegin
12:37:00.035 -> Wire1 receiving from Wire.
12:37:00.035 -> buff: ''
12:37:00.035 -> 

But this is what is peculiar, I have Wire.cpp open in VSCode and I hover over the passed initialization parameters.

TwoWire Wire(__WIRE0_DEVICE, PIN_WIRE0_SDA, PIN_WIRE0_SCL);
TwoWire Wire1(__WIRE1_DEVICE, PIN_WIRE1_SDA, PIN_WIRE1_SCL);

The devices are i2c0, i2c1
The data and clock pins are (GPIO) 4u, 5u for i2c0 and (GPIO) 26u, 27u for i2c1.
I don't mind redefining the pins in the .ino file. But what forces that to be necessary to make the change to 0,1 and 2,3.

I made the change with the same result.

victoryred, 2023-12-01:
Hi Earle,
I'm including some snippets of code.

From /home/penta/Arduino/hardware/pico/rp2040/variants/rpipico/pins_arduino.h
// Wire
#define PIN_WIRE0_SDA  (4u)
#define PIN_WIRE0_SCL  (5u)

#define PIN_WIRE1_SDA  (26u)
#define PIN_WIRE1_SCL  (27u)

From /home/penta/.arduino15/packages/rp2040/hardware/rp2040/3.6.1/libraries/Wire/src/Wire.cpp

My changes-

uint8_t TwoWire::beginTransmission(uint8_t addr) {
    if (!_running || _txBegun) {
        // ERROR
        return 4;
    }
    _addr = addr;
    _buffLen = 0;
    _buffOff = 0;
    _txBegun = true;
    return 0;
}

and the source for the error codes......

// Errors:
//  0 : Success
//  1 : Data too long
//  2 : NACK on transmit of address
//  3 : NACK on transmit of data
//  4 : Other error
//  5 : Timeout
uint8_t TwoWire::endTransmission(bool stopBit) {

closed 2023-12-01 by earlephilhower

earlephilhower, 2023-12-01:
No worries!  Good luck.