earlephilhower/arduino-pico #1868: Pico W WiFi + FreeRTOS leaves core in idle state

GUVWAF, 2023-12-01:
I’m having an issue with WiFi and FreeRTOS on the Pico W. At a certain point, both cores seem to be idling. Although I’m only using one core, in the IRQ as shown below it calls `idleOtherCore()` (since `_multicore` is true when using FreeRTOS). After `resumeOtherCore()`, both are then idling and it never returns to `loop()`. 

```
_freertos_idle_other_core@0x20000120 (~/.platformio/packages/framework-arduinopico/libraries/FreeRTOS/src/variantHooks.cpp:182)
_MFIFO::idleOtherCore@0x100159cc (~/.platformio/packages/framework-arduinopico/cores/rp2040/RP2040Support.h:97)
RP2040::idleOtherCore@0x10015ab8 (~/.platformio/packages/framework-arduinopico/cores/rp2040/RP2040Support.h:242)
__wrap_get_rand_64@0x10015ab8 (~/.platformio/packages/framework-arduinopico/cores/rp2040/_freertos.cpp:80)
__wrap_get_rand_32@0x10015ae6 (/~/.platformio/packages/framework-arduinopico/cores/rp2040/_freertos.cpp:92)
igmp_start_timer@0x10024700 (Unknown Source:695)
igmp_delaying_member@0x10024700 (Unknown Source:718)
igmp_delaying_member@0x10024700 (Unknown Source:713)
igmp_input@0x10024906 (/igmp_input.dbgasm:103)
ip4_input@0x10025672 (/ip4_input.dbgasm:305)
ethernet_input@0x10025ace (/ethernet_input.dbgasm:87)
__wrap_cyw43_cb_process_ethernet@0x10030506 (~/.platformio/packages/framework-arduinopico/cores/rp2040/cyw43_wrappers.cpp:56)
cyw43_ll_process_packets@0x10026568 (/cyw43_ll_process_packets.dbgasm:29)
cyw43_poll_func@0x1001c49c (Unknown Source:238)
cyw43_poll_func@0x1001c49c (Unknown Source:219)
cyw43_do_poll@0x1001caf6 (Unknown Source:90)
async_context_base_execute_once@0x1001d43e (/async_context_base_execute_once.dbgasm:28)
process_under_lock@0x1001d010 (Unknown Source:251)
low_priority_irq_handler@0x1001d172 (Unknown Source:293)
<signal handler called>@0xfffffffd (Unknown Source:0)
```

When I comment out `rp2040.idleOtherCore();` and `rp2040.resumeOtherCore();` in `__wrap_get_rand_64` it doesn’t happen, but obviously that is not the right solution here. 

Unfortunately I’m not able to reproduce this with a MCVE. I tried adding components from the application (Meshtastic) one by one to the WiFiServer example, but I was unable to get this to happen. Despite that, I hope this report gives some clue about what might go wrong.

earlephilhower, 2023-12-01:
Without an MCVE it's really unlikely that anyone but you can dig into this.

There are 2 cores running (FreeRTOS here is SMP), so can you grab both core stacks?  In GDB that's via `thread 0` and `thread 1` (from memory, it may be thread 1 and 2) before the `where`

earlephilhower, 2023-12-01:
(My thinking is that the other core is also trying to idle things at the exact same time and there's some very small window of possibility of things going boom).

earlephilhower, 2023-12-01:
And even more important.  FreeRTOS + WiFi has *severe* [limitations on who can call WiFi stuff](https://github.com/earlephilhower/arduino-pico/blob/master/docs/wifi.rst#important-information), of any kind (this would also include flashing the LED since the LED is driven by the WiFi chip).

a) Only 1 task or setup/loop may ever do any WiFi-related stuff of any kind
b) That task must be pinned to core 0

Otherwise, very bad things will happen in the LWIP and CYW43 interrupt stack.

GUVWAF, 2023-12-01:
Thanks for your answer. 
When this happens, core1's stack is just:
```
IdleThisCore@0x200000e0 (~/.platformio/packages/framework-arduinopico/libraries/FreeRTOS/src/variantHooks.cpp:171)
vPortStartFirstTask@0x10012408 (~/.platformio/packages/framework-arduinopico/libraries/FreeRTOS/lib/FreeRTOS-Kernel/portable/ThirdParty/GCC/RP2040/port.c:208)
```

We're only using one `loop()` and do not create any FreeRTOS task.

earlephilhower, 2023-12-01:
Are you doing any EEPROM or LittleFS or other flash writes?  And could you please double check that there really are no tasks of any kind?

The code looks like it's all well.  The 2nd core has set the "I'm idled" global flag and waiting in a busy loop w/IRQs and preemption disabled.  The 1st core is at the "I'm waiting for the 1st core to set the I'm idled flag".

Can you dump the value of `__otherCoreIdled` to confirm it's `true` during the hang

GUVWAF, 2023-12-02:
> Are you doing any EEPROM or LittleFS or other flash writes? And could you please double check that there really are no tasks of any kind?

The project uses LittleFS, but I ensured there are no writes to flash before it happens. There are two occurrences of `xTaskCreate` in the firmware, but those are not included in the compilation for the Pico W.

I stripped the firmware down as much as I can and then it is quite predictable. If I set a breakpoint in `__wrap_get_rand_64()`, after the initial connection, it enters there a couple of times, and continuous normally. About half a minute afterwards, it enters there again, but then never returns to `loop()`.

> Can you dump the value of `__otherCoreIdled` to confirm it's `true` during the hang?

Sorry, I wasn't clear in my first report. It doesn’t hang in `__wrap_get_rand_64()`, but after it returns from it. In the function, before `idleOtherCore()`, `__otherCoreIdled` is `false`, then after `idleOtherCore()`, `__otherCoreIdled` is `true`, and after `resumeOtherCore()`, it is `false` -- so that seems OK. Then if I pause immediately after that, the stack of core0 looks like: 
```
prvIdleTask@0x10013e58 (~/.platformio/packages/framework-arduinopico/libraries/FreeRTOS/lib/FreeRTOS-Kernel/tasks.c:4476)
vPortStartFirstTask@0x10012408 (~/.platformio/packages/framework-arduinopico/libraries/FreeRTOS/lib/FreeRTOS-Kernel/portable/ThirdParty/GCC/RP2040/port.c:208).
```
…and it stays there. When I comment out `rp2040.idleOtherCore()` and `rp2040.resumeOtherCore()`, it does keep coming back to `loop()`.

I also observed that when it goes wrong, before `idleOtherCore()`, `pxCurrentTCBs[0]` `xIsIdle` is 1 (`pcTaskName` is "IDLE0"), and `pxCurrentTCBs[1]` `xIsIdle` is 1 as well (`pcTaskName` is "IDLE1"). While when it continuous normally (just after initial connection), `pxCurrentTCBs[0]` `xIsIdle` is 0 and its `pcTaskName` is "CORE0". 

> Also, why are you running FreeRTOS if there are no tasks or 2nd core work?

FreeRTOS tasks were used in the past, but replaced by [ArduinoThread.](https://github.com/ivanseidel/ArduinoThread) However, the firmware still uses FreeRTOS’ `queue` and `semaphr`.

GUVWAF, 2023-12-02:
Did some further digging and it doesn’t hang anymore when I move `portENABLE_INTERRUPTS()` to the end of `__freertos_resume_other_core`.

earlephilhower, 2023-12-02:
Very nice debug.  Could you make a PR so I can see exactly the change?  Looks like a race condition where I messed up the order of operations (unfreeze not exact inverse of freeze))...

GUVWAF, 2023-12-02:
Yes, will do that in a minute. 
I believe the same should be done in `IdleThisCore`.