earlephilhower/arduino-pico #1874: SPI beginTransaction does not check if bits are set to 8

Bodmer, 2023-12-03:
If SPI beginTransaction is called for the first time, then the format is correctly set to 8 bits since this is required for the SPI library functions to work correctly.

With the latest 3.6.1 release, if the user calls endTransaction and reconfigures the SPI port for 16 bits, then a subsequent call to beginTransaction with the same settings will not result in the port being configured for 8 bits and the library SPI transfer calls then behave incorrectly.

There is no "get" call in the SDK to check the SPI bits setting and the ["settings" comparison](https://github.com/earlephilhower/arduino-pico/blob/master/libraries/SPI/src/SPI.cpp#L184) will pass if the settings are the same since the bit count is not part of these settings.

One solution is to always set the bit count to 8 in beginTransaction using:
```
hw_write_masked(&spi_get_hw(_spi)->cr0, (8 - 1) << SPI_SSPCR0_DSS_LSB, SPI_SSPCR0_DSS_BITS);
```
This line can be added [here](https://github.com/earlephilhower/arduino-pico/blob/master/libraries/SPI/src/SPI.cpp#L185). This is a register write solely for setting the SPI bits to 8, so it is faster than using:
```
spi_set_format(_spi, 8, cpol(), cpha(), SPI_MSB_FIRST);
```

Another option is to leave "as is" and rely on the user to always reset the SPI bit count to 8 before calling beginTransaction.

earlephilhower, 2023-12-03:
The only spot the bits are set is this line:
https://github.com/earlephilhower/arduino-pico/blob/0e4fd0587be389967e91e0df1c93c5aa33589495/libraries/SPI/src/SPI.cpp#L194

So I'm not sure I see where it could be anything other than 8.

Is your code natively calling the SDK and changing the SPI settings behind the scenes for something?  If that's so, then the whole "has SPISettings changed" check is kind of impossible, no?

Bodmer, 2023-12-04:
Understood, not everything is checked and the beginTransaction proceeds on the assumption that certain things have not changed to make the function run faster. This is helpful (but is a bit of a kludge!) as the SDK spi_init(_spi, _spis.getClockFreq()); function takes an incredible ~80us to execute.

Looking at the SDK spi_init() function:
```
uint spi_init(spi_inst_t *spi, uint baudrate) {
    spi_reset(spi);
    spi_unreset(spi);

    uint baud = spi_set_baudrate(spi, baudrate);
    spi_set_format(spi, 8, SPI_CPOL_0, SPI_CPHA_0, SPI_MSB_FIRST);
    // Always enable DREQ signals -- harmless if DMA is not listening
    hw_set_bits(&spi_get_hw(spi)->dmacr, SPI_SSPDMACR_TXDMAE_BITS | SPI_SSPDMACR_RXDMAE_BITS);

    // Finally enable the SPI
    hw_set_bits(&spi_get_hw(spi)->cr1, SPI_SSPCR1_SSE_BITS);

    return baud;
}
```
 and running some tests indicates that 98% of that delay is in the spi_set_baudrate(spi, baudrate); function, presumably because it iteratively determines the required divider ratios in two "for" loops.

So this means the arduino-pico code could be made more robust by avoiding that single function unless the baudrate changes.
This is untested but shows the general idea:
```
void SPIClassRP2040::beginTransaction(SPISettings settings) {
    noInterrupts(); // Avoid possible race conditions if IRQ comes in while main app is in middle of this
    DEBUGSPI("SPI::beginTransaction(clk=%lu, bo=%s)\n", settings.getClockFreq(), (settings.getBitOrder() == MSBFIRST) ? "MSB" : "LSB");

    if (!_initted) {
      spi_reset(_spi);
      spi_unreset(_spi);
    }
    // Avoid this slow function unless baud rate changes
    if (!_initted  ||  settings.getClockFreq() != _spis.getClockFreq() ) spi_set_baudrate(_spi, _spis.getClockFreq());
    _spis = settings;
    // Always play safe since user may be using 16 bit transfers
    spi_set_format(_spi, 8, cpol(), cpha(), SPI_MSB_FIRST);
    // Always enable DREQ signals -- harmless if DMA is not listening
    f (!_initted) hw_set_bits(&spi_get_hw(_spi)->dmacr, SPI_SSPDMACR_TXDMAE_BITS | SPI_SSPDMACR_RXDMAE_BITS);

    // Finally enable the SPI
    f (!_initted) hw_set_bits(&spi_get_hw(_spi)->cr1, SPI_SSPCR1_SSE_BITS);
   _initted = true;
    // Disable any IRQs that are being used for SPI
    io_irq_ctrl_hw_t *irq_ctrl_base = get_core_num() ? &iobank0_hw->proc1_irq_ctrl : &iobank0_hw->proc0_irq_ctrl;
    DEBUGSPI("SPI: IRQ masks before = %08x %08x %08x %08x\n", (unsigned)irq_ctrl_base->inte[0], (unsigned)irq_ctrl_base->inte[1], (unsigned)irq_ctrl_base->inte[2], (unsigned)irq_ctrl_base->inte[3]);
    for (auto entry : _usingIRQs) {
        int gpio = entry.first;

        // There is no gpio_get_irq, so manually twiddle the register
        io_rw_32 *en_reg = &irq_ctrl_base->inte[gpio / 8];
        uint32_t val = ((*en_reg) >> (4 * (gpio % 8))) & 0xf;
        _usingIRQs.insert_or_assign(gpio, val);
        DEBUGSPI("SPI: GPIO %d = %d\n", gpio, val);
        (*en_reg) ^= val << (4 * (gpio % 8));
    }
    DEBUGSPI("SPI: IRQ masks after = %08x %08x %08x %08x\n", (unsigned)irq_ctrl_base->inte[0], (unsigned)irq_ctrl_base->inte[1], (unsigned)irq_ctrl_base->inte[2], (unsigned)irq_ctrl_base->inte[3]);
    interrupts();
}
```

earlephilhower, 2023-12-04:
While I understand your concern, no other library has anything like that kind of paranoia.  If you're doing something to a shared resource like SPI, outside of the standard library, it seems reasonable to ask that you return it to the state it was when you're done.

So I think for the particular 8/16 bit issue titled here that #1882 will handle it and fix the actual bug in #1879.  I'm happy to look at PRs that are more paranoid but adding in paranoia is not high on the list of things to do right now for me. 😆 

One way might be to simply cache the entire register settings after an initialization and rewrite it every time even if nothing changed (modulo any register write ordering requirements).

Bodmer, 2023-12-05:
OK, that's a fair point!

PR #1882 (2023-12-05): Fix SPI 16-bit transfers
Fixes #1879
Fixes #1874
commit c3a3526a (2023-12-05): Fix SPI 16-bit transfers (#1882)

Fixes #1879
Fixes #1874
closed 2023-12-05 by Bodmer