earlephilhower/arduino-pico #1894: WiFi Server.begin Compiler Error

djaus2, 2023-12-12:
- C:\Users\david\Documents\Arduino\RPIPicoWWifi/RPIPicoWWifi.ino:22: undefined reference to `_ZN9WiFiClass5beginEPKcS1_PKh'
- Line 22 is WiFi.begin(ssid,  password);
- So ClassBegin in 2nd line above refers to WiFi.begin

> Code is below

All Messages
========
c:/users/david/appdata/local/arduino15/packages/rp2040/tools/pqt-gcc/2.1.0-a-d3d2e6b/bin/../lib/gcc/arm-none-eabi/12.3.0/../../../../arm-none-eabi/bin/ld.exe: C:\Users\david\AppData\Local\Temp\arduino\sketches\5E67FB60FE2FE9AE4FFD169CA908468C\sketch\RPIPicoWWifi.ino.cpp.o: in function `setup':
C:\Users\david\Documents\Arduino\RPIPicoWWifi/RPIPicoWWifi.ino:22: undefined reference to `_ZN9WiFiClass5beginEPKcS1_PKh'
c:/users/david/appdata/local/arduino15/packages/rp2040/tools/pqt-gcc/2.1.0-a-d3d2e6b/bin/../lib/gcc/arm-none-eabi/12.3.0/../../../../arm-none-eabi/bin/ld.exe: C:\Users\david\AppData\Local\arduino15\packages\rp2040\hardware\rp2040\3.6.2/lib/libpicow-noipv6-nobtc-noble.a(cyw43_ctrl.c.obj): in function `cyw43_ensure_up':
/home/earle/Arduino/hardware/pico/rp2040/pico-sdk/lib/cyw43-driver/src/cyw43_ctrl.c:164: undefined reference to `__wrap_cyw43_cb_tcpip_deinit'
c:/users/david/appdata/local/arduino15/packages/rp2040/tools/pqt-gcc/2.1.0-a-d3d2e6b/bin/../lib/gcc/arm-none-eabi/12.3.0/../../../../arm-none-eabi/bin/ld.exe: /home/earle/Arduino/hardware/pico/rp2040/pico-sdk/lib/cyw43-driver/src/cyw43_ctrl.c:165: undefined reference to `__wrap_cyw43_cb_tcpip_deinit'
c:/users/david/appdata/local/arduino15/packages/rp2040/tools/pqt-gcc/2.1.0-a-d3d2e6b/bin/../lib/gcc/arm-none-eabi/12.3.0/../../../../arm-none-eabi/bin/ld.exe: C:\Users\david\AppData\Local\arduino15\packages\rp2040\hardware\rp2040\3.6.2/lib/libpicow-noipv6-nobtc-noble.a(cyw43_ctrl.c.obj): in function `cyw43_deinit':
/home/earle/Arduino/hardware/pico/rp2040/pico-sdk/lib/cyw43-driver/src/cyw43_ctrl.c:130: undefined reference to `__wrap_cyw43_cb_tcpip_deinit'
c:/users/david/appdata/local/arduino15/packages/rp2040/tools/pqt-gcc/2.1.0-a-d3d2e6b/bin/../lib/gcc/arm-none-eabi/12.3.0/../../../../arm-none-eabi/bin/ld.exe: /home/earle/Arduino/hardware/pico/rp2040/pico-sdk/lib/cyw43-driver/src/cyw43_ctrl.c:131: undefined reference to `__wrap_cyw43_cb_tcpip_deinit'
c:/users/david/appdata/local/arduino15/packages/rp2040/tools/pqt-gcc/2.1.0-a-d3d2e6b/bin/../lib/gcc/arm-none-eabi/12.3.0/../../../../arm-none-eabi/bin/ld.exe: C:\Users\david\AppData\Local\arduino15\packages\rp2040\hardware\rp2040\3.6.2/lib/libpicow-noipv6-nobtc-noble.a(cyw43_ctrl.c.obj): in function `cyw43_cb_process_async_event':
/home/earle/Arduino/hardware/pico/rp2040/pico-sdk/lib/cyw43-driver/src/cyw43_ctrl.c:429: undefined reference to `__wrap_cyw43_cb_tcpip_set_link_up'
c:/users/david/appdata/local/arduino15/packages/rp2040/tools/pqt-gcc/2.1.0-a-d3d2e6b/bin/../lib/gcc/arm-none-eabi/12.3.0/../../../../arm-none-eabi/bin/ld.exe: /home/earle/Arduino/hardware/pico/rp2040/pico-sdk/lib/cyw43-driver/src/cyw43_ctrl.c:350: undefined reference to `__wrap_cyw43_cb_tcpip_set_link_down'
c:/users/david/appdata/local/arduino15/packages/rp2040/tools/pqt-gcc/2.1.0-a-d3d2e6b/bin/../lib/gcc/arm-none-eabi/12.3.0/../../../../arm-none-eabi/bin/ld.exe: /home/earle/Arduino/hardware/pico/rp2040/pico-sdk/lib/cyw43-driver/src/cyw43_ctrl.c:405: undefined reference to `__wrap_cyw43_cb_tcpip_set_link_up'
c:/users/david/appdata/local/arduino15/packages/rp2040/tools/pqt-gcc/2.1.0-a-d3d2e6b/bin/../lib/gcc/arm-none-eabi/12.3.0/../../../../arm-none-eabi/bin/ld.exe: /home/earle/Arduino/hardware/pico/rp2040/pico-sdk/lib/cyw43-driver/src/cyw43_ctrl.c:409: undefined reference to `__wrap_cyw43_cb_tcpip_set_link_down'
c:/users/david/appdata/local/arduino15/packages/rp2040/tools/pqt-gcc/2.1.0-a-d3d2e6b/bin/../lib/gcc/arm-none-eabi/12.3.0/../../../../arm-none-eabi/bin/ld.exe: C:\Users\david\AppData\Local\arduino15\packages\rp2040\hardware\rp2040\3.6.2/lib/libpicow-noipv6-nobtc-noble.a(cyw43_ll.c.obj): in function `cyw43_ll_process_packets':
/home/earle/Arduino/hardware/pico/rp2040/pico-sdk/lib/cyw43-driver/src/cyw43_ll.c:1169: undefined reference to `__wrap_cyw43_cb_process_ethernet'
c:/users/david/appdata/local/arduino15/packages/rp2040/tools/pqt-gcc/2.1.0-a-d3d2e6b/bin/../lib/gcc/arm-none-eabi/12.3.0/../../../../arm-none-eabi/bin/ld.exe: C:\Users\david\AppData\Local\arduino15\packages\rp2040\hardware\rp2040\3.6.2/lib/libpicow-noipv6-nobtc-noble.a(cyw43_ll.c.obj): in function `cyw43_do_ioctl':
/home/earle/Arduino/hardware/pico/rp2040/pico-sdk/lib/cyw43-driver/src/cyw43_ll.c:1203: undefined reference to `__wrap_cyw43_cb_process_ethernet'
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

Code
===

```
#include <WiFi.h>

#ifndef STASSID
#define STASSID "SSID"
#define STAPSK "PWD"
#endif

const char* ssid = STASSID;
const char* password = STAPSK;

int port = 4242;

WiFiServer server(port);

void setup() {
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.setHostname("PicoW2");
  Serial.printf("Connecting to '%s' with '%s'\n", ssid, password);
  WiFi.begin(ssid,  password);
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(100);
  }
  Serial.printf("\nConnected to WiFi\n\nConnect to server at %s:%d\n", WiFi.localIP().toString().c_str(), port);

  server.begin();
}

void loop() {
  // put your main code here, to run repeatedly:
  Serial.print('-'); delay(1000);
}
```

djaus2, 2023-12-12:
Using Latest BSP

earlephilhower, 2023-12-12:
Set the board to the PicoW and try again.  There's no WiFi otherwise.

closed 2023-12-12 by earlephilhower

djaus2, 2023-12-12:
Ah ... ! Silly me! Thanks for the quick response.  Maybe this might help someone in the future. :)