earlephilhower/arduino-pico #1896: Compilation error - Help

baritonomarchetto, 2023-12-12:
Hello, I have recently updated the library from an old version and I am experiencing some issue now. 
I am using Arduino IDE 2.2.1 and Pi Pico board package 3.6.2. Unfortunately I did not take notice of the version I was using before.
In particular, I am having the following error message trying to upload the Joystick-AllFunctions example sketch (verbose, sorry):
```
FQBN: rp2040:rp2040:rpipico
Using board 'rpipico' from platform in folder: C:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2
Using core 'rp2040' from platform in folder: C:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2

Detecting libraries used...
C:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\2.1.0-a-d3d2e6b/bin/arm-none-eabi-g++ -I C:\Users\HOME\AppData\Local\Temp\arduino\sketches\B5D3088AA6B6BC2E9E69B28F2F38CBB8/core -c -Werror=return-type -Wno-psabi -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSBD_PID=0x000a -DUSBD_VID=0x2e8a -DUSBD_MAX_POWER_MA=250 -DUSB_MANUFACTURER="Raspberry Pi" -DUSB_PRODUCT="Pico" -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1 -DCYW43_LWIP=1 -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 -DARDUINO_VARIANT="rpipico" -DTARGET_RP2040 -DPICO_FLASH_SIZE_BYTES=2097152 -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions -DARM_MATH_CM0_FAMILY -DARM_MATH_CM0_PLUS -iprefixC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2/ @C:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2/lib/platform_inc.txt -IC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2/include -fno-rtti -std=gnu++17 -g -pipe -w -x c++ -E -CC -DF_CPU=133000000L -DARDUINO=10607 -DARDUINO_RASPBERRY_PI_PICO -DBOARD_NAME="RASPBERRY_PI_PICO" -DARDUINO_ARCH_RP2040 -Os -DWIFICC=CYW43_COUNTRY_WORLDWIDE -IC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2\cores\rp2040 -IC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2\variants\rpipico C:\Users\HOME\AppData\Local\Temp\arduino\sketches\B5D3088AA6B6BC2E9E69B28F2F38CBB8\sketch\Joystick-AllFunctions.ino.cpp -o nul
Alternatives for Joystick.h: [Joystick@2.1.1 Joystick@1.0.1]
ResolveLibrary(Joystick.h)
  -> candidates: [Joystick@2.1.1 Joystick@1.0.1]
C:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\2.1.0-a-d3d2e6b/bin/arm-none-eabi-g++ -I C:\Users\HOME\AppData\Local\Temp\arduino\sketches\B5D3088AA6B6BC2E9E69B28F2F38CBB8/core -c -Werror=return-type -Wno-psabi -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSBD_PID=0x000a -DUSBD_VID=0x2e8a -DUSBD_MAX_POWER_MA=250 -DUSB_MANUFACTURER="Raspberry Pi" -DUSB_PRODUCT="Pico" -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1 -DCYW43_LWIP=1 -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 -DARDUINO_VARIANT="rpipico" -DTARGET_RP2040 -DPICO_FLASH_SIZE_BYTES=2097152 -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions -DARM_MATH_CM0_FAMILY -DARM_MATH_CM0_PLUS -iprefixC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2/ @C:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2/lib/platform_inc.txt -IC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2/include -fno-rtti -std=gnu++17 -g -pipe -w -x c++ -E -CC -DF_CPU=133000000L -DARDUINO=10607 -DARDUINO_RASPBERRY_PI_PICO -DBOARD_NAME="RASPBERRY_PI_PICO" -DARDUINO_ARCH_RP2040 -Os -DWIFICC=CYW43_COUNTRY_WORLDWIDE -IC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2\cores\rp2040 -IC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2\variants\rpipico -IC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2\libraries\Joystick\src C:\Users\HOME\AppData\Local\Temp\arduino\sketches\B5D3088AA6B6BC2E9E69B28F2F38CBB8\sketch\Joystick-AllFunctions.ino.cpp -o nul
Alternatives for HID_Joystick.h: [HID_Joystick@1.0.1]
ResolveLibrary(HID_Joystick.h)
  -> candidates: [HID_Joystick@1.0.1]
C:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\2.1.0-a-d3d2e6b/bin/arm-none-eabi-g++ -I C:\Users\HOME\AppData\Local\Temp\arduino\sketches\B5D3088AA6B6BC2E9E69B28F2F38CBB8/core -c -Werror=return-type -Wno-psabi -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSBD_PID=0x000a -DUSBD_VID=0x2e8a -DUSBD_MAX_POWER_MA=250 -DUSB_MANUFACTURER="Raspberry Pi" -DUSB_PRODUCT="Pico" -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1 -DCYW43_LWIP=1 -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 -DARDUINO_VARIANT="rpipico" -DTARGET_RP2040 -DPICO_FLASH_SIZE_BYTES=2097152 -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions -DARM_MATH_CM0_FAMILY -DARM_MATH_CM0_PLUS -iprefixC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2/ @C:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2/lib/platform_inc.txt -IC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2/include -fno-rtti -std=gnu++17 -g -pipe -w -x c++ -E -CC -DF_CPU=133000000L -DARDUINO=10607 -DARDUINO_RASPBERRY_PI_PICO -DBOARD_NAME="RASPBERRY_PI_PICO" -DARDUINO_ARCH_RP2040 -Os -DWIFICC=CYW43_COUNTRY_WORLDWIDE -IC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2\cores\rp2040 -IC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2\variants\rpipico -IC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2\libraries\Joystick\src -IC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2\libraries\HID_Joystick\src C:\Users\HOME\AppData\Local\Temp\arduino\sketches\B5D3088AA6B6BC2E9E69B28F2F38CBB8\sketch\Joystick-AllFunctions.ino.cpp -o nul
C:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\2.1.0-a-d3d2e6b/bin/arm-none-eabi-g++ -I C:\Users\HOME\AppData\Local\Temp\arduino\sketches\B5D3088AA6B6BC2E9E69B28F2F38CBB8/core -c -Werror=return-type -Wno-psabi -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSBD_PID=0x000a -DUSBD_VID=0x2e8a -DUSBD_MAX_POWER_MA=250 -DUSB_MANUFACTURER="Raspberry Pi" -DUSB_PRODUCT="Pico" -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1 -DCYW43_LWIP=1 -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 -DARDUINO_VARIANT="rpipico" -DTARGET_RP2040 -DPICO_FLASH_SIZE_BYTES=2097152 -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions -DARM_MATH_CM0_FAMILY -DARM_MATH_CM0_PLUS -iprefixC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2/ @C:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2/lib/platform_inc.txt -IC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2/include -fno-rtti -std=gnu++17 -g -pipe -w -x c++ -E -CC -DF_CPU=133000000L -DARDUINO=10607 -DARDUINO_RASPBERRY_PI_PICO -DBOARD_NAME="RASPBERRY_PI_PICO" -DARDUINO_ARCH_RP2040 -Os -DWIFICC=CYW43_COUNTRY_WORLDWIDE -IC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2\cores\rp2040 -IC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2\variants\rpipico -IC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2\libraries\Joystick\src -IC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2\libraries\HID_Joystick\src C:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2\libraries\Joystick\src\Joystick.cpp -o nul
C:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\2.1.0-a-d3d2e6b/bin/arm-none-eabi-g++ -I C:\Users\HOME\AppData\Local\Temp\arduino\sketches\B5D3088AA6B6BC2E9E69B28F2F38CBB8/core -c -Werror=return-type -Wno-psabi -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DUSBD_PID=0x000a -DUSBD_VID=0x2e8a -DUSBD_MAX_POWER_MA=250 -DUSB_MANUFACTURER="Raspberry Pi" -DUSB_PRODUCT="Pico" -DPICO_CYW43_ARCH_THREADSAFE_BACKGROUND=1 -DCYW43_LWIP=1 -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 -DARDUINO_VARIANT="rpipico" -DTARGET_RP2040 -DPICO_FLASH_SIZE_BYTES=2097152 -march=armv6-m -mcpu=cortex-m0plus -mthumb -ffunction-sections -fdata-sections -fno-exceptions -DARM_MATH_CM0_FAMILY -DARM_MATH_CM0_PLUS -iprefixC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2/ @C:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2/lib/platform_inc.txt -IC:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2/include -fno-rtti -std=gnu++17 -g -pipe -w -x c++ -E - [... cut ...]
Using library Joystick at version 1.0.1 in folder: C:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2\libraries\Joystick 
Using library HID_Joystick at version 1.0.1 in folder: C:\Users\HOME\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.2\libraries\HID_Joystick 
exit status 0xc0000409

Compilation error: exit status 0xc0000409
```
Tried both with and without the BOOTSEL button kept pressed before connecting procedure: same output.
Any idea what am I doing wrong? Thank you very much for the help :)

baritonomarchetto, 2023-12-12:
I installed back an older version (2.5.0) and the upload started back. I could then install more recent boards packages and I had no issues uploading new sketches. Oh well....

Anyway, I noticed that Joystick library works up to ver. 3.1.0. Newer versions (starting from 3.2.0) doesn't. 

Could anyone confirm it's a board package issue, or is it something happening to my setup only?

Thanks very much :)

earlephilhower, 2023-12-12:
> Fatal Python error: initfsencoding: unable to load the file system codec
ModuleNotFoundError: No module named 'encodings'

This is a known Windows IDE bug.  A reinstall fixes it in most cases.  The IDE just didn't expand all files properly as the message said.

The Joystick library included here is running fine with the demo, I'm watching it right now run through all buttons and axes.

Good luck!

closed 2023-12-12 by earlephilhower

baritonomarchetto, 2023-12-15:
Reinstalling the package after launching Arduino IDE in admin mode solves the issue: no more errors occur