earlephilhower/arduino-pico #1898: Joystick.h and Keyboard.h not compatible

baritonomarchetto, 2023-12-13:
This gave me some headache recently. I was used to merge these two libs toghether with no problems and after one year stuck with an old version of pi pico core I decided to update... 

They are no more compatible. You can check it with this simple sketch:

`//Short pin 10 to ground and joy button 1 will be pressed

#include <Joystick.h>
//#include <Keyboard.h>

const byte btn = 10;

void setup (){
  pinMode(btn, INPUT_PULLUP);
  Joystick.begin();// start joystick
}  // end of setup

void loop (){
  Joystick.button(1, !digitalRead(btn));
  delay(8);
}`

This one works. Uncomment Keyboard.h and it will not work.
The issue arose between ver 3.1.0 and 3.1.1

maxgerhardt, 2023-12-13:
Hm, the core does try to accomodate the Joystick + Keyboard case in its USB code, even in the latest version.

https://github.com/earlephilhower/arduino-pico/blob/3068cd0af004e2575f9d47b0851f3abab64124dc/cores/rp2040/RP2040USB.cpp#L166-L177

so maybe something broke somewhere else.. 

Do I understand it correct that "it will not work" means the USB setup fails or only shows on USB device instead of two? Or do both USB device show up in the device manager but the joystick doesn't report button presses?

earlephilhower, 2023-12-13:
I just tried the [Joystick-Alltest.ino](https://github.com/earlephilhower/arduino-pico/blob/master/libraries/Joystick/examples/Joystick-AllFunctions/Joystick-AllFunctions.ino) example and added `#include <Keyboard.h>` and it seems to work just fine.

The example you posted also works fine (I replaced the pin /wBOOTSEL for simplicity).  No issues seen at all:
````

#include <Joystick.h>
#include <Keyboard.h>

const byte btn = 10;

void setup (){
  pinMode(btn, INPUT_PULLUP);
  Joystick.begin();// start joystick
} 

void loop (){
  Joystick.button(1, !BOOTSEL);
  delay(8);
}
````

Please explain exactly what the issue you're seeing is, because it's not reproducible here...

earlephilhower, 2023-12-13:
Dumping the HID report descriptor I see the keyboard + special keys + joystick:
````
earle@amd:~/Downloads$ sudo usbhid-dump --model 2e8a:810a
003:030:002:DESCRIPTOR         1702480477.319117
 05 01 09 06 A1 01 85 01 05 07 19 E0 29 E7 15 00
 25 01 95 08 75 01 81 02 95 01 75 08 81 01 05 08
 19 01 29 05 95 05 75 01 91 02 95 01 75 03 91 01
 05 07 19 00 2A FF 00 15 00 26 FF 00 95 06 75 08
 81 00 C0 05 0C 09 01 A1 01 85 02 15 00 26 FF 03
 19 00 2A FF 03 95 01 75 10 81 00 C0 05 01 09 05
 A1 01 85 03 05 01 09 30 09 31 09 32 09 35 09 33
 09 34 15 81 25 7F 95 06 75 08 81 02 05 01 09 39
 15 01 25 08 35 00 46 3B 01 95 01 75 08 81 02 05
 09 19 01 29 20 15 00 25 01 95 20 75 01 81 02 C0
````
Decoded becomes:
````
0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x06,        // Usage (Keyboard)
0xA1, 0x01,        // Collection (Application)
0x85, 0x01,        //   Report ID (1)
0x05, 0x07,        //   Usage Page (Kbrd/Keypad)
0x19, 0xE0,        //   Usage Minimum (0xE0)
0x29, 0xE7,        //   Usage Maximum (0xE7)
0x15, 0x00,        //   Logical Minimum (0)
0x25, 0x01,        //   Logical Maximum (1)
0x95, 0x08,        //   Report Count (8)
0x75, 0x01,        //   Report Size (1)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x01,        //   Report Count (1)
0x75, 0x08,        //   Report Size (8)
0x81, 0x01,        //   Input (Const,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x08,        //   Usage Page (LEDs)
0x19, 0x01,        //   Usage Minimum (Num Lock)
0x29, 0x05,        //   Usage Maximum (Kana)
0x95, 0x05,        //   Report Count (5)
0x75, 0x01,        //   Report Size (1)
0x91, 0x02,        //   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x95, 0x01,        //   Report Count (1)
0x75, 0x03,        //   Report Size (3)
0x91, 0x01,        //   Output (Const,Array,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x05, 0x07,        //   Usage Page (Kbrd/Keypad)
0x19, 0x00,        //   Usage Minimum (0x00)
0x2A, 0xFF, 0x00,  //   Usage Maximum (0xFF)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x95, 0x06,        //   Report Count (6)
0x75, 0x08,        //   Report Size (8)
0x81, 0x00,        //   Input (Data,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              // End Collection
0x05, 0x0C,        // Usage Page (Consumer)
0x09, 0x01,        // Usage (Consumer Control)
0xA1, 0x01,        // Collection (Application)
0x85, 0x02,        //   Report ID (2)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x03,  //   Logical Maximum (1023)
0x19, 0x00,        //   Usage Minimum (Unassigned)
0x2A, 0xFF, 0x03,  //   Usage Maximum (0x03FF)
0x95, 0x01,        //   Report Count (1)
0x75, 0x10,        //   Report Size (16)
0x81, 0x00,        //   Input (Data,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              // End Collection
0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x05,        // Usage (Game Pad)
0xA1, 0x01,        // Collection (Application)
0x85, 0x03,        //   Report ID (3)
0x05, 0x01,        //   Usage Page (Generic Desktop Ctrls)
0x09, 0x30,        //   Usage (X)
0x09, 0x31,        //   Usage (Y)
0x09, 0x32,        //   Usage (Z)
0x09, 0x35,        //   Usage (Rz)
0x09, 0x33,        //   Usage (Rx)
0x09, 0x34,        //   Usage (Ry)
0x15, 0x81,        //   Logical Minimum (-127)
0x25, 0x7F,        //   Logical Maximum (127)
0x95, 0x06,        //   Report Count (6)
0x75, 0x08,        //   Report Size (8)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x01,        //   Usage Page (Generic Desktop Ctrls)
0x09, 0x39,        //   Usage (Hat switch)
0x15, 0x01,        //   Logical Minimum (1)
0x25, 0x08,        //   Logical Maximum (8)
0x35, 0x00,        //   Physical Minimum (0)
0x46, 0x3B, 0x01,  //   Physical Maximum (315)
0x95, 0x01,        //   Report Count (1)
0x75, 0x08,        //   Report Size (8)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x09,        //   Usage Page (Button)
0x19, 0x01,        //   Usage Minimum (0x01)
0x29, 0x20,        //   Usage Maximum (0x20)
0x15, 0x00,        //   Logical Minimum (0)
0x25, 0x01,        //   Logical Maximum (1)
0x95, 0x20,        //   Report Count (32)
0x75, 0x01,        //   Report Size (1)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              // End Collection

// 160 bytes
````

baritonomarchetto, 2023-12-14:
Thank you both for your replyes.
The issue I see is that on windows 10 joy.cpl if I press a button, nothing happens when the keyboard.h is included. In previous versions (3.1.0) the button press was recordee.

earlephilhower, 2023-12-14:
Your problem is you're using a version 3.1.1 that's about a year old with bugs that were fixed in the subsequent 3.2.0 version (also about a year old) per the release notes.  Use the latest 3.6.2 release.  Just tested the sketch on Win 11 w/the current core and no issues reporting the joystick data.

closed 2023-12-14 by earlephilhower

baritonomarchetto, 2023-12-14:
Sorry to bother, maybe I was not able to describe the issue or I am not understanding well your answer :)

If I use rel 3.6.2, win 10 joypad monitor (joy.cpl) shows no activity when I press a button if keyboard.h is included. It shows button press if I comment keyboard.h.

I would absolutely prefer to use the latest release, but it's not working as it should in my understanding.

Could it be a win 10 issue then?

I will also try an online joypad activity monitor tomorrow.

Hope I was more clear now, thank you very much for you time!

earlephilhower, 2023-12-14:
I just took the USB cable down to a Win10 laptop, same as was on Win 11.  No issues seen, button flicks when I hit BOOTSEL.  I'm sorry, but it's not reproducible and there haven't been any reports like this since before the 3.2.0 fix.

maxgerhardt, 2023-12-14:
OP might have a weird state of the Arduino-Pico core files with a half-botched update process. Maybe try a clean remove of the core (+ deletion of the rp2040 packages from the Arduino IDE intstall folder) and a clean reinstall.

baritonomarchetto, 2023-12-15:
Thank you very much for the help!

I made further testings and spotted the issue. Someway, the program I used to monitor joypad activity (joy.cpl) is no more compatible. My first thinking was a win32 related issue. It's not being that some other win32 programs work.

Other online gamepad monitors record the button press with no issues.

earlephilhower, 2023-12-15:
Thanks for closing the loop here!