earlephilhower/arduino-pico #1900: SPI: SCLK polarity (CPOL) is inverted

maxgerhardt, 2023-12-16:
Wikipedia says

[image]

so for CPOL = 0, the SCLK pin should idle LOW before and after a SPI transfer.

However, testing with
```cpp
#include <Arduino.h>
#include <SPI.h>
static int _sck = 2;
static int _mosi = 3;
static int _miso = 4;
static int _cs = 5;

void setup() {
    SPI.setSCK(_sck);
    SPI.setTX(_mosi);
    SPI.setRX(_miso);
    SPI.setCS(_cs);
    SPI.begin(true /* HW CS */);
}

void loop() {
  SPI.beginTransaction(SPISettings(100000, MSBFIRST, SPI_MODE0));
  uint8_t value = 0xAA;
  SPI.transfer(value);
  SPI.endTransaction();
  delay(100);
}
```
I see

[image]

on GP2, 3, 4 and 5 respectively. This is exactly inverted, it idles HIGH. Wut?

The signal also looks exactly the same when using native Pico-SDK functions:

```cpp
#include <Arduino.h>
#include <hardware/spi.h>
static int _sck = 2;
static int _mosi = 3;
static int _miso = 4;
static int _cs = 5;
static spi_inst_t* _spi = spi0;

void setup() {
  gpio_set_function(_sck, GPIO_FUNC_SPI);
  gpio_set_function(_mosi, GPIO_FUNC_SPI);
  gpio_set_function(_miso, GPIO_FUNC_SPI);
  gpio_set_function(_cs, GPIO_FUNC_SPI);

  spi_init(_spi, 100000);
  spi_set_format(_spi, 8, SPI_CPOL_0, SPI_CPHA_0, SPI_MSB_FIRST);
}

void loop() {
  uint8_t value = 0xAA;
  (void) spi_write_blocking(_spi, &value, sizeof(value));
  delay(100);
}
```

earlephilhower, 2023-12-18:
I'm kind of confused here, sorry.  Are you saying the SDK is inverted, or this core is?  And if it was inverted wouldn't there be serious trouble communicating with a wide array of devices?

The logic for converting from Arduino modes to SDK modes is here:
https://github.com/earlephilhower/arduino-pico/blob/c4f36170d4858b8115e50e2a0caba7c875f1751c/libraries/SPI/src/SPI.cpp#L47-L75

Looking at those 2 `switch`es and the Wikipedia article, the values themselves seem to be correct.

The actual values are set in `beginTransaction`, just as in your SDK code:
https://github.com/earlephilhower/arduino-pico/blob/c4f36170d4858b8115e50e2a0caba7c875f1751c/libraries/SPI/src/SPI.cpp#L193-194

maxgerhardt, 2023-12-18:
The conversion of the core from SPI mode to CPOL and CPHA is correct, what I'm saying is that SPI Mode 0 (equates CPOL = 0, CPHA = 0) should have a clock idling LOW (per wikipedia), but is actually idling HIGH on real hardware. It's as if the RP2040 chip has inverted SCLK idle levels for CPOL = 0 / 1.

earlephilhower, 2023-12-18:
Ah, gotcha.  Maybe this would be better handled at the [Pico-SDK repo](https://github.com/raspberrypi/pico-sdk) or the [Raspberry Pi forum?](https://forums.raspberrypi.com/viewforum.php?f=144).  

Looking at the [chip docs](https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf), they do show the clock low during idle for CPOL=0 (SPO=0)
[image]

and high during idle for CPOL=1:  (SPO=1)
[image]

The `SSPCR0` register has the 2 bits for CPOL/CPHA, and I see nothing else related to clock polarity elsewhere.

maxgerhardt, 2023-12-18:
You're right, issue in pico-sdk opened per above.

earlephilhower, 2024-01-03:
Closing here, being followed up on (hopefully) in the SDK repo...

closed 2024-01-03 by earlephilhower