earlephilhower/arduino-pico #1905: Compile Errors

sk8board, 2023-12-18:
I have cloned this arduino-pico github using Platform IO with Visual Studio.

`gh repo clone earlephilhower/arduino-pico`

Then added a platform.ini file

```
[platformio]
default_envs = pipico

[env:pipico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board_build.core = earlephilhower
board = pico
framework = arduino
```

Then selected build and this is the result.

```
Linking .pio/build/pipico/firmware.elf
/home/home14/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/12.3.0/../../../../arm-none-eabi/bin/ld: .pio/build/pipico/libFrameworkArduino.a(main.cpp.o): in function `main':
/home/home14/.platformio/packages/framework-arduinopico/cores/rp2040/main.cpp:148: undefined reference to `setup'
/home/home14/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/12.3.0/../../../../arm-none-eabi/bin/ld: /home/home14/.platformio/packages/framework-arduinopico/cores/rp2040/main.cpp:150: undefined reference to `loop'
collect2: error: ld returned 1 exit status
*** [.pio/build/pipico/firmware.elf] Error 1
```

Where did I go wrong?

Shouldn't this project build without modification?

earlephilhower, 2023-12-18:
This is an Arduino compatible core.  Your own app supplies the `setup` and `loop`.  Just compiling without them will give the error you showed.  See the examples for some ideas.

closed 2023-12-18 by earlephilhower

sk8board, 2023-12-18:
The servo example worked perfectly when I renamed it to main.cpp. 

I am new to Platform IO. Thank you for your patience.