earlephilhower/arduino-pico #1909: Built-in LED dose not work (Raspberry Pi Pico W)

alchg, 2023-12-22:
Arduino IDE 1.8.19
version 3.6.2

#define WIFI_PIN  23
void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  pinMode(WIFI_PIN, OUTPUT);
  digitalWrite(WIFI_PIN, LOW);
  digitalWrite(LED_BUILTIN, HIGH);
  digitalWrite(WIFI_PIN, LOW);
}
void loop(){
  digitalWrite(LED_BUILTIN, HIGH);
  delay(100);
  digitalWrite(LED_BUILTIN, LOW);
  delay(100);
}

earlephilhower, 2023-12-22:
The LED works fine, this is probably just a setting problem.

Make sure you set the board to `Raspberry Pi Pico W` in the Tools menu, and then try the Blink.ino example.

closed 2023-12-22 by earlephilhower

earlephilhower, 2023-12-22:
Also, GPIO23 is the WL_ON pin and owned by the WiFi interface.  Do not mess with it or you'll kill the WiFi chip and hence kill the LED that's connected to it.