earlephilhower/arduino-pico #1912: A clarification of changing the I2C pins

djaus2, 2023-12-24:
In the [Doumentation- for Wire(I2C)](https://arduino-pico.readthedocs.io/en/latest/wire.html) its says:

> You may change these pins before calling Wire.begin() or Wire1.begin() using:
```
bool setSDA(pin_size_t sda);
bool setSCL(pin_size_t scl);
```

In coding that for Wire(I2C0) it would be, _for example_:
```
  Wire.setSDA(4);
  Wire.setSCL(5);
```

- 4 and 5 are the defaults for the Pico for IC0 so not needed there.
- NB: For the Grove Pico Shield, 8 and 9  _(undocumented)_ respectively are used for IC0 so the above is required there using 8 and 9 in place of 4 and 5.
- IC1 on the Grove Shield is connected to 6 and 7 _(as documented)_.

earlephilhower, 2023-12-26:
Thanks, but I'm not quite following you.  If you could do this as a PR that would be helpful.  I'd also rather not call out one specific board for pinouts in the docs.  Those can be set right in the variant header to be used automatically.

closed 2023-12-26 by earlephilhower

djaus2, 2023-12-27:
The primary issue was that I was clarifying for others that when changing the I2C SDA and SCL pins, the code for the method calls require Wire or Wire1 to be prepended. For example
        Wire.setSDA(8);
        Wire.setSCL(9);
The specific RPi Pico issues I’ll blog about.
I’ll do a PR.