earlephilhower/arduino-pico #1918: Setting up AP no longer sets default IP address

zhiquanyeo, 2023-12-29:
In earlier versions of the Arduino Core, the following code would result in `IP Address: 192.168.42.1` getting printed to Serial. However, with 3.6.2, I get `IP Address: (IP unset)` instead.

```
#include <WiFi.h>

void setup() {
  // put your setup code here, to run once:
  while (!Serial) {}

  Serial.begin(115200);

  Serial.println("Creating AP");
  WiFi.softAP("test-network", "test-password");
  delay(1000);

  Serial.print("IP Address: ");
  Serial.println(WiFi.softAPIP());
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

To work around this, I had to add the following before the `WiFi.softAP(...)` call:

```
  // Workaround
  IPAddress ip {192, 168, 42, 1};
  IPAddress subnet {255, 255, 255, 0};
  WiFi.softAPConfig(ip, ip, subnet);
```

Was this an expected change?

earlephilhower, 2024-01-03:
No, this was a regression.   #1862 seems to wipe the AP config, need to investigate.

earlephilhower, 2024-01-03:
Fixed with https://github.com/earlephilhower/arduino-pico/commit/afbba6854973abc6806dba47d2defe1d0de5940b

closed 2024-01-03 by earlephilhower