earlephilhower/arduino-pico #1919: Cannot program Pi Pico from Apple Silicon Mac

rgov, 2023-12-30:
Arduino IDE 2.2.1, board package 3.6.2

When I try to upload a program, I get:

```

Waiting for gdb server to start...[2023-12-30T21:16:53.799Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
~/Library/Arduino15/packages/rp2040/tools/pqt-openocd/2.1.0-a-d3d2e6b/bin/openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s /private/var/folders/q_/s9k7ck0s1j7065yndz6njy300000gn/T/.arduinoIDE-unsaved20231130-2052-14s0gy3.kjva/sketch_dec30a -f "/Applications/Arduino IDE.app/Contents/Resources/app/plugins/cortex-debug/extension/support/openocd-helpers.tcl" -f ~/Library/Arduino15/packages/rp2040/hardware/rp2040/3.6.2/lib/picoprobe_cmsis_dap.tcl
dyld[8210]: Library not loaded: /usr/local/opt/libusb/lib/libusb-1.0.0.dylib
  Referenced from: <266FCA80-426A-3486-B127-7DC7E66393DE> ~/Library/Arduino15/packages/rp2040/tools/pqt-openocd/2.1.0-a-d3d2e6b/bin/openocd
  Reason: tried: '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/usr/local/opt/libusb/lib/libusb-1.0.0.dylib' (no such file), '/usr/local/lib/libusb-1.0.0.dylib' (no such file), '/usr/lib/libusb-1.0.0.dylib' (no such file, not in dyld cache)
[2023-12-30T21:16:53.826Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next sessio to start...
```

There are a few issues:

1. It's assuming `/usr/local/opt/libusb/lib/libusb-1.0.0.dylib` exists. When installed with Homebrew on Apple Silicon, the default path is under `/opt/homebrew/opt/libusb`.
2. The `openocd` binary is an Intel binary, so loading the Apple Silicon library would fail even if it were at that path.

earlephilhower, 2023-12-30:
Sorry, this is known and won't change until there is either a) a working osxcross that can build for ARM or b) GitHub provides Apple ARM VMs for OSS so it can be built on real silicon w/the expected paths using a GH action (like is done for openocd on Windows and Mac x64).  Feel free to try the homebrew versions of openocd (but I can't say if the support the Picoprobe CMSIS-DAP adapter).

You'll just need to use the UF2 (standard file-based) upload in the meantime.  Good luck!

rgov, 2023-12-31:
I was using UF2 upload actually, but maybe I was pressing the 'Debug' button instead of 'Upload'? I will try.

I understand this is a known limitation but it would be useful to keep an issue like this open/pinned, or make a note in the README, because virtually all Macs sold for the past few years are Apple Silicon-based and therefore it will be a common issue for Mac users.

rgov, 2023-12-31:
> or b) GitHub provides Apple ARM VMs for OSS so it can be built on real silicon w/the expected paths using a GH action

This [seems to be available](https://github.blog/2023-10-02-introducing-the-new-apple-silicon-powered-m1-macos-larger-runner-for-github-actions/) since October, also.

earlephilhower, 2023-12-31:
Unfortunately that's only in the paid pro tier for now.  I think I did the math and it was cheaper to buy a new Mac Mini than to run an instance for 3 days of wall time.  It'll eventually come down, I'm sure, but definitely not for us yet.

rgov, 2023-12-31:
If you point me to the build system I could take a glance at it. Intel Macs can cross-compile for Apple Silicon relatively easily (but cannot execute the binaries, so it has to be smart enough to build any intermediate tools for the host architecture).

I think it makes sense to delegate as much as possible to Homebrew since they have more bandwidth available to support them. There's OpenOCD and the arm-none-eabi-* toolchain for example (which is what the Pico SDK suggests to use).

Is it possible to give the user a way to configure the search path? Or perhaps it's easier to run the user's default shell and extract the `PATH` from it:

```bash
$(dscl . -read /Users/$USER UserShell | awk '{print $2}') -l -c 'echo $PATH'
```

earlephilhower, 2024-01-01:
Here is the uber toolchain builder:  https://github.com/earlephilhower/pico-quick-toolchain .  It's set up for Linux hosted cross-building in a Docker container, so one parallel `make -j16` and 30 minutes and you have ARM64, RPI, Win32, Win64, Mac x64, Linux x32, and Linux x64 binaries for everything (gcc, gdb, mklittlefs, etc.).

There are 2 GH actions I use to build the OpenOCD for Win32/64 and Mac x64 using GH's native builders, see https://github.com/earlephilhower/pico-quick-toolchain/blob/master/.github/workflows/build-macos.yml

Using Homebrew's GCC bundle is not really recommended.  We have a specific version of Newlib (the C runtime library) configured with hooks that let us support multicore and FreeRTOS.  Simple sketches may work, but more complicated ones will go boom in very unpleasant ways.  You'd also end up with different binaries when building on Mac ARM vs. any other one which is going to be one heck of a nightmare for support.

For OpenOCD and Picotool, using homebrew should be fine assuming the versions are relatively close to each other (and of course if the OpenOCD from homebrew supports the Picoprobe!).

What might make sense for Macs is to ship a *script* instead of a binary as `{runtime.platform.tools}/openocd/bin/openocd`.  That script could just be something like
````
#!/bin/bash
exec openocd "$@"`
````
If there's one in the path, you're golden.  If not, you're no worse off than now, no?

rgov, 2024-01-01:
### For OpenOCD specifically

I see that you are copying from Homebrew, `cp -a /usr/local/Cellar/hidapi/*/lib/*.dylib /tmp/picotool/.` etc. This isn't generalizable, since these hardcode library load paths:

```
$ otool -L ~/Library/Arduino15/packages/rp2040/tools/pqt-openocd/2.1.0-a-d3d2e6b/bin/libhidapi.dylib 
~/Library/Arduino15/packages/rp2040/tools/pqt-openocd/2.1.0-a-d3d2e6b/bin/libhidapi.dylib:
	/usr/local/opt/hidapi/lib/libhidapi.0.dylib (compatibility version 0.0.0, current version 0.14.0)
```

What you can do is use `install_name_tool` to patch the binaries to use relative load paths. [Here is a script for doing so](https://gist.github.com/rgov/c1702279a18cac2eac8662445a4da80d), and I confirmed it works to get `openocd` running under Rosetta (but didn't exercise the tool).

### For true native ARM builds of the toolchain

Am I correct that your [gcc-cross](https://github.com/earlephilhower/gcc-cross) container needs to be updated to include the macOS 11 SDK in order for [osxcross](https://github.com/tpoechtrager/osxcross) to be able to target ARM64?

> arm64 requires macOS 11.0 SDK (or later).  
> arm64e [requires a recent Apple clang compiler.](https://github.com/apple/llvm-project)

Your build process copies the SDK from a local path but perhaps with a GitHub Action you could spin up a macOS VM, run the [gen_sdk_package.sh](https://github.com/tpoechtrager/osxcross/blob/master/tools/gen_sdk_package.sh) script, and then copy the resulting SDK tarball to another Linux job where you actually build GCC.

([multiarch/crossbuild](https://github.com/multiarch/crossbuild) seems to be very similar but _also_ doesn't support ARM64, https://github.com/multiarch/crossbuild/issues/51, but there's a [fork](https://github.com/multiarch/crossbuild/compare/master...gotson:crossbuild:m1) that makes it look pretty simple to adapt.)

It looks that you can't compile universal (Intel + ARM) binaries with osxcross GCC without building for the targets separately and then merging them.

earlephilhower, 2024-01-02:
@rgov , could you please check the `Artifacts` of the GH action at https://github.com/earlephilhower/pico-quick-toolchain/actions/runs/7389568090 ?  I've applied your script and changes to the Picotool and OpenOCD packaging and it *looks* like it's working from the outputs:

````
/tmp/picotool/picotool:
	/usr/local/opt/libusb/lib/libusb-1.0.0.dylib (compatibility version 4.0.0, current version 4.0.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 1200.3.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1311.0.0)
[*] Processing libusb-1.0.0.dylib...
/Users/runner/work/pico-quick-toolchain/pico-quick-toolchain/picotool/build/../../osx-relative-dynlib.sh: line 20: continue: only meaningful in a `for', `while', or `until' loop
[*] Processing libusb-1.0.dylib...
/Users/runner/work/pico-quick-toolchain/pico-quick-toolchain/picotool/build/../../osx-relative-dynlib.sh: line 20: continue: only meaningful in a `for', `while', or `until' loop
[*] Processing ./picotool...
  [+] Made relative link to libusb-1.0.0.dylib
/tmp/picotool/picotool:
	@executable_path/libusb-1.0.0.dylib (compatibility version 4.0.0, current version 4.0.0)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 1200.3.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1311.0.0)
````
and
````
/tmp/openocd/bin/openocd:
	/usr/local/opt/libusb/lib/libusb-1.0.0.dylib (compatibility version 4.0.0, current version 4.0.0)
	/usr/local/opt/hidapi/lib/libhidapi.0.dylib (compatibility version 0.0.0, current version 0.14.0)
	/usr/local/opt/capstone/lib/libcapstone.5.dylib (compatibility version 5.0.0, current version 5.0.1)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1311.0.0)
[*] Processing libcapstone.5.dylib...
/Users/runner/work/pico-quick-toolchain/pico-quick-toolchain/osx-relative-dynlib.sh: line 20: continue: only meaningful in a `for', `while', or `until' loop
[*] Processing libcapstone.dylib...
/Users/runner/work/pico-quick-toolchain/pico-quick-toolchain/osx-relative-dynlib.sh: line 20: continue: only meaningful in a `for', `while', or `until' loop
[*] Processing libhidapi.0.14.0.dylib...
/Users/runner/work/pico-quick-toolchain/pico-quick-toolchain/osx-relative-dynlib.sh: line 20: continue: only meaningful in a `for', `while', or `until' loop
[*] Processing libhidapi.0.dylib...
/Users/runner/work/pico-quick-toolchain/pico-quick-toolchain/osx-relative-dynlib.sh: line 20: continue: only meaningful in a `for', `while', or `until' loop
[*] Processing libhidapi.dylib...
/Users/runner/work/pico-quick-toolchain/pico-quick-toolchain/osx-relative-dynlib.sh: line 20: continue: only meaningful in a `for', `while', or `until' loop
[*] Processing libusb-1.0.0.dylib...
/Users/runner/work/pico-quick-toolchain/pico-quick-toolchain/osx-relative-dynlib.sh: line 20: continue: only meaningful in a `for', `while', or `until' loop
[*] Processing libusb-1.0.dylib...
/Users/runner/work/pico-quick-toolchain/pico-quick-toolchain/osx-relative-dynlib.sh: line 20: continue: only meaningful in a `for', `while', or `until' loop
[*] Processing ./openocd...
  [+] Made relative link to libusb-1.0.0.dylib
  [+] Made relative link to libhidapi.0.dylib
  [+] Made relative link to libcapstone.5.dylib
/tmp/openocd/bin/openocd:
	@executable_path/libusb-1.0.0.dylib (compatibility version 4.0.0, current version 4.0.0)
	@executable_path/libhidapi.0.dylib (compatibility version 0.0.0, current version 0.14.0)
	@executable_path/libcapstone.5.dylib (compatibility version 5.0.0, current version 5.0.1)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1311.0.0)
````

The PR is https://github.com/earlephilhower/pico-quick-toolchain/pull/35 .  Also, what attribution (if any) should there be on the bash script?  I'll update the commit message to reference this discussion, of course, but for now it was just a quick and dirty hack.

Re full GCC toolchain on Apple ARM:  It's actually a little more complicated that that.  I can easily update the docker image with a later SDK (and have done locally while experimenting).  The real problem is that GCC 12.3 doesn't build out of the box for Apple ARM.  I think there's a GCC 12.2 patch set available in homebrew's recipes but haven't spent any time seeing how far it is from working on GCC 12.3.  About a year ago I went thru the motions, documented here https://github.com/earlephilhower/pico-quick-toolchain/issues/8 .

rgov, 2024-01-02:
I downloaded both and could execute them via the command line. When I tried to run them via the Arduino IDE, macOS complained about the quarantine which I resolved with `xattr -d com.apple.quarantine -r ~/Library/Arduino15/packages/rp2040` — hopefully not needed when installed by Arduino.

I tried setting the upload method to "picotool" and selecting the serial port for an attached Pico, but it errors with the same as #1344:

```
$ ./picotool info
Device at bus 1, address 1 appears to be a RP2040 device with a USB serial connection, so consider -f (or -F) to force reboot in order to run the command.

$ ./picotool reboot --bus 1 --address 1 -f
ERROR: Unable to locate reset interface on the device
```

earlephilhower, 2024-01-02:
Much appreciate the quick feedback!

The picotool error is expected.  There's no special custom picotool USB device emulated by the core unless you actually upload an image using picotool.  The `picotool info` is the same issue, and additionally the RPI specific "mark this pin as this usage" data structure does not work unless you run without an OTA boot stage.  So it too should work in bootloader mode but report nothing.

Rebooting into BOOTSEL UF2 uploader mode should expose a picotool USB compound device that the `reboot` command would work on.

> OpenOCD can likewise be started from the Arduino IDE but I'm not sure how to use it; it fails with "unable to find a matching CMSIS-DAP device". I don't have a debug probe attached.

I assume you do have one, just not connected now, or the initial bug here (uploading via a picoprobe CMSIS_DAP adapter) wouldn't make any sense.

rgov, 2024-01-02:
I rebooted with <kbd>BOOTSEL</kbd> then programmed using `picotool`, which seemed to work:

```
Loading into Flash: [==============================]  100%
```

```
$ ./picotool info                            
Program Information
 none
```

This allows me to keep uploading in UF2 mode.

---

Seems like you ought to also revert the macOS instructions in #1355.

earlephilhower, 2024-01-02:
Will note that when I do the package.json update PR in this repo, thx!

closed 2024-01-03 by earlephilhower

earlephilhower, 2024-01-05:
Looks like GCC12.3 does now have a patchset for Apple M1:  https://raw.githubusercontent.com/Homebrew/formula-patches/f1188b90d610e2ed170b22512ff7435ba5c891e2/gcc/gcc-12.3.0.diff

earlephilhower, 2024-01-05:
And unfortunately that patch seems to break the binutils quadmath library `configure` in some strange way when building without C++ exceptions.  😱

commit ff3e8969 (2024-01-03): Update Max Picotool/OpenOCD to use bundled dylibs

Fixes #1919 by using binaries from https://github.com/earlephilhower/pico-quick-toolchain/pull/37
commit 8495eabc (2024-01-03): Update Max Picotool/OpenOCD to use bundled dylibs

Fixes #1919 by using binaries from https://github.com/earlephilhower/pico-quick-toolchain/pull/37
commit b3628e69 (2024-01-03): Update Mac Picotool/OpenOCD to use bundled dylibs

Fixes #1919 by using binaries from https://github.com/earlephilhower/pico-quick-toolchain/pull/37
PR #1922 (2024-01-03): Update Mac Picotool/OpenOCD to use bundled dylibs
Fixes #1919 by using binaries from https://github.com/earlephilhower/pico-quick-toolchain/pull/37