earlephilhower/arduino-pico #1925: Feature Request: provide ping functionality

5chufti, 2024-01-04:
is it possible to implement ping functionality directly ot like in ESPping library (as both are based on lwIP) ? at the moment ESPping lib misses at least some export from lwIP:

C:\Daten\arduino\portable\sketchbook\libraries\ESPping\src/ESPping.h:63:5: error: 'ping_option' does not name a type; did you mean 'ip_get_option'?
   63 |     ping_option _options;
      |     ^~~~~~~~~~~
      |     ip_get_option

thanks and keep up the great work!

5chufti, 2024-01-04:
Hi,
sorry but I couldn't find any traces in the documentation (arduino-pico.readthedocs.io WiFi) or the examples - so I assumed it not to be implemented. 
rgds

p.s.: with new infor tried     "WiFi.ping(WiFi.gatewayIP());" to no avail ...

earlephilhower, 2024-01-04:
We implement pretty much the Arduino `WiFi` library, so for docs the official website is probably the best bet.

That said, ping seems to be working fine here with a very simple MCVE:
````
#include <WiFi.h>

#ifndef STASSID
#define STASSID "your-ssid"
#define STAPSK "your-password"
#endif

const char* ssid = STASSID;
const char* password = STAPSK;

WiFiMulti multi;

void setup() {
  Serial.begin(115200);
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  multi.addAP(ssid, password);
  if (multi.run() != WL_CONNECTED) {
    Serial.println("Unable to connect to network, rebooting in 10 seconds...");
    delay(10000);
    rp2040.reboot();
  }
  Serial.println();
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  Serial.printf("Ping 8.8.8.8 result: %d\n", WiFi.ping(IPAddress(8,8,8,8)));
  Serial.printf("Ping GW result: %d\n", WiFi.ping(WiFi.gatewayIP()));
  Serial.printf("Ping 10.0.0.1 result (should fail): %d\n", WiFi.ping(IPAddress(10,0,0,1)));
  delay(10000);
}
````

Gives
````
11:06:20.291 -> WiFi connected
11:06:20.291 -> IP address: 
11:06:20.291 -> 192.168.1.131
11:06:20.325 -> Ping 8.8.8.8 result: 128
11:06:20.325 -> Ping GW result: 128
11:06:35.320 -> Ping 10.0.0.1 result (should fail): -1
````

closed 2024-01-04 by 5chufti

earlephilhower, 2024-01-04:
Yeah, that's kind of on me.  The WiFi class itself (i.e. WiFi.begin, WiFi.end, etc.) is pretty much Arduino, but the support classes like WiFiClient, WebServer, etc. are much closer to the ESP8266/32 (if there even is an official Arduino equivalent).