earlephilhower/arduino-pico #1929: Libraries not found when compiling for Pico W with PlatformIO

DoomHammer, 2024-01-09:
I'm not sure if this problem is related to the core or to PlatformIO and it is a bit confusing.

I have several projects with more or less identical `platformio.ini` settings.

One of the projects compiles without problem and finds Wifi.h, DNSServer.h, and all the other netowrking related libraries.

In two other projects, regardless whether I select the `pico` or `rpipicow` core, I get `fatal error: WiFi.h: No such file or directory`

I compared the verbose output from the working and non-working projects and it turns out, the non-working one is missing a bunch of `-I` include dir directives. To be specific: the ones responsible for WiFi, DNSServer, etc.

I worked around this problem by just copying the hardcoded flags and pasting them in `lib_deps` in `platformio.ini`, but let's just say I don't feel great about it.

Any ideas what might be causing such problems?

larrybud2004, 2024-01-09:
"In two other projects, regardless whether I select the pico or rpipicow core,"

You state the "core", but do you have the proper BOARD set in your platformio.ini? e.g.

[env:rpipicow]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = rpipicow
framework = arduino
board_build.core = earlephilhower

This works for me.

Wifi.h should be in

c:\users\yourusername\.platformio\packages\framework-arduinopico\libraries\wifi\src\wifi.h

DoomHammer, 2024-01-09:
The exact configuration I use is:

```
[env:default]
framework = arduino
monitor_speed = 115200
monitor_rts = 0
monitor_dtr = 0
check_tool = clangtidy
platform = raspberrypi
board_build.core = earlephilhower
board_build.filesystem = littlefs
board_build.filesystem_size = 0.5m
#board = rpipicow
board = pico
lib_deps =
    robtillaart/RS485 @ ^0.4.0
   https://gitlab.com/doomhammerng/wifi-manager#rpi-picow
build_flags =
    -DDEBUG_RP2040_PORT=Serial
    -D PICOW
 ```

larrybud2004, 2024-01-09:
Yeah, there's your problem. 

#board = rpipicow
board = pico

The pound sign comments out that line, so your board definition is "pico" and not "rpipicow".

"pico" doesn't have wifi.

Remove 
board = pico

and make it

board = rpipicow

DoomHammer, 2024-01-09:
Sorry, I pasted a wrong config. I just modified this to uplaod the same project to the regular Pico. Here's the original that fails for me:

```
[env:default]
framework = arduino
monitor_speed = 115200
monitor_rts = 0
monitor_dtr = 0
check_tool = clangtidy
platform = raspberrypi
board_build.core = earlephilhower
board_build.filesystem = littlefs
board_build.filesystem_size = 0.5m
board = rpipicow
lib_deps =
    robtillaart/RS485 @ ^0.4.0
   https://gitlab.com/doomhammerng/wifi-manager#rpi-picow
build_flags =
    -DDEBUG_RP2040_PORT=Serial
    -D PICOW
 ```

larrybud2004, 2024-01-09:
I don't understand this:
https://gitlab.com/doomhammerng/wifi-manager#rpi-picow

Are you wanting to use a 3rd party wifi library or the built in one? And why are you using a wifi manager for an ESP32 when you're using a pico?

This configuration doesn't compile for me. When "platform" is set to raspberrypi, board ID rpipicow is unknown. Seems like you're not reading the compilation errors properly. Often the "root" error will lead to other compilation errors which aren't really errors. Here's what you need in your platformio.ini

Here's what you need for the picow

[env:rpipicow]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board_build.core = earlephilhower
board = rpipicow
framework = arduino

DoomHammer, 2024-01-09:
The WiFi Manager is a library that's providing a Captive Portal feature that my project requires.

The project's name is ESP32 WiFi Manager, but I've ported it to Pico W. Same story as with ESP8266 Audio which works on ESP8266, ESP32, Pico, and a few others. Or STM32-DMD which again works on STM32 and Pico. Not ideal, but it is what it is.

I've changed the platform as suggested and again I get `
.pio/libdeps/default/AsyncTCP_RP2040W/src/AsyncTCP_RP2040W.h:84:10: fatal error: WiFi.h: No such file or directory`

Here' the current config:

```
[env:default]
framework = arduino
monitor_speed = 115200
monitor_rts = 0
monitor_dtr = 0
check_tool = clangtidy
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board_build.core = earlephilhower
board_build.filesystem = littlefs
board_build.filesystem_size = 0.5m
board = rpipicow
lib_deps =
    robtillaart/RS485 @ ^0.4.0
    https://gitlab.com/doomhammerng/wifi-manager#rpi-picow
build_flags =
    -DDEBUG_RP2040_PORT=Serial
    -D PICOW
```

larrybud2004, 2024-01-09:
That's clearly a problem in this library:
AsyncTCP_RP2040W

What is that? I don't know what that is. I don't see it referenced in your project, do you?

I think you probably should clean your project or start with a very basic example of doing wifi and then add to it.

DoomHammer, 2024-01-09:
This is a file from the external library. This library compiles fine when used with ESPHome (yes, I know it's called ESPHome but it's compatible with Pico W as well). In general this library is fine and what it does is it tries to include "WiFi.h" but fails to do so.

But thanks to your suggestion I started thinking where else the problem might lie and I realized the code never calls `
#include "WifiManager.h"`. When I added this to the main.cpp file, the compilation seems to proceed fine. So this is probably a problem with PlatformIO.

larrybud2004, 2024-01-09:
Why would except PlatformIO to automatically include a header file from a 3rd party library?

earlephilhower, 2024-01-09:
This seems not related to the core here.  Feel free to open something up on the [P.IO repo ](https://github.com/platformio/platformio-core)(although they do not officially support any RP2040 cores) or [their forums](https://community.platformio.org/).  Good luck!

closed 2024-01-09 by earlephilhower

DoomHammer, 2024-01-09:
I'm not sure if this is only relevant to RP2040, I'll check with other platforms. But clearly it is not related to this repo but to PIO.

Thanks!

maxgerhardt, 2024-01-17:
Coming late to the party, but if your project includes a library which itself includes `WiFi.h` and it doesn't find the header, chances are you'ure using the wrong [lib_ldf_mode](https://docs.platformio.org/en/latest/projectconf/sections/env/options/library/lib_ldf_mode.html). For further analysis you would need to post your *exact* project files.