earlephilhower/arduino-pico #1952: Help with crt0 SIGTRAP error?

rivimey, 2024-01-23:
Hi

I'm building a project using platformio, which I'm also new to. I have got the arduino-pico framework building a `firmware.elf` and using the RPi Picoprobe I can program it into the RP2040 Stamp I'm using. However it fails to run properly, failing with a `SIGTRAP` / `isr_hardfault`  somewhere in `crt0.S` Here is the session log. I would very much appreciate any help you can give. Ask if you need more info!

``` platformio.ini
[platformio]
default_envs = ardu-philhower

; custom common options
[common]
build_flags =

lib_deps_builtin =
    SPI
    Wire

lib_deps_external =
    ../libraries/Adafruit_MCP23017_Arduino_Library
    ../libraries/AT24Cxx
    ../libraries/I2CRotEnc
    ../libraries/Adafruit_BusIO
    olikraus/U8g2@^2.35.9

[env:ardu-philhower]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git

framework = arduino
board = solderparty_rp2040_stamp
board_build.core = earlephilhower
board_build.f_cpu = 133000000
board_build.filesystem_size = 2m

; Build options
build_flags =
    ${common.build_flags}

; Library options
lib_deps =
    ${common.lib_deps_builtin}
    ${common.lib_deps_external}

upload_protocol = cmsis-dap
debug_tool = cmsis-dap
monitor_speed = 19200
```

***

``` session log file
Reading symbols from /Users/ruth/Arduino/solarpump/.pio/build/ardu-philhower/firmware.elf...
Warning: 'set target-async', an alias for the command 'set mi-async', is deprecated.
Use 'set mi-async'.
PlatformIO Unified Debugger -> https://bit.ly/pio-debug
PlatformIO: debug_tool = cmsis-dap
PlatformIO: Initializing remote target...
Open On-Chip Debugger 0.12.0-g4d87f6dca-dirty (2023-08-24-20:43)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : Hardware thread awareness created
Info : Hardware thread awareness created
adapter speed: 1000 kHz

Info : tcl server disabled
Info : telnet server disabled
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E6614103E780A82E
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 1000 kHz
Info : SWD DPIDR 0x0bc12477, DLPIDR 0x00000001
Info : SWD DPIDR 0x0bc12477, DLPIDR 0x10000001
Info : [rp2040.core0] Cortex-M0+ r0p1 processor detected
Info : [rp2040.core0] target has 4 breakpoints, 2 watchpoints
Info : [rp2040.core1] Cortex-M0+ r0p1 processor detected
Info : [rp2040.core1] target has 4 breakpoints, 2 watchpoints
Info : starting gdb server for rp2040.core0 on pipe
Info : accepting 'gdb' connection from pipe
Info : Found flash device 'gd gd25q64c' (ID 0x001740c8)
Info : RP2040 B0 Flash Probe: 8388608 bytes @0x10000000, in 128 sectors

Info : New GDB Connection: 1, Target rp2040.core0, state: halted
[rp2040.core1] halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x00000184 msp: 0x20041f00
warning: multi-threaded target stopped without sending a thread-id, using first non-exited thread
isr_hardfault () at /home/earle/Arduino/hardware/pico/rp2040/pico-sdk/src/rp2_common/pico_standard_link/crt0.S:98
98	/home/earle/Arduino/hardware/pico/rp2040/pico-sdk/src/rp2_common/pico_standard_link/crt0.S: No such file or directory.
[rp2040.core0] halted due to debug-request, current mode: Thread
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
[rp2040.core1] halted due to debug-request, current mode: Thread
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
Loading section .boot2, size 0x100 lma 0x10000000
Loading section .ota, size 0x24e4 lma 0x10000100
Loading section .partition, size 0xa1c lma 0x100025e4
Loading section .text, size 0x12758 lma 0x10003000
Loading section .rodata, size 0x11cc lma 0x10015758
Loading section .data, size 0x17ac lma 0x10016924
Info : Padding image section 0 at 0x100180d0 with 48 bytes (bank write end alignment)
Warn : keep_alive() was not invoked in the 1000 ms timelimit. GDB alive packet not sent! (5731 ms). Workaround: increase "set remotetimeout" in GDB
[rp2040.core0] halted due to debug-request, current mode: Thread
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
[rp2040.core1] halted due to debug-request, current mode: Thread
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
Start address 0x100030d4, load size 98512
Transfer rate: 14 KB/sec, 9851 bytes/write.
[rp2040.core0] halted due to debug-request, current mode: Thread
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
[rp2040.core1] halted due to debug-request, current mode: Thread
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
[rp2040.core0] halted due to debug-request, current mode: Thread
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
[rp2040.core1] halted due to debug-request, current mode: Thread
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
Temporary breakpoint 1 at 0x100081a8: file /Users/ruth/.platformio/packages/framework-arduinopico/cores/rp2040/main.cpp, line 81.
Note: automatically using hardware breakpoints for read-only addresses.
PlatformIO: Initialization completed
PlatformIO: Resume the execution to `debug_init_break = tbreak main`
PlatformIO: More configuration options -> https://bit.ly/pio-debug
[rp2040.core1] halted due to debug-request, current mode: Thread
xPSR: 0x01000000 pc: 0x00000184 msp: 0x20041f00

Thread
1 "rp2040.core0" received signal SIGTRAP, Trace/breakpoint trap.
isr_hardfault () at /home/earle/Arduino/hardware/pico/rp2040/pico-sdk/src/rp2_common/pico_standard_link/crt0.S:98
98	in /home/earle/Arduino/hardware/pico/rp2040/pico-sdk/src/rp2_common/pico_standard_link/crt0.S
Info : SWD DPIDR 0x0bc12477, DLPIDR 0x00000001
Error: Failed to read memory at 0x88022300

Fatal signal:
Abort trap: 6

----- Backtrace -----
Backtrace unavailable
---------------------
A fatal error internal to GDB has been detected, further
debugging is not possible.  GDB will now terminate.
This is a bug, please report it.
  For instructions, see:
<https://www.gnu.org/software/gdb/bugs/>
.

```

maxgerhardt, 2024-01-26:
When you upload a blinky firmware in the current configuration (and not start debugging), does that run?

maxgerhardt, 2024-01-26:
Not really, I meant a blinky written with this regular Arduino-Pico core. It should be flashable via the cmsis-dap protocol, setup as you now have.
```cpp
#include <Arduino.h>
#define LED LED_BUILTIN // or whatever
void setup() { pinMode(LED, OUTPUT); }
void loop() { digitalWrite(LED, digitalRead(LED) ^ 1); delay(1000); }
```

rivimey, 2024-01-26:
Aside: before I used platformio, I compiled my project using a simple Makefile with compiler flags cribbed from an Arduino build; this built and linked ok but also failed in the crt0 code in what seems to be a very similar way. The switch to platformio was supposed to avoid the 'handcrafted' aspect of this build. Sadly this doesn't seem to have worked.

I will say I'm not doing anything at all weird, especially in during crt0/start/main startup. There are some c++ constructors which could perhaps start doing something but as far as I know that happens after crt0, not during.

earlephilhower, 2024-01-26:
Without a full MCVE we're really just guessing here and not likely to get down to the root cause, but...

> I will say I'm not doing anything at all weird, especially in during crt0/start/main startup. There are some c++ constructors which could perhaps start doing something but as far as I know that happens after crt0, not during.

C++ constructors are all called as part of the C++ runtime initialization, **in random order**.  If a C++ global object constructor uses another global object there is no guarantee that the called object's constructor will have been called.  Recipe for confusion, at least, or a crash, more likely.

That said, please run @maxgerhardt 's very basic blink example and verify that works.  If it does, there's a problem in the crashing app's code.  If blink doesn't work, there's a toolchain config problem (wrong flash type specified, maybe?)...

rivimey, 2024-01-27:
Thanks both for your suggestions, appreciated. I first tried the blinky code (by replacing src/* with a cpp file containing that code) and it worked fine.

I then tried reinstating all my project's code (except u8g2) but commenting out any line that could cause code execution, so both of the global c++ objects (one for each core) and so on. I then included the lines from blinky so it would run that within the context of my code. That worked too.

I then re-enabled one core's global constructor, and it failed immediately, as (now) expected. It seems the point about constructor initialisation is well made, though I'm unsure if this is about ordering or not, as there is only one global constructor being enabled here.

I then made the global object a pointer, with an initialisation within arduino setup(). That also worked (in so far as it didn't crash -- there was still no loop() code).

I then thought I would single-step through the setup() code, see what was happening. So, in debug mode, set a breakpoint in setup() and upload. It stopped at the beginning of the framework's main.cpp as expected. I click on 'step' once and it dies with "trap 6". I can't see anything else wrong in the log, and a breakpoint set a couple of lines later in main() [on the freeRTOS check line] is not hit.

```
Fatal signal: 
Abort trap: 6

----- Backtrace -----
Backtrace unavailable
---------------------
A fatal error internal to GDB has been detected, further
debugging is not possible.  GDB will now terminate.
```

Out of all this I am somewhat confused why bad/faulty app code can cause a crash in crt0 code -- is this just about c++ initialisation doing unexpected things or is something else going on?