earlephilhower/arduino-pico #1967: W5500 example compile error

lazaj, 2024-01-28:
Hello,

I've tried to modify an old project that rely on W5500 on a new computer.
I use platformio. The old project compiled fine, but now I have this error :
/Users/julien/.platformio/packages/framework-arduinopico/libraries/WiFi/src/WiFiClass.cpp:275:18: error: no matching function for call to 'NoDriverLwIP::end()'

So I create a new project with the code from the example file WiFiClient-W5500.ino.
I have the same error.

/Users/julien/.platformio/packages/framework-arduinopico/libraries/WiFi/src/WiFiClass.cpp: In member function 'int WiFiClass::disconnect(bool)':
/Users/julien/.platformio/packages/framework-arduinopico/libraries/WiFi/src/WiFiClass.cpp:275:18: error: no matching function for call to 'NoDriverLwIP::end()'
  275 |         _wifi.end();
      |         ~~~~~~~~~^~
In file included from /Users/julien/.platformio/packages/framework-arduinopico/libraries/WiFi/src/WiFiClass.h:29,
                 from /Users/julien/.platformio/packages/framework-arduinopico/libraries/WiFi/src/WiFi.h:6,
                 from /Users/julien/.platformio/packages/framework-arduinopico/libraries/WiFi/src/WiFiClass.cpp:28:
/Users/julien/.platformio/packages/framework-arduinopico/libraries/WiFi/src/utility/lwIP_nodriver.h:40:10: note: candidate: 'void NoDriverLwIP::end(bool)'
   40 |     void end(bool apMode) {
      |          ^~~
/Users/julien/.platformio/packages/framework-arduinopico/libraries/WiFi/src/utility/lwIP_nodriver.h:40:10: note:   candidate expects 1 argument, 0 provided

here my platformio.ini file :
[env:adafruit_feather]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = adafruit_feather
framework = arduino
board_build.core = earlephilhower

earlephilhower, 2024-01-28:
Fixed with #1966 which was just merged.  Please pull the latest `git master` and try again.

closed 2024-01-28 by earlephilhower

lazaj, 2024-01-28:
Compile issue is fixed but WiFiClient-W5500.ino still doesn't work it hang on eth.begin()