earlephilhower/arduino-pico #1969: WiFiClient-W5500.ino despite https://github.com/earlephilhower/arduino-pico/pull/1966

lazaj, 2024-01-28:
Despite patch https://github.com/earlephilhower/arduino-pico/pull/1966

It hang on eth.begin().

lazaj, 2024-01-28:
still with platformio.

// Platformio.ini
[env:adafruit_feather]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = adafruit_feather
framework = arduino
board_build.core = earlephilhower
platform_packages =
   framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git#master

earlephilhower, 2024-01-28:
Sorry, can't reproduce that on either a W5100 board or my W5500 mini board using latest `git master`:

Using WiFiClient-W5500 with no changes (my w5500 mini board is wired to pins 0-3 of a standard Pico) gives the expected output, no hangs:
````
12:36:51.519 -> Starting Ethernet port
12:36:51.519 -> .
12:36:52.048 -> Ethernet connected
12:36:52.048 -> IP address: 
12:36:52.048 -> 192.168.1.181
12:36:52.048 -> connecting to djxmmx.net:17
12:36:52.224 -> sending data to server
12:36:52.252 -> receiving from remote server
12:36:52.252 -> "Someone's not dead, until they're completely dead..."
12:36:52.252 -> 	 - Munro Kelly (Congo)
````

I suggest checking your wiring and if that checks out then hook up a PicoProbe and see where your code is stuck and maybe how to reproduce it outside of your own board.

earlephilhower, 2024-01-28:
Actually, there might be a difference between your binary and mine:  I'm running the plain Arduino IDE built version since I don't(can't) run Platform.IO.  Would you be able to try your app on the exact same sketch in the Arduino IDE?   Possibly there is some linking difference between the two toolsets.

If the IDE works but the P.IO doesn't, that's a good pointer of where to start digging.

earlephilhower, 2024-01-29:
Well, I did some more digging and it does turn out there was a regression in the SPI clock setting optimization and I was able to show your failure, @lazaj .  Trivial fix implemented, and when #1970 merges update and you should be good.

closed 2024-01-29 by earlephilhower

lazaj, 2024-01-29:
Thanks, it works now.

commit ba16fa33 (2024-01-29): Fix initial SPI startup

Clocks were not being set at all after #1934.  Ensure they are by initting
to an impossible 0hz clock on creation and ::end

Fixes #1969
PR #1970 (2024-01-29): Fix initial SPI startup
Clocks were not being set at all after #1934.  Ensure they are by initting to an impossible 0hz clock on creation and ::end

Fixes #1969
commit c64a4a58 (2024-01-29): Fix initial SPI startup (#1970)

Clocks were not being set at all after #1934.  Ensure they are by initting
to an impossible 0hz clock on creation and ::end

Fixes #1969