earlephilhower/arduino-pico #1971: [Feature Request] Wiznet W6100

Wxchris071, 2024-01-29:
Is it possible to add support for the W6100?
I'd like to use IPv6 with WebServerSecure on W6100-EVB-Pico.
The Ethernet Library already supports the W6100 and IPv6, but no Secure Webserver.

earlephilhower, 2024-01-29:
I don't have any W6100 boards, but if you do then you may want to try your hand at writing a driver yourself.

The [W5100](https://github.com/earlephilhower/arduino-pico/tree/master/libraries/lwIP_w5100) or [W5500](https://github.com/earlephilhower/arduino-pico/tree/master/libraries/lwIP_w5500) libraries are very simple and basically set up a raw bypass mode for those chips and stream raw bits in and out.

JAndrassy, 2024-01-29:
here the IPv6 and WebServerSecure are not in the chip. so you can use the m with W5500 too

Wxchris071, 2024-01-30:
@earlephilhower Maybe in my freetime i can try writing a driver for it.
@JAndrassy You're right. I successfully established an IPv6 Connection with WebServerSecure on W5100S-EVB-Pico.

earlephilhower, 2024-02-27:
Closing for now since the other, lower-end WizNet boards are running and give the needed  IPV6

closed 2024-02-27 by earlephilhower