earlephilhower/arduino-pico #1974: Multicore on Arduino RP2040 Nano unsuccessful

tech9492, 2024-01-30:
Hi all,

I am trying to run the code below on an Arduino RP2040 Nano in dual core mode. The upload seem successful based on IDE outputs. However, the built-in LED would stay on and RGB LED flashes. 

Is the setup of my code correct for multicore? 

What are the limitations for multicore? Can WiFi be call on either core?

Thank you.

```
#include <SPI.h>
#include <WiFiNINA.h>
#include <ArduinoMqttClient.h>
#include "arduino_secrets.h"

const int PSON = A3;
const int SSR = 8;
const int FAN = 6;

unsigned long ssrTimer = 0;  // SSR timer
unsigned long fanTimer = 0;  // fan timer

int psState = 0;   // PSON status
int ssState = 0;   // SSR status
int fnState = 0;   // FAN status

//Enter data in secret tab(arduino_secrets.h)
char ssid[] = SECRET_SSID;
char pass[] = SECRET_PASS;

WiFiClient wifiClient;
MqttClient mqttClient(wifiClient);

const char broker[] = "20.0.0.2";
int port = 1883;
const char topic[] = "RP2040Nano/PSon";
const char topic2[] = "RP2040Nano/SSR";
const char topic3[] = "RP2040Nano/FAN";

void setup() {
  // Attempt to connect to WiFi network:
  WiFi.begin(ssid, pass);

  mqttClient.setKeepAliveInterval(0);
  // Username and Password for MQTT authentication
  mqttClient.setUsernamePassword(SECRET_MQTT_USER, SECRET_MQTT_PASS);
  mqttClient.connect(broker, port);
}

void loop() {
  // call poll() regularly to allow the library to send MQTT and prevent disconnection by the broker
  mqttClient.poll();  

  if (WiFi.status() != WL_CONNECTED) {
    WiFi.end();
    WiFi.begin(ssid, pass);
  }

  if (WiFi.status() == WL_CONNECTED && !mqttClient.connected()) {
    mqttClient.stop();
    mqttClient.connect(broker, port);
  }

  if (WiFi.status() == WL_CONNECTED && mqttClient.connected()) {
    tele();
  }
}

void setup1() {
  pinMode(PSON, INPUT);  // analog pin A3 as input
  pinMode(SSR, OUTPUT);  // digital pin 8 as output
  pinMode(FAN, OUTPUT);  // digital pin 6 as output

  digitalWrite(SSR, LOW);
  digitalWrite(FAN, LOW);
}

void loop1() {
  psState = analogRead(PSON);  // read analog input status
  ssState = digitalRead(SSR);  // read digital output status
  fnState = digitalRead(FAN);  // read digital output status

  if (psState > 50) {
    ssrTimer = 100;  // delay SSR for 10 seconds
  }

  psOFF();  // run psOFF

  if (ssState == 1) {
    fanTimer = 100;  // delay Fan for 10 seconds
  }

  fnOFF();  // run psOFF
}

void psOFF() {
  static unsigned long sspreviousTime;
  unsigned long sscurrentTime = millis();  // return time since board active

  if (sscurrentTime - sspreviousTime < 100) {  // move forward if 100 ms has elapsed
    return;
  }
  
  sspreviousTime = sscurrentTime;

  if (!ssrTimer) {
    digitalWrite(SSR, LOW);  // if timer is done or still 0, set digital output LOW
  } else {
    digitalWrite(SSR, HIGH);  // set digital output HIGH
    digitalWrite(FAN, HIGH);  // set ditigal output HIGH
    ssrTimer--;
  }
}

void fnOFF() {
  static unsigned long fnpreviousTime;
  unsigned long fncurrentTime = millis();  // return time since board active

  if (fncurrentTime - fnpreviousTime < 100) {  // move forward if 100 ms has elapsed
    return;
  }

  fnpreviousTime = fncurrentTime;

  if (!fanTimer) {
    digitalWrite(FAN, LOW);  // if timer is done or still 0, set digital output LOW
  } else {
    digitalWrite(FAN, HIGH);  // set digital output HIGH
    fanTimer--;
  }
}

void tele() {
  static unsigned long previousMillis;
  unsigned long currentMillis = millis();

  if (currentMillis - previousMillis >= 10000) {
    // save the last time a message was sent
    previousMillis = currentMillis;

    // send message, the Print interface can be used to set the message contents
    mqttClient.beginMessage(topic);
    mqttClient.print(psState);
    mqttClient.endMessage();

    mqttClient.beginMessage(topic2);
    mqttClient.print(ssState);
    mqttClient.endMessage();

    mqttClient.beginMessage(topic3);
    mqttClient.print(fnState);
    mqttClient.endMessage();

    mqttClient.beginMessage(topic4);
    mqttClient.print(temp_deg);
    mqttClient.endMessage();
  }
}
```

earlephilhower, 2024-01-30:
> What are the limitations for multicore? Can WiFi be call on either core?

First of all, I'm amazed that `WiFiNINA` works on this core.  I would have thought it would need MBED OS stuff that only the official core supports.  It's really not recommended and I have no idea if it works and no way of testing or fixing anything should it not (or, say, should it fail in interesting ways!).

**For your sanity, I would really recommend sticking w/the official MBED OS core from Arduino if you're using their WiFi stack.**

That said, I would bet that if it did work, that it would have the same requirements as the PicoW native WiFi included here, namely only on core 0.  Anything at all doing TCP/IP comms needs to be on core 0.  `WiFiClient`, `WebServer`, `MQTT`, etc. included since they'd use TCP/IP.

Also, I would bet that there are exactly 0 Arduino libraries which are multicore safe.  So you'd need to make sure that all use of any library is on the same core you started it on.

closed 2024-01-30 by earlephilhower

tech9492, 2024-01-30:
Thank you very much for the detailed explanation of your core.

I will stick to the official core then, considering that libraries might not be compatible with multicore mode.

Thank you.