earlephilhower/arduino-pico #1983: uf2conv.py cause warning message during upload phase

AlainF-ing, 2024-02-05:
During upload preparation I receive this Warning message:
.arduino15/packages/rp2040/hardware/rp2040/3.7.0/tools/uf2conv.py:264: SyntaxWarning: invalid escape sequence '\s'
The upload complete correctly.
I have seen a similar warning from Python.
  words = re.split('\s+', line)
 was replaced by   words = re.split('\\s+', line) 
I do not have experience with the "Pull request ..." 
I may be wrong.

I am impressed by this rp2040 lib, very good work !

earlephilhower, 2024-02-05:
I think you meant to change it to the following (adding `r` to make the string a regexp), no?  Your before and after were the same...
````
 words = re.split(r'\s+', line)
````

AlainF-ing, 2024-02-05:
Sorry for this typo.
I actually get rid of the warning using this 
words = re.split('\\s+', line)

Your choice is probably better than mine.

Best regards Earle

commit a41618fa (2024-02-06): Make Python3 re.split() use a r-string (#1985)

Fixes #1983
closed 2024-02-06 by earlephilhower